CadQuery/OCP #1: Compiling

jmwright, 2019-12-20:
The build finished almost instantly, but didn't throw any errors. I'm not sure if I did something wrong or not. I copied `makefile` over from pywrap (all my cloned repos are in the same parent directory):
```
$ cd OCP
$ cp ../pywrap/OCP/makefile ./
```
Then I ran `make`:
```
$ make
g++-8 -shared -L/home/jwright/anaconda3/envs/cpp-py-bindgen/lib -o OCP.cpython-36m-x86_64-linux-gnu.so  -l:libTKMath.so -l:libTKernel.so -l:libTKG2d.so -l:libTKG3d.so -l:libTKGeomBase.so -l:libTKBRep.so -l:libTKGeomAlgo.so -l:libTKTopAlgo.so -l:libTKPrim.so -l:libTKShHealing.so -l:libTKHLR.so -l:libTKMesh.so -l:libTKBO.so -l:libTKBool.so -l:libTKFeat.so -l:libTKOffset.so -l:libTKFillet.so -l:libTKGeomAlgo.so -l:libTKHLR.so -l:libTKMesh.so -l:libTKOffset.so -l:libTKPrim.so -l:libTKShHealing.so -l:libTKTopAlgo.so -l:libTKXMesh.so -l:libTKXSBase.so -l:libTKService.so -l:libTKV3d.so -l:libTKOpenGl.so -l:libTKMeshVS.so -l:libTKBin.so -l:libTKBinL.so -l:libTKBinTObj.so -l:libTKCAF.so -l:libTKCDF.so -l:libTKLCAF.so -l:libTKStd.so -l:libTKStdL.so -l:libTKTObj.so -l:libTKVCAF.so -l:libTKXml.so -l:libTKXmlL.so -l:libTKXmlTObj.so -l:libTKIGES.so -l:libTKSTEP.so -l:libTKSTEP209.so -l:libTKSTEPAttr.so -l:libTKSTEPBase.so -l:libTKSTL.so -l:libTKXSBase.so
```
This is what `ls` shows:
```
$ ls -l
total 48
-rw-rw-r-- 1 jwright jwright  1418 Dec 20 11:01 makefile
-rwxrwxr-x 1 jwright jwright  7232 Dec 20 11:01 OCP.cpython-36m-x86_64-linux-gnu.so
drwxrwxr-x 2 jwright jwright 36864 Dec 20 11:01 src
```
Is that correct?

adam-urbanczyk, 2019-12-20:
I think you need to put the makefile in `src`

jmwright, 2019-12-20:
With makefile in `src` this is what I get:
```
$ make
g++-8 -O3 -Wall --std=c++17 -fPIC  -fpermissive -fvisibility=hidden -fvisibility-inlines-hidden -D_GLIBCXX_USE_CXX11_ABI=0 -I/home/jwright/anaconda3/envs/cpp-py-bindgen/include/python3.6m -I/home/jwright/.local/include/python3.6m -I/home/jwright/anaconda3/envs/cpp-py-bindgen/include/opencascade  -c -o TopLoc_enums.o TopLoc_enums.cpp
In file included from TopLoc_enums.cpp:36:
NCollection.hxx:81:10: fatal error: OCP_specific.inc: No such file or directory
 #include "OCP_specific.inc"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
<builtin>: recipe for target 'TopLoc_enums.o' failed
make: *** [TopLoc_enums.o] Error 1
```

adam-urbanczyk, 2019-12-20:
OK, I forgot to commit copy `OCP_specific.inc`

adam-urbanczyk, 2019-12-30:
I managed to setup a pipeline using azure. Code generation and compilation works. I publish the generated sources and makefile as an artifact. You could download and try it locally.

NB: for now the binary has some missing symbols - I'll try to experiment with different OCCT builds. One of them did work on my local machine

adam-urbanczyk, 2020-01-01:
It compiles and at least imports on azure: [image].

adam-urbanczyk, 2020-01-01:
I added `OCP_specific.inc` - you should be able to compile locally as well.

jmwright, 2020-01-02:
It compiles now. How do I get set up to do the import? I see that a .so library file is created. I tried to run the python repl from within the src directory and then ran `import OCP`, but got the following error:
```
$ python
Python 3.6.6 | packaged by conda-forge | (default, Jul 26 2018, 09:53:17) 
[GCC 4.8.2 20140120 (Red Hat 4.8.2-15)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import OCP
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: generic_type: type "OSD_File" does not have a non-default holder type while its base "OSD_FileNode" does
```
I'm guessing I'm missing a few steps.

adam-urbanczyk, 2020-01-02:
Great, it should import without issues. Could you try with pybind11 2.4.3 (latest release)?

adam-urbanczyk, 2020-01-03:
Note that I updated the `env.yml` file in the pywrap project.

jmwright, 2020-01-03:
It imports now using pybind11 2.4.3. Thanks!

What's the next step? Is it tracking down an OCCT build that has the missing symbols? When you say "build" [above](https://github.com/CadQuery/OCP/issues/1#issuecomment-569759963), do you mean a pre-built version, or are you compiling OCCT yourself?

adam-urbanczyk, 2020-01-04:
Note that I updated the env file. I use prebuilt OCCT. You can play around with the OCP branch of cadquery. I'm still working though on making some tests pass.

adam-urbanczyk, 2020-01-09:
@jmwright any updates? For me quite significant portion of CQ seems to be working on the OCP branch.

jmwright, 2020-01-10:
@adam-urbanczyk How did you set up OCP (or install the compiled libraries) so that the test suite on the branch can find the OCP imports?

jmwright, 2020-01-10:
Nevermind, I was over-thinking it. I got the tests to run.
```
Ran 178 tests in 2.693s

FAILED (failures=1, errors=86)
```
I see some that just look like the function definition changed in the newer OCCT version (`incompatible function arguments`), but I also see `'Tesselator' is not defined`. I'm not sure if they just renamed it, or if they moved it to a new/different package.

I'm also wondering how we're going to replace the visualization infrastructure that PythonOCC seems to have. I do see some errors related to that as well. Maybe that just becomes part of CQ-editor or a separate repo since it's not really CQ's responsibility to display the models?

adam-urbanczyk, 2020-01-10:
With the latest generated OCP (obtained by using HEAD of pywrap; also available as an artifact from azure) I get slightly better result
```
Ran 178 tests in 15.220s

FAILED (failures=8, errors=5)
```

`Tesselator ` is a custom extension of PythonOCC (I'll probably need to reimplement it) and so is the visualization infrastructure. The latter is only used by CQ-editor and I want to implement it on python side because now (almost) all classes are exposed so there is no need to write custom intermediate layer in C++.

jmwright, 2020-01-12:
Interesting, I get 6 errors on my Ubuntu 18.04 system.
```
Ran 178 tests in 7.882s

FAILED (failures=8, errors=6)

```

jmwright, 2020-01-12:
Here's the last part of the test run log if that helps.
```
testZYPlaneBasics (tests.TestWorkplanes.TestWorkplanes) ... ok

======================================================================
ERROR: testEdgeWrapperCenter (tests.TestCadObjects.TestCadObjects)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/cadquery/tests/TestCadObjects.py", line 68, in testEdgeWrapperCenter
    e = self._make_circle()
  File "/home/jwright/cadquery/tests/TestCadObjects.py", line 19, in _make_circle
    circle = gp_Circ(gp_Ax2(gp_Pnt(1, 2, 3), gp.gp_DZ_s()),
AttributeError: type object 'OCP.gp.gp' has no attribute 'gp_DZ_s'

======================================================================
ERROR: testText (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/cadquery/tests/TestCadQuery.py", line 1871, in testText
    .text('CQ 2.0',0.5,-.05,cut=True,halign='left',valign='bottom', font='Sans')
  File "/home/jwright/cadquery/cadquery/cq.py", line 2922, in text
    halign=halign, valign=valign, position=self.plane)
  File "/home/jwright/cadquery/cadquery/occ_impl/shapes.py", line 1520, in makeText
    font_kind = {'regular' : Font_FA_Regular,
NameError: name 'Font_FA_Regular' is not defined

======================================================================
ERROR: testToOCC (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/cadquery/tests/TestCadQuery.py", line 87, in testToOCC
    self.assertEqual(type(r), OCC.TopoDS.TopoDS_Compound)
AttributeError: module 'OCC.Core' has no attribute 'TopoDS'

======================================================================
ERROR: testAMF (tests.TestExporters.TestExporters)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/cadquery/tests/TestExporters.py", line 44, in testAMF
    self._exportBox(exporters.ExportTypes.AMF, ['<amf units', '</object>'])
  File "/home/jwright/cadquery/tests/TestExporters.py", line 29, in _exportBox
    exporters.exportShape(p, eType, s, 0.1)
  File "/home/jwright/cadquery/cadquery/occ_impl/exporters.py", line 90, in exportShape
    tess = tessellate(shape)
  File "/home/jwright/cadquery/cadquery/occ_impl/exporters.py", line 63, in tessellate
    tess = Tesselator(shape.wrapped)
NameError: name 'Tesselator' is not defined

======================================================================
ERROR: testTJS (tests.TestExporters.TestExporters)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/cadquery/tests/TestExporters.py", line 51, in testTJS
    'vertices', 'formatVersion', 'faces'])
  File "/home/jwright/cadquery/tests/TestExporters.py", line 29, in _exportBox
    exporters.exportShape(p, eType, s, 0.1)
  File "/home/jwright/cadquery/cadquery/occ_impl/exporters.py", line 72, in exportShape
    tess = tessellate(shape)
  File "/home/jwright/cadquery/cadquery/occ_impl/exporters.py", line 63, in tessellate
    tess = Tesselator(shape.wrapped)
NameError: name 'Tesselator' is not defined

======================================================================
ERROR: test_repr_html (tests.TestJupyter.TestJupyter)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/cadquery/tests/TestJupyter.py", line 12, in test_repr_html
    html = shape._repr_html_()
  File "/home/jwright/cadquery/cadquery/occ_impl/shapes.py", line 578, in _repr_html_
    return x3d_display(self.wrapped, export_edges=True)
[... 53 lines cut ...]
  File "/home/jwright/cadquery/tests/TestCadQuery.py", line 1275, in testSplitKeepingBoth
    self.assertEqual(7, c.faces().size())
AssertionError: 7 != 9

======================================================================
FAIL: testSplitKeepingBottom (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/cadquery/tests/TestCadQuery.py", line 1294, in testSplitKeepingBottom
    self.assertEqual(7, c.faces().size())
AssertionError: 7 != 9

======================================================================
FAIL: testSplitKeepingHalf (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/cadquery/tests/TestCadQuery.py", line 1260, in testSplitKeepingHalf
    self.assertEqual(7, c.faces().size())
AssertionError: 7 != 9

----------------------------------------------------------------------
Ran 178 tests in 7.882s

FAILED (failures=8, errors=6)
```

adam-urbanczyk, 2020-01-20:
I managed to go down to one error (x3d export) and AFAIR 6 failures. The OCP code is pushed to this repo and the OCP branch of CQ is also updated.

I am mostly happy with the binding process, there two things to improve(protected destructor handling in BRep_Font and return by value form iterators), but the whole project seems to be totally workable. I think I will tag this conf and try to generate for OCCT 7.4 just to see how much work it is to get things running between OCCT releases.

I did not investigate the failed tests in detail but the failures seemed to be related to not properly merged faces after a cut operation. I am somewhat disappointed that this did not work out of the box (cf. https://www.opencascade.com/doc/occt-7.3.0/refman/html/class_shape_upgrade___unify_same_domain.html ), but let's see in 7.4.

What do you think @jmwright ?

jmwright, 2020-01-20:
@adam-urbanczyk It's sounding good. When I tried to run the tests of the OCP branch of cadquery on Ubuntu 18.04 64-bit, I got the following error.
```
testAMF (tests.TestExporters.TestExporters) ... free(): invalid pointer
Fatal Python error: Aborted

Current thread 0x00007fa6fc8cd740 (most recent call first):
  File "/home/jwright/Downloads/repos/cadquery/cadquery/occ_impl/shapes.py", line 1033 in <genexpr>
  File "/home/jwright/Downloads/repos/cadquery/cadquery/occ_impl/shapes.py", line 1033 in <listcomp>
  File "/home/jwright/Downloads/repos/cadquery/cadquery/occ_impl/shapes.py", line 1033 in tessellate
  File "/home/jwright/Downloads/repos/cadquery/cadquery/occ_impl/exporters.py", line 64 in tessellate
  File "/home/jwright/Downloads/repos/cadquery/cadquery/occ_impl/exporters.py", line 86 in exportShape
  File "/home/jwright/Downloads/repos/cadquery/tests/TestExporters.py", line 29 in _exportBox
  File "/home/jwright/Downloads/repos/cadquery/tests/TestExporters.py", line 44 in testAMF
  File "/home/jwright/anaconda3/envs/cpp-py-bindgen/lib/python3.6/unittest/case.py", line 605 in run
  File "/home/jwright/anaconda3/envs/cpp-py-bindgen/lib/python3.6/unittest/case.py", line 653 in __call__
  File "/home/jwright/anaconda3/envs/cpp-py-bindgen/lib/python3.6/unittest/suite.py", line 122 in run
  File "/home/jwright/anaconda3/envs/cpp-py-bindgen/lib/python3.6/unittest/suite.py", line 84 in __call__
  File "/home/jwright/anaconda3/envs/cpp-py-bindgen/lib/python3.6/unittest/runner.py", line 176 in run
  File "./runtests.py", line 21 in <module>
Aborted (core dumped)
```
Please note that I just pulled the latest of what's on the OCP branch. I did not rebuild the conda environment.

jmwright, 2020-01-20:
I'll try rebuilding the environment from `pywrap/t1` when I get a chance and then report back on whether or not that fixes it.

jmwright, 2020-01-21:
I didn't realize `pywrap/t1` was an older release. I'm setting up a clean environment and will compile OCP from master, and then try to run the tests against that.

jmwright, 2020-01-21:
Ok, with OCP built from master, I get:
```
Ran 178 tests in 9.033s

FAILED (failures=8, errors=1)
```
Here's the [X3D exporter from PythonOCC](https://github.com/tpaviot/pythonocc-core/blob/master/src/Display/WebGl/x3dom_renderer.py). It seems like our implementation of that exporter should be placed in [exporters.py](https://github.com/CadQuery/cadquery/blob/master/cadquery/occ_impl/exporters.py) rather than having it in OCP. I'm guessing [this line](https://github.com/tpaviot/pythonocc-core/blob/master/src/Display/WebGl/x3dom_renderer.py#L27) might be an issue though if those objects are needed for the conversion.

> I did not investigate the failed tests in detail but the failures seemed to be related to not properly merged faces after a cut operation. I am somewhat disappointed that this did not work out of [...]

Yes, that is disappointing. If it's not fixed in 7.4 is our only option to always clean/repair faces after cut operations?
Here's the extra output:
```
testZYPlaneBasics (tests.TestWorkplanes.TestWorkplanes) ... ok

======================================================================
ERROR: test_repr_html (tests.TestJupyter.TestJupyter)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cadquery/tests/TestJupyter.py", line 12, in test_repr_html
    html = shape._repr_html_()
  File "/home/jwright/Downloads/repos/cadquery/cadquery/occ_impl/shapes.py", line 582, in _repr_html_
    from .jupyter_tools import x3d_display
  File "/home/jwright/Downloads/repos/cadquery/cadquery/occ_impl/jupyter_tools.py", line 1, in <module>
    from OCC.Display.WebGl.x3dom_renderer import X3DExporter
ModuleNotFoundError: No module named 'OCC'

======================================================================
FAIL: testBasicLines (tests.TestCadQuery.TestCadQuery)
Make a triangluar boss
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cadquery/tests/TestCadQuery.py", line 929, in testBasicLines
    self.assertEqual(6, r1.faces().size())
AssertionError: 6 != 8

======================================================================
FAIL: testCutThroughAll (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cadquery/tests/TestCadQuery.py", line 868, in testCutThroughAll
    self.assertEqual(11, t.faces().size())
AssertionError: 11 != 13

======================================================================
FAIL: testLargestDimension (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cadquery/tests/TestCadQuery.py", line 1023, in testLargestDimension
    self.assertAlmostEqual(8.7, dim, 1)
AssertionError: 8.7 != 8.833459118601274 within 1 places

======================================================================
FAIL: testMakeShellSolid (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cadquery/tests/TestCadQuery.py", line 1934, in testMakeShellSolid
    self.assertEqual(len(solid.Vertices()),4)
AssertionError: 24 != 4

======================================================================
FAIL: testPointList (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cadquery/tests/TestCadQuery.py", line 203, in testPointList
    self.assertEqual(9, body.faces().size())
AssertionError: 9 != 15

======================================================================
FAIL: testSplitKeepingBoth (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cadquery/tests/TestCadQuery.py", line 1275, in testSplitKeepingBoth
    self.assertEqual(7, c.faces().size())
AssertionError: 7 != 9

======================================================================
FAIL: testSplitKeepingBottom (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cadquery/tests/TestCadQuery.py", line 1294, in testSplitKeepingBottom
    self.assertEqual(7, c.faces().size())
AssertionError: 7 != 9

======================================================================
FAIL: testSplitKeepingHalf (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cadquery/tests/TestCadQuery.py", line 1260, in testSplitKeepingHalf
    self.assertEqual(7, c.faces().size())
AssertionError: 7 != 9

----------------------------------------------------------------------
Ran 178 tests in 9.033s

FAILED (failures=8, errors=1)
```

jmwright, 2020-01-21:
Actually, where did `jupyter_tools.py` come from? I don't see it in the codebase until the PythonOCC transition, and it doesn't seem like [jupyter-cadquery](https://github.com/bernhard-42/jupyter-cadquery) is using it.

adam-urbanczyk, 2020-01-21:
I wrote it AFAIR. There is no native x3d support in OCCT so I think this functionality should be removed. I'd like to wrap the OCCT glTF exporter though.

jmwright, 2020-01-21:
Yeah, glTF seems to be where the our community, and a lot of other projects, seem to be moving.

jmwright, 2020-02-06:
@adam-urbanczyk Where are you getting the pre-built versions of OCCT?

adam-urbanczyk, 2020-02-07:
I'm getting them from the conda-forge channel `conda install -c conda-forge occt` (cf. azure pipeline specification),

jmwright, 2020-02-10:
I'm closing this issue now since I've pushed it way outside of it's original scope.

closed 2020-02-10 by jmwright

jmwright, 2020-04-25:
@adam-urbanczyk What's the process to compile this locally now? I can't find the pywrap makefile I used before.

jmwright, 2020-04-25:
Please ignore the previous question. I haven't tried to build OCP in awhile and forgot the workflow. I've got it figured out now.