CadQuery/OCP #2: OCCT Quirks

jmwright, 2020-02-10:
@adam-urbanczyk I'm about 80% of the way through a cqparts update (not counting documentation) to make it compatible with CQ 2.0. During that process, I've been using the OCP branch of CQ. I'm still fixing things, but I've noticed differences in bounding boxes, volumes, and one case where the number of polygons is different. This seems consistent with what you're seeing with OCCT 7.3.x. If it will help, I can post the test output here.

jmwright, 2020-02-12:
One of the most interesting assert failures is this one:
```
======================================================================
FAIL: test_tolerance (t_cqparts.test_codecs.TestGltf)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cqparts/tests/t_cqparts/test_codecs.py", line 354, in test_tolerance
    get_polycount(tolerance=0.2)
AssertionError: 200 not greater than 200
```
The tolerance is not having an effect on the polygon count, which I assume is related to [tessellation](https://github.com/jmwright/cqparts/blob/master/src/cqparts/codec/gltf.py#L506).

Here's the assert that fails: https://github.com/jmwright/cqparts/blob/master/tests/t_cqparts/test_codecs.py#L352

adam-urbanczyk, 2020-02-14:
Could you try with more extreme tolerance values?

jmwright, 2020-02-14:
Switching to the following definitely does make the assert failure go away.
```
self.assertGreater(
            get_polycount(tolerance=0.001),
            get_polycount(tolerance=0.2)
        )
```
With `0.001` the polygon count is 396 instead of 200.

I'll just update that test. It seems like the tessellate functionality is working, it just doesn't act quite like the old one did. One other thing to note is that with OCCT 7.3.x tessellation doesn't seem to hang with high CPU utilization when the tolerance is set to 0.001 and smaller, which is a problem we've had fairly often with the version of OCE we've been running on. Generating 3972 polygons using a tolerance of 0.00001 is not noticeably slower than using the tolerance of 0.2.

jmwright, 2020-02-16:
I'm going to close this issue and open issues for specific issues as needed.

closed 2020-02-16 by jmwright