CadQuery/OCP #3: OCCT 7.4.x Update

jmwright, 2020-02-10:
@adam-urbanczyk Please let me know when the 7.4.x version is ready to test.

adam-urbanczyk, 2020-02-14:
This is what I have for now. I think it is good enough for testing.

```
======================================================================
ERROR: test_repr_html (tests.TestJupyter.TestJupyter)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/adam/cq/cadquery/tests/TestJupyter.py", line 12, in test_repr_html
    html = shape._repr_html_()
  File "/home/adam/cq/cadquery/cadquery/occ_impl/shapes.py", line 580, in _repr_html_
    from .jupyter_tools import x3d_display
  File "/home/adam/cq/cadquery/cadquery/occ_impl/jupyter_tools.py", line 1, in <module>
    from OCC.Display.WebGl.x3dom_renderer import X3DExporter
ModuleNotFoundError: No module named 'OCC'

======================================================================
FAIL: testBoundingBox (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/adam/cq/cadquery/tests/TestCadQuery.py", line 849, in testBoundingBox
    self.assertAlmostEqual(0.0, bb.xmin, 2)
AssertionError: 0.0 != -0.05000010010000022 within 2 places

======================================================================
FAIL: testLargestDimension (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/adam/cq/cadquery/tests/TestCadQuery.py", line 1023, in testLargestDimension
    self.assertAlmostEqual(8.7, dim, 1)
AssertionError: 8.7 != 8.833459118601274 within 1 places

======================================================================
FAIL: testMakeShellSolid (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/adam/cq/cadquery/tests/TestCadQuery.py", line 1937, in testMakeShellSolid
    self.assertEqual(len(solid.Vertices()),4)
AssertionError: 24 != 4

----------------------------------------------------------------------
Ran 178 tests in 17.400s
```

jmwright, 2020-02-14:
Ok, I have the cqparts tests running with OCP generated and built from the latest pywrap master.
```
FAILED (failures=8, errors=4, skipped=4)
```
The errors are the same ones as with OCCT 7.3.x. I'll dig into those. What I'm less sure about is how to interpret the difference in values returned when using OCCT 7.x vs OCE. I guess I should probably assume that OCCT is more correct. Here are a couple of examples of asserts that failed.
```
======================================================================
FAIL: test_simplify (t_cqparts.test_part.MakeSimpleTests)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cqparts/tests/t_cqparts/test_part.py", line 66, in test_simplify
    self.assertAlmostEqual((cbb.xmin, cbb.xmax), (-0.5, 0.5))
  File "/home/jwright/Downloads/repos/cqparts/tests/base.py", line 152, in assertAlmostEqual
    a, b, places=places, msg=msg, delta=delta,
AssertionError: -0.51 != -0.5 within 7 places
```
```
======================================================================
FAIL: test_arithmetic_add_workplane (t_cqparts.test_coordsystem.CoordSystemTests)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cqparts/tests/t_cqparts/test_coordsystem.py", line 280, in test_arithmetic_add_workplane
    -1.6845977586686152, -0.1705982413313848,
  File "/home/jwright/Downloads/repos/cqparts/tests/t_cqparts/test_coordsystem.py", line 54, in assertBoundingBoxEquals
    self.assertAlmostEqual(a, b, places=places)
  File "/home/jwright/Downloads/repos/cqparts/tests/base.py", line 156, in assertAlmostEqual
    first, second, places=places, msg=msg, delta=delta,
AssertionError: -1.31115279738595 != -1.3011527973859502 within 6 places
```

adam-urbanczyk, 2020-02-15:
I pushed one commit to the OCP branch (now "optimal" bounding box is used). It does solve one failure related to dimensions in the CQ test suite. I'd expect that it will have similar effect on cqparts.

jmwright, 2020-02-16:
Awesome, that dropped it down to one failure, and it's so close that it really shouldn't count. I'm inclined to change the assert to use 6 decimal places.
```
======================================================================
FAIL: test_multipart_part (t_cqparts.test_codecs.TestStep)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cqparts/tests/t_cqparts/test_codecs.py", line 244, in test_multipart_part
    self.assertAlmostEqual(thing.bounding_box.xmax, 15)
  File "/home/jwright/Downloads/repos/cqparts/tests/base.py", line 156, in assertAlmostEqual
    first, second, places=places, msg=msg, delta=delta,
AssertionError: 15.0000001 != 15 within 7 places

----------------------------------------------------------------------
```

jmwright, 2020-02-16:
I added `places = 6` to that specific `assertAlmostEquals` call. I'm down to only 4 errors with no assert failures.

Closing this issue to open more specific issues as needed.

closed 2020-02-16 by jmwright