CadQuery/OCP #4: Long-Term Plans

jmwright, 2020-02-24:
@adam-urbanczyk With the imminent release of PythonOCC 7.4.0, what are the long-term plans for OCP? Are we planning to switch the CQ 2.0 codebase over to OCP, or will it simply remain an alternative to PythonOCC?

adam-urbanczyk, 2020-02-25:
I'd say it is still open. I see the following advantages:
* Faster tracking of OCCT (NB: 7.3 --> 7.4 conversion was really quick)
* More complete API
* More control, CQ specific solutions
* Modern binding technology (libclang+pybind11 vs SWIG)

Disadvantages:
* Additional work

I personally think it makes sense to go for it and I'm willing to do the work. But on the other hand I'd prefer not to push my opinion on everyone. I asked if the PythonOCC folks are willing to update their binding technology to pybind11 but that does not seem the case. I did contribute a couple of lines to PythonOCC but in general I'd rather spend time sharpening my `pybind11` i.so. `SWIG` skills.

What is your take on it?

jmwright, 2020-02-25:
I can certainly see the advantages you mention above. Some people may see the lack of display functionality like PythonOCC has as a disadvantage. Personally, I see it as an advantage for CQ. The core of CQ should only concerned with generating the models. It's the execution environment's (CQ-editor, jupyter-cadquery, etc) responsibility to display the models. If someone really wanted to, they could add PythonOCC and Qt/wx to their environment and display their CQ models that way.

I definitely like the idea of being in control of our own bindings, and not depending on another project who's priorities may differ from ours. I'm not knocking the PythonOCC devs. I think PythonOCC is great. It would just be nice to have the upgrade speed, control and flexibility that custom bindings give us.

adam-urbanczyk, 2020-03-21:
In the meantime I got Windows and Mac builds working:

https://github.com/CadQuery/OCP/releases/tag/7.4-alpha

frmdstryr, 2020-03-27:
Just found this repo, looks great! 

Are there any plans on releasing it on conda and/or pypi?

adam-urbanczyk, 2020-03-27:
Yes, definitely conda package, maybe a wheel. You could be also interested in the project that generates OCP https://github.com/CadQuery/pywrap .

frmdstryr, 2020-03-27:
Sounds great! Once it's released I'd like to give it a try.

Yes I saw pywrap as well, looks similar to how [OCCT](https://github.com/LaughlinResearch/pyOCCT) does it, although that project seems to have stopped development.

adam-urbanczyk, 2020-03-27:
Pywrap/OCP was actually inspired by pyOCCT. I started working on pywrap before pyOCCT_binder was open sourced.

adam-urbanczyk, 2020-07-19:
CQ is based on OCP now and is a conda package provided. The ambition is to switch to OCCT 7.5 when it is released.

closed 2020-07-19 by adam-urbanczyk