CadQuery/OCP #15: Some modules/bindings missing...

frmdstryr, 2020-04-25:
Hi! Glad to see OCP has been released!

I tried porting [DeclaraCAD](https://github.com/codelv/declaracad) to OCP with just a find and replace and hit a few errors.

Version:
```
ocp                       7.4                     1_py3.7    cadquery
```

A few errors (after commenting some things out to try and catch others)

```python
# 1st
from OCP.TopTools import (
ImportError: cannot import name 'TopTools_ListIteratorOfListOfShape' from 'OCP.TopTools' (unknown location)

# 2nd
ModuleNotFoundError: No module named 'OCP.STEPCAFControl'; 'OCP' is not a package

# 3rd (Cannot get further without making a lot more changes...)
AttributeError: type object 'OCP.TopoDS.TopoDS' has no attribute 'Vertex_'
AttributeError: type object 'OCP.Font.Font_FontMgr' has no attribute 'GetInstance_'
AttributeError: type object 'OCP.gp.gp' has no attribute 'Origin_'
AttributeError: type object 'OCP.AIS.AIS_Shape' has no attribute 'SelectionMode_
```

Does OCP add bindings for static members in a different way than pyOCCT?

If there's any way I can help test in the future let me know, thanks!

adam-urbanczyk, 2020-04-25:
From the top of my head:

(1) I did not wrap the nested iterators but there should be an `__iter__` method exposed
(2) STEPCAFControl is not wrapped at the moment
(3) Static methods end with `_s`

adam-urbanczyk, 2020-04-26:
@frmdstryr regarding `TopTools_ListIteratorOfListOfShape` I see that you are using it here:
https://github.com/codelv/declaracad/blob/ec6d05eeece2a5b3785cfc4f90c9ab3cf4051323/declaracad/occ/impl/occ_shape.py#L110
it should be possible to skip it completely:
```python
for topo_to_add in occ_seq:
    seq.append(topo_to_add)
```

adam-urbanczyk, 2020-04-26:
BTW: I'll try wrapping `STEPCAFControl` once #14 is solved. If you want to help you could try generating and compiling bindings with this module added to the conf file (`ocp.toml`)

frmdstryr, 2020-04-27:
Ok, I gave another try. Iterating over the `occ_seq` as you mentioned is working well. Replacing `_s` is also working. Do you happen to know what convention python-occ uses? 

Hit one more error...

```
AttributeError: type object 'OCP.Graphic3d.Graphic3d_RenderingParams' has no attribute 'Anaglyph_RedCyan_Optimized'
```

But after commenting out that everything else seems to be working. 

Nice work!

adam-urbanczyk, 2020-04-30:
@frmdstryr I added `STEPCAFControl` in https://github.com/CadQuery/OCP/commit/8aaad0eb0cf2fcb577a91f64ac4f77ad7610d6f7, azure should be finished in a couple of hours and then you can just pull the conda package from our channel. Let me know how it works. If you have access to osx I could use some help in testing the UI related functionality in OCP (`Cocoa_Window` and related).

Some of the fields of `Graphic3d_RenderingParams` are not exposed indeed, for now pywrap is not binding nested types.

jmwright, 2020-05-05:
It seems like the `Interface_Static` binding may be missing as mentioned [here](https://github.com/CadQuery/cadquery/pull/329#issuecomment-623048457).

jmwright, 2020-05-17:
Please ignore my post about `Interface_Static`. It's not `OCP.Interface_Static`, it's `OCP.Interface.Interface_Static` which is present.

adam-urbanczyk, 2021-02-06:
All nested enums should be wrapped now.

PR #42 (2021-02-06): Wrap nested enums for 7.5
This PR adds nested enum support. Will resolve #15 .
closed 2021-02-06 by adam-urbanczyk