CadQuery/OCP #19: Build from sources

bragostin, 2020-05-29:
Thank you for this formidable job, can't wait to try it! I am trying to build OCP from sources, as it is stated that "Building from sources is also possible, for details take a look at azure-pipelines.yml", but I am struggling. 
I already have a stable python environment under Debian 10 so that using Conda is not really appealing to me.
I should be able to use cmake to install OCP but somehow I am not able to generate a CMakeLists file from azure-pipelines.yml.
Any hint on how to attack this problem?

jmwright, 2020-05-29:
I'd wait for @adam-urbanczyk to give the official answer, but here's what I do.

You should be able to grab [an OCP release](https://github.com/CadQuery/OCP/releases/tag/7.4-RC1) and do this:
```
cmake -B build -S "." -G Ninja -DCMAKE_BUILD_TYPE=Release && cmake --build build -j 2 -- -k 0
```
However, it complains about my cmake version (wants 3.16, I had 3.10 on my Ubuntu 16.04 install). I had to alter the line pinning the version of cmake in CMakeLists.txt inside the Ubuntu 16.04 release directory to get it to work.

Alternatively, starting all the way from pywrap I run the following to do an end-to-end build.
```
./run.sh && cd OCP && cmake -B build -S "." -G Ninja -DCMAKE_BUILD_TYPE=Release && cmake --build build -j 2 -- -k 0
```

adam-urbanczyk, 2020-05-29:
To save time I'd seriously consider installing conda to a local dir (in this example `~/miniconda`):
```
wget https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh -O miniconda.sh
bash miniconda.sh -b -p $HOME/miniconda
```
Then activate it like so:
```
source $HOME/miniconda/bin/activate
```
If you insist on building then easiest is to pull source from the release zip and use cmake:
```
cmake -B build_dir -S src_dir -G Ninja && cmake --build build_dir
```
You'll likely need to solve some issues with cmake not being able to find all the dependencies (python, pybind11, OCCT) and such.

I would not recommend going all the way down to the pywrap level -- you'll need to setup non-trivial build environment.

bragostin, 2020-05-30:
@jmwright @adam-urbanczyk 

I started by exploring the building from sources option and I think I am really close:
In ``CMakeLists.txt`` I changed:
``cmake_minimum_required( VERSION 3.13 )``
In ``FindOpenCascade.cmake`` I modified the OpenCascade path to:
``/usr/local/include/opencascade``
Then running ``cmake -B build_dir -S "." -G Ninja && cmake --build build_dir``
it does the compiling and produces this big library file:
``OCP.cpython-37m-x86_64-linux-gnu.so``

What I am missing is where to put it so that CadQuery can find it? I tried the usual libraries paths, but each time CadQuery complains that ``ModuleNotFoundError: No module named 'OCP'
``

By the way, I like conda, and advise my colleagues and friends to use it, it's just that I would like to avoid redundancies and unnecessary installs on my linux machine to preserve stability.

Thanks a lot for your help!

adam-urbanczyk, 2020-05-30:
Before trying CQ I'd run this in the output folder:
```
python -c "import OCP"
```
just to verify that your so is correct. In the end, you need to put it somewhere in the `PYTHONPATH`. Note that the so is python version specific (in your case it is for 3.7).

bragostin, 2020-05-30:
``python -c "import OCP"`` gives
``ImportError: /home/bruno/Downloads/CadQuery/OCP_7.4/OCP-master/OCP/build/OCP.cpython-37m-x86_64-linux-gnu.so: undefined symbol: _ZNK17BRepBlend_AppSurf7SurfaceER18NCollection_Array2I6gp_PntERS0_IdER18NCollection_Array1IdES8_RS6_IiESA_``

I guess it's a problem with my OpenCascade 7.4 install. That's too complex for me to solve, I'll take your advice and try miniconda now.

bragostin, 2020-05-30:
Victory! I simply recompiled OpenCascade 7.4 and now OCP loads seamlessly. I just ran ``Ex001_Simple_Block.py`` and I get my simple block. Thanks for all the advice!

closed 2020-05-30 by bragostin