CadQuery/OCP #20: TopTools_DataMapIteratorOfDataMapOfShapeShape not in OCP.TopTools

bragostin, 2020-06-09:
I am trying to import ``TopTools_DataMapIteratorOfDataMapOfShapeShape`` (https://www.opencascade.com/doc/occt-7.4.0/refman/html/_top_tools___data_map_of_shape_shape_8hxx.html#a61da0422c6bddef13d54239baf54be23)
```python
from OCP.TopTools import TopTools_DataMapIteratorOfDataMapOfShapeShape
```
but it does not seem to be there:
``ImportError: cannot import name 'TopTools_DataMapIteratorOfDataMapOfShapeShape' from 'OCP.TopTools' (unknown location)
``

Is it to be expected?

adam-urbanczyk, 2020-06-09:
I'd need to dig a little to be sure, but these kind of typdefs were often omitted due to compilation issues.  How/where would you like to use such an interator?

bragostin, 2020-06-09:
@adam-urbanczyk 
OCP 7.4 has already solved a lot of issues for me, and it opens new perspectives. Now I am trying to union triply periodic fins with straights walls in the most efficient way. Here is what I tried:
- standard union now works quite fast, if I shave the sides of the fins to be flush with the walls (with OCC 6.9 the shaving stage was already crashing)
- with these flush surfaces gluing is mush faster, but later I have issues with cuts, which I don't get with standard union
- in FreeCAD there is a ``JoinConnect`` function that performs this union successfully without the need to have flush surfaces. From what I understand it performs a General Fusion first to isolate the intersecting elements and after performs a union. It works nicely with my weird fins, so now I am trying to implement this in CadQuery.

``TopTools_DataMapIteratorOfDataMapOfShapeShape`` allows to iterate these intersecting elements.
 That said, I don't think I will need to use it. I can isolate the intersections by number of common faces with other objects or even by volume.

adam-urbanczyk, 2020-06-09:
Could you paste a link to the `JoinConnect` implementation? Maybe it can be done without the offending iterator.

bragostin, 2020-06-09:
It's there: https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Part/BOPTools/JoinAPI.py
I am quite convinced it can be done without TopTools_DataMapIteratorOfDataMapOfShapeShape now.

adam-urbanczyk, 2020-06-09:
Here, I think, is the underlying C++ part https://github.com/FreeCAD/FreeCAD/blob/da4baa74efeac8cd1eb6008c3c549c7d06afbf11/src/Mod/Part/App/TopoShape.cpp#L1950 .
 Looks like they are using `BRepAlgoAPI_BuilderAlgo`. It has an API just like any other bool op. You need the `TopTools_ListOfShape` class which can be iterated over in the normal python way (i.e. `for el in X:`).

bragostin, 2020-06-10:
@adam-urbanczyk  thank you, that's very helpful!

closed 2020-06-10 by bragostin