CadQuery/OCP #21: g++-8 is not present in Ubuntu 16.04 repos

marcus7070, 2020-06-10:
I'm trying to recreate the environment and build process from the various yaml files (in order to package it in Nix) and I'm getting confused.

The `build-bindings-job.yml` contains:
```yaml
parameters:
  vmImage: 'Ubuntu-16.04'
  CXX: g++-8
jobs:
- job: ${{ parameters.name }}
  - ${{ if contains(parameters.vmImage, 'Ubuntu') }}:
    - bash: sudo apt-get install ${{ parameters.CXX }} mesa-common-dev libgl1-mesa-dev libglu1-mesa-dev freeglut3-dev
...
    - ${{ if contains(parameters.vmImage, 'Ubuntu') }}:
      - bash: |
          source activate cpp-py-bindgen && \
          cmake -B build -S "../$(OCP_src)" -G Ninja -DCMAKE_BUILD_TYPE=Release && \
          cmake --build build -j $(n_cores) -- -k 0 && \
          rm -rf build/CMakeFiles
        displayName: 'Compile'
        env:
          CXX: ${{ parameters.CXX }}
```
The Ubuntu 16.04 repo does not contain g++-8: https://packages.ubuntu.com/search?keywords=g%2B%2B-8
Running `sudo apt-get install g++-8` in a fresh Ubuntu 16.04 server results in some combination of gcc5 and clang-8. Here is a screenshot of the apt log.
[image]

I don't know how to get the exact Ubuntu-16.04 image the Azure pipeline uses so I don't know if it's got updated repos or something funny. I'm also not very experienced with c++ compilers. Is OCP intended to to be compiled with clang-8 or gcc8?

adam-urbanczyk, 2020-06-10:
I also don't know how the azure images are constructed. This ppa seems to be of interest: https://launchpad.net/~ubuntu-toolchain-r/+archive/ubuntu/test. I principle it should build with any C++17 compiler provided that it will be compatible with what was used to build OCCT.

marcus7070, 2020-06-10:
Thanks. Nixpkgs builds OCCT with gcc9 and it works fine, so I'll just use that. 

I found this: https://github.com/actions/virtual-environments/blob/master/images/linux/Ubuntu1604-README.md
That is a surprising amount of extra packages.

closed 2020-06-10 by marcus7070