CadQuery/OCP #22: What errors are expected during build?

marcus7070, 2020-06-10:
@adam-urbanczyk, can you share a build log from the 7.4-RC1 build or similar? Including the pywrap phases please.

When I try to recreate your build process I get a heap of warnings and errors but I suspect most of them are actually acceptable. Being able to compare them against the standard build would be a big help.

adam-urbanczyk, 2020-06-10:
The logs of azure ci are publicly visible, you can take a look here:
https://dev.azure.com/cadquery/OCP/_build/results?buildId=546&view=results

Indeed there will be quite some warnings/errors in the C++ code generation phase. You could pull the sources from github release to be sure that you start with the same code.

marcus7070, 2020-06-10:
Perfect, thanks. And thanks for setting out the job stages so cleanly, these logs are easy to read.

closed 2020-06-10 by marcus7070

adam-urbanczyk, 2020-06-10:
NB: you can also find the intermediate results for every build in the artifacts:
https://dev.azure.com/cadquery/OCP/_build/results?buildId=546&view=artifacts&type=publishedArtifacts
It is sometimes handy for debugging.