CadQuery/OCP #24: CadQuery's DXF import test fails under NixOS

marcus7070, 2020-06-13:
I doubt anyone will be able to recreate this error, but it's so weird I thought I'd mention it here incase anyone has any advice.

The complete CadQuery test suite from commit https://github.com/CadQuery/cadquery/tree/f9fe7b1b842bce3a590ec8e6a1becdc2f04bd054 passes with OCP under NixOS, except `testImportDXF`. 

Stepping through with `pdb`, everything looks to be going well up to https://github.com/CadQuery/cadquery/blob/f9fe7b1b842bce3a590ec8e6a1becdc2f04bd054/cadquery/occ_impl/importers.py#L138, 
```python
ShapeAnalysis_FreeBounds.ConnectEdgesToWires_s(edges_in, tol, False, wires_out)
```
`edges_in.IsEmpty()` is False and it appears to contain the correct edges, but after running that line `wires_out` is still empty. No errors are raised.

The test goes on to fail because the CQ object is empty.

I don't use DXF imports so I've just disabled that particular test and carried on. 

One thing that does occur to me is that I use a default OCCT build. Perhaps I need a plugin or to enable something at build time for this?

adam-urbanczyk, 2020-06-13:
Assuming you are using RC1, could you try with HEAD?

marcus7070, 2020-06-13:
Thanks, https://github.com/CadQuery/OCP/commit/0059e425875fb6fa3e8b3f0335c9d08924e6726c solves the problem!

closed 2020-06-13 by marcus7070

adam-urbanczyk, 2020-06-13:
Good I'll mark it as RC2. BTW: `ConnectEdgesToWires` has a weird interface on the C++ side - pointers passed by reference.