CadQuery/OCP #31: header file mismatch

greyltc, 2020-11-16:
The header files in the opencascade folder here don't seem to match those I have for my system opencascade install.
``` bash
$ pwd
/home/REDACTED/git/python-ocp/src/OCP-7.5-RC1
$ diff opencascade/ /usr/include/opencascade/
Only in /usr/include/opencascade/: BOPTest_DrawableShape.hxx                                                                                        
Only in /usr/include/opencascade/: BOPTest.hxx                                                                                                      
Only in /usr/include/opencascade/: BOPTest_Objects.hxx                                                                                              
Only in /usr/include/opencascade/: BRepTest_DrawableHistory.hxx                                                                                     
Only in /usr/include/opencascade/: BRepTest.hxx                                                                                                     
Only in /usr/include/opencascade/: BRepTest_Objects.hxx                                                                                             
Only in /usr/include/opencascade/: CommandWindow.h                                                                                                  
Only in /usr/include/opencascade/: DBRep_DrawableShape.hxx                                                                                          
Only in /usr/include/opencascade/: DBRep_Edge.hxx
Only in /usr/include/opencascade/: DBRep_Edge.lxx
Only in /usr/include/opencascade/: DBRep_Face.hxx
Only in /usr/include/opencascade/: DBRep_Face.lxx
Only in /usr/include/opencascade/: DBRep_HideData.hxx
Only in /usr/include/opencascade/: DBRep.hxx
Only in /usr/include/opencascade/: DBRep_IsoBuilder.hxx
Only in /usr/include/opencascade/: DBRep_ListIteratorOfListOfEdge.hxx
Only in /usr/include/opencascade/: DBRep_ListIteratorOfListOfFace.hxx
Only in /usr/include/opencascade/: DBRep_ListIteratorOfListOfHideData.hxx
Only in /usr/include/opencascade/: DBRep_ListOfEdge.hxx
Only in /usr/include/opencascade/: DBRep_ListOfFace.hxx
Only in /usr/include/opencascade/: DBRep_ListOfHideData.hxx
Only in /usr/include/opencascade/: DDataStd_DrawDriver.hxx
Only in /usr/include/opencascade/: DDataStd_DrawPresentation.hxx
Only in /usr/include/opencascade/: DDataStd.hxx
Only in /usr/include/opencascade/: DDataStd_TreeBrowser.hxx
Only in /usr/include/opencascade/: DDF_AttributeBrowser.hxx
Only in /usr/include/opencascade/: DDF_Browser.hxx
Only in /usr/include/opencascade/: DDF_Data.hxx
Only in /usr/include/opencascade/: DDF.hxx
Only in /usr/include/opencascade/: DDF_ListIteratorOfTransactionStack.hxx
Only in /usr/include/opencascade/: DDF_Transaction.hxx
Only in /usr/include/opencascade/: DDF_TransactionStack.hxx
Only in /usr/include/opencascade/: DDocStd_DrawDocument.hxx
Only in /usr/include/opencascade/: DDocStd.hxx
Only in /usr/include/opencascade/: DNaming_BooleanOperationDriver.hxx
Only in /usr/include/opencascade/: DNaming_BoxDriver.hxx
Only in /usr/include/opencascade/: DNaming_CylinderDriver.hxx
Only in /usr/include/opencascade/: DNaming_DataMapIteratorOfDataMapOfShapeOfName.hxx
Only in /usr/include/opencascade/: DNaming_DataMapOfShapeOfName.hxx
Only in /usr/include/opencascade/: DNaming_FilletDriver.hxx
Only in /usr/include/opencascade/: DNaming.hxx
Only in /usr/include/opencascade/: DNaming_Line3DDriver.hxx
Only in /usr/include/opencascade/: DNaming_PointDriver.hxx
Only in /usr/include/opencascade/: DNaming_PrismDriver.hxx
Only in /usr/include/opencascade/: DNaming_RevolutionDriver.hxx
Only in /usr/include/opencascade/: DNaming_SelectionDriver.hxx
Only in /usr/include/opencascade/: DNaming_SphereDriver.hxx
Only in /usr/include/opencascade/: DNaming_TransformationDriver.hxx
Only in /usr/include/opencascade/: DPrsStd.hxx
Only in /usr/include/opencascade/: Draw_Appli.hxx
Only in /usr/include/opencascade/: Draw_Axis2D.hxx
Only in /usr/include/opencascade/: Draw_Axis3D.hxx
Only in /usr/include/opencascade/: Draw_Box.hxx
Only in /usr/include/opencascade/: Draw_Chronometer.hxx
Only in /usr/include/opencascade/: Draw_Circle2D.hxx
Only in /usr/include/opencascade/: Draw_Circle3D.hxx
Only in /usr/include/opencascade/: Draw_Color.hxx
Only in /usr/include/opencascade/: Draw_ColorKind.hxx
Only in /usr/include/opencascade/: DrawDim_Angle.hxx
Only in /usr/include/opencascade/: DrawDim_Dimension.hxx
Only in /usr/include/opencascade/: DrawDim_Distance.hxx
Only in /usr/include/opencascade/: DrawDim.hxx
Only in /usr/include/opencascade/: DrawDim_PlanarAngle.hxx
Only in /usr/include/opencascade/: DrawDim_PlanarDiameter.hxx
Only in /usr/include/opencascade/: DrawDim_PlanarDimension.hxx
[... 114 lines cut ...]
Only in /usr/include/opencascade/: XDEDRAW_Colors.hxx
Only in /usr/include/opencascade/: XDEDRAW_Common.hxx
Only in /usr/include/opencascade/: XDEDRAW_GDTs.hxx
Only in /usr/include/opencascade/: XDEDRAW.hxx
Only in /usr/include/opencascade/: XDEDRAW_Layers.hxx
Only in /usr/include/opencascade/: XDEDRAW_Notes.hxx
Only in /usr/include/opencascade/: XDEDRAW_Props.hxx
Only in /usr/include/opencascade/: XDEDRAW_Shapes.hxx
Only in /usr/include/opencascade/: XDEDRAW_Views.hxx
Only in /usr/include/opencascade/: XSDRAW_Functions.hxx
Only in /usr/include/opencascade/: XSDRAW.hxx
Only in /usr/include/opencascade/: XSDRAWIGES.hxx
Only in /usr/include/opencascade/: XSDRAWSTEP.hxx
Only in /usr/include/opencascade/: XSDRAWSTLVRML_CoordsMap.hxx
Only in /usr/include/opencascade/: XSDRAWSTLVRML_DataMapIteratorOfCoordsMap.hxx
Only in /usr/include/opencascade/: XSDRAWSTLVRML_DataMapIteratorOfElemNodesMap.hxx
Only in /usr/include/opencascade/: XSDRAWSTLVRML_DataSource3D.hxx
Only in /usr/include/opencascade/: XSDRAWSTLVRML_DataSource.hxx
Only in /usr/include/opencascade/: XSDRAWSTLVRML_DrawableMesh.hxx
Only in /usr/include/opencascade/: XSDRAWSTLVRML_ElemNodesMap.hxx
Only in /usr/include/opencascade/: XSDRAWSTLVRML.hxx
Only in /usr/include/opencascade/: XSDRAW_Vars.hxx
```
Is that on purpose?
I'm able to complete the build with the modified headers here, but not the ones released in the official opencascade 7.5 source tarball.

greyltc, 2020-11-16:
Seems like the problem might be this commit: https://github.com/CadQuery/OCP/commit/6d4f0b28104cc4b3820fb3102bd439f5205d3ee5

greyltc, 2020-11-16:
Here is how my build breaks when I try to build without that modification to ShapeFix_Solid.hxx :
```
[434/620] Building CXX object CMakeFiles/OCP.dir/ShapeFix.cpp.o                                                                                     
FAILED: CMakeFiles/OCP.dir/ShapeFix.cpp.o                                                                                                           /usr/bin/g++-8 -DOCP_EXPORTS -I/home/REDACTED/git/python-ocp/src/OCP-7.5-RC1/OCP -I/usr/include/opencascade -isystem /usr/include/python3.8 -D_FORTIFY_S
OURCE=2 -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt -O3 -DNDEBUG -fPIC -fpermissive -fvisibility=hidden -fvisibility-in
lines-hidden -std=gnu++17 -MD -MT CMakeFiles/OCP.dir/ShapeFix.cpp.o -MF CMakeFiles/OCP.dir/ShapeFix.cpp.o.d -o CMakeFiles/OCP.dir/ShapeFix.cpp.o -c 
/home/REDACTED/git/python-ocp/src/OCP-7.5-RC1/OCP/ShapeFix.cpp                                                                                          
/home/REDACTED/git/python-ocp/src/OCP-7.5-RC1/OCP/ShapeFix.cpp: In function ‘void register_ShapeFix(pybind11::module&)’:                                
/home/REDACTED/git/python-ocp/src/OCP-7.5-RC1/OCP/ShapeFix.cpp:1551:161: error: invalid static_cast from type ‘Standard_Boolean (ShapeFix_Solid::*)(cons
t Message_ProgressRange&)’ {aka ‘bool (ShapeFix_Solid::*)(const Message_ProgressRange&)’} to type ‘Standard_Boolean (ShapeFix_Solid::*)(const int&)’
 {aka ‘bool (ShapeFix_Solid::*)(const int&)’}                                                                                                       
              (Standard_Boolean (ShapeFix_Solid::*)( const int &  ) ) static_cast<Standard_Boolean (ShapeFix_Solid::*)( const int &  ) >(&ShapeFix_S
olid::Perform),                                                                                                                                                                                                                                                                                         
             ^                                                                                                                                      /home/REDACTED/git/python-ocp/src/OCP-7.5-RC1/OCP/ShapeFix.cpp:1552:288: error: invalid static_cast from type ‘Message_ProgressRange’ to type ‘const int
&’                                                                                                                                                  
              R"#(Iterates on shells and performs fixes (calls ShapeFix_Shell for each subshell). The passed progress indicator allows user to consu
lt the current progress stage and abort algorithm if needed.)#"  , py::arg("theProgress")=static_cast<const int &>(Message_ProgressRange ( ))
```

adam-urbanczyk, 2020-11-16:
Yes it is on purpose, otherwise I would not include the headers here.

greyltc, 2020-11-16:
Is there some other way this issue can be solved without modifying the headers?

adam-urbanczyk, 2020-11-16:
Modify headers on the fly, If someone implements it in the wrapper generator.

greyltc, 2020-11-16:
Seems like there could be runtime issues if the wrapper is operating on headers that are not the ones that the opencascade .so files, are built from, no?

greyltc, 2020-11-16:
Do I need to build a modified version of opencascade for use with CadQuery?

greyltc, 2020-11-16:
I think I'm getting a bit confused by this similar convo we had about the opencascade headers in pywrap where you told me the headers included there were not modified and were only included for testing:

https://github.com/CadQuery/pywrap/issues/22

adam-urbanczyk, 2020-11-16:
The header modification is new for 7.5. Long story short, it did not parse proprely without it. Note that the modification is just a forward declaration of a class defined in another header, I'm using a stock build of OCCT.

Pywrap is (one day will be...) a generic binding generator. This repo contains all the input files and CI configuration used for building OCP.

adam-urbanczyk, 2020-11-19:
Do you have further questions / alternative ideas how to solve this or should I close the issue @greyltc ?

greyltc, 2020-11-19:
I don't really have much of substance to add. I don't understand why the header mods are required in the first place and I don't understand why wrapping modified headers wouldn't cause runtime issues when the precompiled binaries aren't aware of the header mods that are being made here.

If these header modifications can cause runtime issues, I'd say this should stay open until that's fixed.

adam-urbanczyk, 2021-01-02:
Forward declaration has no consequences in runtime - closing.

adam-urbanczyk, 2021-01-02:
Actually I can do it on the fly now via the settings file - reopening.

adam-urbanczyk, 2021-01-30:
Solved in #40

closed 2021-01-30 by adam-urbanczyk