CadQuery/OCP #33: Unable to build on FreeBSD

nealie, 2020-12-03:
Since the FreeBSD opencascade port has been updated to 7.5.0 my port of OCP 7.4 using pre-generated files no longer works, so I thought it was time to revisit trying to get it all to generate using bindgen. Unfortunately this still does not work for me, so I thought this was a more suitable forum than the google groups to try and work out why it's failing.

I've attached a log of my failing build, which has quite a lot of bindgen failures, so it's no surprise that the results don't all compile. I've also attached my Makefile (called Makefile.txt because this silly web interface won't accept anything else).

Any insight into what I can do to get this working would be highly appreciated.

[Makefile.txt](https://github.com/CadQuery/OCP/files/5637143/Makefile.txt)

[build.log](https://github.com/CadQuery/OCP/files/5637112/build.log)

adam-urbanczyk, 2020-12-10:
Thanks, we'll switch to 7.5 after CQ2.1 is released. I'll look into this later. Note that you have to use the headers included in this project (AFAIR I did one forward declaration to be able to compile).

nealie, 2020-12-11:
Thanks for the tip. I was compiling against the headers installed on the system. I've changed that now and I get the following error on the first file being compiled (which is different to before):

`FAILED: CMakeFiles/OCP.dir/Aspect.cpp.o 
/usr/bin/c++ -DOCP_EXPORTS -I/usr/home/nealie/ports/cad/py-ocp-pywrap/work/OCP-7bf1e8d/OCP -I/usr/home/nealie/ports/cad/py-ocp-pywrap/work/OCP-7bf1e8d/opencascade -isystem /usr/local/include/python3.7m -isystem /usr/local/include -O2 -pipe -fstack-protector-strong -fno-strict-aliasing -O2 -pipe -fstack-protector-strong -fno-strict-aliasing -fPIC -fpermissive -fvisibility=hidden -fvisibility-inlines-hidden -std=gnu++17 -MD -MT CMakeFiles/OCP.dir/Aspect.cpp.o -MF CMakeFiles/OCP.dir/Aspect.cpp.o.d -o CMakeFiles/OCP.dir/Aspect.cpp.o -c /usr/home/nealie/ports/cad/py-ocp-pywrap/work/OCP-7bf1e8d/OCP/Aspect.cpp
/usr/home/nealie/ports/cad/py-ocp-pywrap/work/OCP-7bf1e8d/OCP/Aspect.cpp:219:79: error: static_cast from 'Atom (Aspect_DisplayConnection::*)(const Aspect_XAtom) const' to 'int (Aspect_DisplayConnection::*)(const Aspect_XAtom) const' is not allowed
             (int (Aspect_DisplayConnection::*)( const Aspect_XAtom  ) const) static_cast<int (Aspect_DisplayConnection::*)( const Aspect_XAtom  ) const>(&Aspect_DisplayConnection::GetAtom),
                                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
`

adam-urbanczyk, 2021-01-02:
Could you try again with the latest master and also share all the logging form the first phase (`parse`)?

nealie, 2021-01-04:
I've updated my port to use the latest prwrap and OCP and with the addition of a few more command line switches to pywrap, everything seems to have gone well. The generated files compile and I at last have a locally generated OCP. Awesome.

The only thing that concerns me is right at the end where some QA tests are done by the ports infrastructure and it complains that various opencascade libraries are lacking an SONAME. I would need to contact the port maintainer for cad/opencascade to see if anything could be done about this.

I've attached a log of the complete build for interest:
[typescript.txt](https://github.com/CadQuery/OCP/files/5765932/typescript.txt)

nealie, 2021-01-04:
I also get the following error when trying to run the latest cadquery (master) using the generated OCP.so:

`Traceback (most recent call last):
  File "test.py", line 1, in <module>
    import cadquery as cq
  File "/usr/local/lib/python3.7/site-packages/cadquery/__init__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "/usr/local/lib/python3.7/site-packages/cadquery/occ_impl/geom.py", line 5, in <module>
    from OCP.gp import gp_Vec, gp_Ax1, gp_Ax3, gp_Pnt, gp_Dir, gp_Trsf, gp_GTrsf, gp, gp_XYZ
ImportError: /usr/local/lib/python3.7/site-packages/OCP.so: Undefined symbol "glRotated"
`
I forgot to mention that I'm using opencascade 7.5.

adam-urbanczyk, 2021-01-04:
Good to hear that there is progress. You have now two distinct issues:

1. looks like you need to link against libGL (or something similar specific for BSD)
2. cadquery is not currently compatible with ocp7.5 (yet).

greyltc, 2021-01-12:
Hey @nealie, Arch Linux also ships opencascade 7.5.0 so I've had to make some mods to get my cadquery package working there with a few caveats:

- I had to completely drop `text()`:
https://github.com/greyltc/cadquery/commit/48bab8ad9e176590f67980a263262b4c6b0b3198.patch
- I had to relax some volume check restrictions:
https://github.com/greyltc/cadquery/commit/3fdf585e640bbee0e41cfd1c3ba817e28e132937.patch
https://github.com/greyltc/cadquery/commit/a0ecf8ec23003859654087ba16343621aaecb8f3.patch

Hope that helps!

adam-urbanczyk, 2021-01-12:
@greyltc  thanks but those are not related to the issues above. 

Note that  CQ does not support 7.5 yet - it will come after 2.1 release.

nealie, 2021-01-21:
Has something changed with the code since I last got it working, as I cannot get it to link to libgl. Before it just magically happened, but I've tried adding it in to cmake using find_package and target_link_libraries, but to no avail. Admittedly I have zero cmake experience so I've been just googling around for ideas.

adam-urbanczyk, 2021-01-27:
Nothing changed in the OCP code regarding OpenGL. Maybe your distro links OCCT with OpenGL using the `PRIVATE` option.

I think that this should do the trick (modify `templates/CMakeLists.j2`):
```
find_package(OpenGL REQUIRED COMPONENTS OpenGL )
target_link_libraries( {{ name }} PRIVATE OpenGL::OpenGL  )
```
Once you confirm that this works for you, I can add this change to the repo.

nealie, 2021-01-29:
I've updated to the latest code from github as the template directory wasn't present on the version I was using. This has necessitated adding VTK as a dependency. The code calls for vtk9, but the FreeBSD opencascade port uses vtk8, so lets hope there isn't a clash.

So, I have a weird problem now:

`===>  Performing out-of-source build
/bin/mkdir -p /usr/home/nealie/ports/cad/py-ocp-pywrap/work/.build
-- Found pybind11: /usr/local/include (found version "2.6.1" )
CMake Warning (dev) at /usr/local/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:426 (message):
  The package name passed to `find_package_handle_standard_args`
  (OPENCASCADE) does not match the name of the calling package (OpenCascade).
  This can lead to problems in calling code that expects `find_package`
  result variables (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  FindOpenCascade.cmake:79 (find_package_handle_standard_args)
  CMakeLists.txt:19 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at CMakeLists.txt:23 (target_link_libraries):
  Cannot specify link libraries for target "OCP" which is not built by this
  project.
`

I can see the project specified as OCP in the CMakeLists.txtx file, so I have no idea why it thinks that it's not building OCP.

nealie, 2021-01-29:
Oops, my mistake. I didn't understand cmake enough to put the lines you told me about into the right place. I think I've corrected that now and I get this: 

`===>  Performing out-of-source build
/bin/mkdir -p /usr/home/nealie/ports/cad/py-ocp-pywrap/work/.build
-- Found pybind11: /usr/local/include (found version "2.6.1" )
CMake Warning (dev) at /usr/local/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:426 (message):
  The package name passed to `find_package_handle_standard_args`
  (OPENCASCADE) does not match the name of the calling package (OpenCascade).
  This can lead to problems in calling code that expects `find_package`
  result variables (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  FindOpenCascade.cmake:79 (find_package_handle_standard_args)
  CMakeLists.txt:18 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
CMake Error at CMakeLists.txt:30 (add_library):
  Target "OCP" links to target "OpenGL::OpenGL" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
`

nealie, 2021-01-29:
The magic incantation seems to be: `target_link_libraries( {{ name }} PRIVATE ${OPENGL_gl_LIBRARY} )`

Everything now builds successfully, but alas all is not yet well:

`Traceback (most recent call last):
  File "test.py", line 1, in <module>
    import cadquery as cq
  File "/usr/local/lib/python3.7/site-packages/cadquery/__init__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "/usr/local/lib/python3.7/site-packages/cadquery/occ_impl/geom.py", line 5, in <module>
    from OCP.gp import gp_Vec, gp_Ax1, gp_Ax3, gp_Pnt, gp_Dir, gp_Trsf, gp_GTrsf, gp, gp_XYZ
ImportError: /usr/local/lib/python3.7/site-packages/OCP.so: Undefined symbol "_ZN19OpenGl_VertexBuffer16unbindFixedColorERKN11opencascade6handleI14OpenGl_ContextEE"
`

adam-urbanczyk, 2021-01-29:
Good, progress. Can you list all undefined symbols in OCP (using `ld`)?

nealie, 2021-02-02:
Here's what ld says:

`/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyUnicode_InternFromString'
/usr/local/bin/ld: /usr/local/lib/libTKIVtk.so.7: undefined reference to `vtkFiltersParallelDIY2_AutoInit_Construct()'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyExc_KeyError'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyExc_NotImplementedError'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyByteArray_Type'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyNumber_Absolute'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyErr_GivenExceptionMatches'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyNumber_Power'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `_PyThreadState_UncheckedGet'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyObject_GC_Track'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyDict_Size'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyExc_MemoryError'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyStaticMethod_New'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyUnicode_DecodeUTF8'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyErr_Print'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyUnicode_FromString'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyNumber_Negative'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyBytes_FromString'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyBaseObject_Type'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyNumber_Check'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyModule_Type'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyThread_tss_create'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyObject_HasAttrString'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyObject_GetBuffer'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyTuple_Pack'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyBytes_AsString'
/usr/local/bin/ld: /lib/libc.so.7: undefined reference to `__progname'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyCapsule_GetContext'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `_PyObject_NextNotImplemented'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyNumber_Subtract'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `_PyObject_GC_New'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `_PyType_Lookup'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyThreadState_Clear'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyNumber_Positive'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyErr_Clear'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyCallable_Check'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyNumber_Xor'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyCapsule_New'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyMem_Calloc'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyLong_AsUnsignedLongLong'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyErr_ExceptionMatches'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyBuffer_Release'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `OpenGl_VertexBuffer::unbindFixedColor(opencascade::handle<OpenGl_Context> const&)'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyBytes_Size'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyNumber_Or'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyObject_GC_Del'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PySequence_Size'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyExc_TypeError'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyNumber_Index'
[... cut ...]
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyLong_FromUnsignedLong'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyStaticMethod_Type'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `PyUnicode_AsUTF8AndSize'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to `PyTuple_Size'
/usr/local/bin/ld: /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1: undefined reference to `_PyByteArray_empty_string'`

adam-urbanczyk, 2021-02-06:
These look interesting:
```
/usr/local/bin/ld: /usr/local/lib/libTKIVtk.so.7: undefined reference to vtkFiltersParallelDIY2_AutoInit_Construct()'
/usr/local/bin/ld: /usr/local/lib/python3.7/site-packages/OCP.so: undefined reference to OpenGl_VertexBuffer::unbindFixedColor(opencascade::handle<OpenGl_Context> const&)'
/usr/local/bin/ld: /usr/local/lib/libTKIVtk.so.7: undefined reference to vtkFiltersParallelDIY2_AutoInit_Destruct()'
```

Only `OpenGl_VertexBuffer::unbindFixedColor`  looks relevant for OCP. Could you add a manual exclusion of this method to ocp.toml and check if this solves the issue?

nealie, 2021-02-09:
I added `OpenGl_VertexBuffer::unbindFixedColor` to `exclude_methods` of `Modules.OpenGl`, but alas, I seem to get exactly the same error. This is with the very latest pywrap and OCP.

adam-urbanczyk, 2021-02-10:
Are you sure it was indeed excluded from the generated code? If so, I'll check locally if this function is not appearring in some inline.

nealie, 2021-02-12:
Here's the patch I applied to ocp.toml. Maybe I haven't specified the exclusion correctly:

`--- ocp.toml.orig       2021-02-06 15:41:02 UTC
+++ ocp.toml
@@ -361,6 +361,10 @@ parsing_header = '''#pragma clang diagnostic ignored "
     modules = ["Xw"]
     symbols = "symbols_mangled_linux.dat"
 
+[FreeBSD]
+    modules = ["Xw"]
+    symbols = "symbols_mangled_linux.dat"
+
 [Windows]
     modules = ["WNT"]
     exclude_classes = ["Handle_*"]
@@ -632,7 +636,8 @@ parsing_header = '''#pragma clang diagnostic ignored "
 struct NSOpenGLContext {};"""
         include_body_pre = 'py::class_<NSOpenGLContext>(m, "NSOpenGLContext");'
         exclude_methods = ["OpenGl_GlCore11::glGetPointerv","OpenGl_RaytraceGeometry::Clear",
-                           "OpenGl_View::ChangeGraduatedTrihedron","OpenGl_View::LayerList"]
+                           "OpenGl_View::ChangeGraduatedTrihedron","OpenGl_View::LayerList",
+                           "OpenGl_VertexBuffer::unbindFixedColor"]
         exclude_typedefs = ["OpenGl_SequenceOfLayers","OpenGl_SetterList","OpenGl_MapOfHatchStylesAndIds","OpenGl_MapOfZLayerSettings","OpenGl_IndexedMapOfStructure","OpenGl_MapOfShaderPrograms","OpenGl_IndexedMapOfStructure","OpenGl_LayerSeqIds"]
         exclude_classes = ["OpenGl_GraduatedTrihedron","OpenGl_LayerList","OpenGl_VariableSetterSelector","OpenGl_RaytraceGeometry"]
         
`

nealie, 2021-02-12:
Here's the patch I applied to ocp.toml, in case I specified the exclusion incorrectly.

[patch-ocp.toml.txt](https://github.com/CadQuery/OCP/files/5970486/patch-ocp.toml.txt)

nealie, 2021-02-23:
I just updated my build to use the latest head of OCP (1abdb76) and now I get a different undefined symbol:

`Traceback (most recent call last):
  File "test.py", line 1, in <module>
    import cadquery as cq
  File "/usr/local/lib/python3.7/site-packages/cadquery/__init__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "/usr/local/lib/python3.7/site-packages/cadquery/occ_impl/geom.py", line 5, in <module>
    from OCP.gp import gp_Vec, gp_Ax1, gp_Ax3, gp_Pnt, gp_Dir, gp_Trsf, gp_GTrsf, gp, gp_XYZ
ImportError: /usr/local/lib/python3.7/site-packages/OCP.so: Undefined symbol "_ZN13BRepAlgo_Loop10SetImageVVERK14BRepAlgo_Image"
`
I'm wondering if it's worth my trying to get anything to work until the code works with opencascade 7.5?

adam-urbanczyk, 2021-03-17:
The code in the head is for 7.5.1 , are you linking against it? I assume so. If all fails, you can generate your own `symbols_mangled_*.dat` and point to it in the `ocp.toml` file.

Here is the script (you'll likely need to modify it to work on BSD): https://github.com/CadQuery/OCP/blob/master/dump_symbols.py

nealie, 2021-04-15:
Just to keep you up to date: I am still working on this, but with no further success. I have generated my own symbols_mangled_freebsd.dat as you suggested, but with no difference (it meant creating yet another port for lief). I have worked using the headers within the OCP repository as well as the locally installed headers. Both compile and link, yet both exhibit the same undefined symbol at execution time as before.

Unfortunately you seem to have skipped from OpenCascade 7.4 to 7.5.1, whereas the FreeBSD port is currently at 7.5. Maybe this is a problem.

I am now trying to enlist help from within the FreeBSD project itself, as this is obviously beyond my skills. If there is no progress there, the only way forward that I can see is to go back to the old method of using your pre-generated Linux files, which would require them to be attached to a release as before. I was trying to avoid that as it means more work for you and probably constant reminders from me for each release, which seemed unnecessary since others seems to have got this process to work.

nealie, 2021-04-15:
Using c++filt I'm able to see that the undefined symbol "_ZN19OpenGl_VertexBuffer16unbindFixedColorERKN11opencascade6handleI14OpenGl_ContextEE" is in fact: `OpenGl_VertexBuffer::unbindFixedColor(opencascade::handle<OpenGl_Context> const&)`

Opencascade (my version at least) defines it as:
`Standard_EXPORT static void unbindFixedColor (const Handle(OpenGl_Context)& theCtx)`

I don't know if that difference is an issue. This is conditional upon: #if !defined(GL_ES_VERSION_2_0)

My OCP library is linked against the following:

`        libvtkWrappingPythonCore-9.0.so.1 => /usr/local/lib/libvtkWrappingPythonCore-9.0.so.1 (0x80067d000)
        libvtkRenderingCore-9.0.so.1 => /usr/local/lib/libvtkRenderingCore-9.0.so.1 (0x8073ac000)
        libGL.so.1 => /usr/local/lib/libGL.so.1 (0x8006bb000)
        libTKMath.so.7 => /usr/local/lib/libTKMath.so.7 (0x807618000)
        libTKernel.so.7 => /usr/local/lib/libTKernel.so.7 (0x807898000)
        libTKG2d.so.7 => /usr/local/lib/libTKG2d.so.7 (0x800752000)
        libTKG3d.so.7 => /usr/local/lib/libTKG3d.so.7 (0x807a79000)
        libTKGeomBase.so.7 => /usr/local/lib/libTKGeomBase.so.7 (0x807c00000)
        libTKBRep.so.7 => /usr/local/lib/libTKBRep.so.7 (0x80809a000)
        libTKGeomAlgo.so.7 => /usr/local/lib/libTKGeomAlgo.so.7 (0x80819f000)
        libTKTopAlgo.so.7 => /usr/local/lib/libTKTopAlgo.so.7 (0x80863e000)
        libTKPrim.so.7 => /usr/local/lib/libTKPrim.so.7 (0x807b95000)
        libTKShHealing.so.7 => /usr/local/lib/libTKShHealing.so.7 (0x8088ff000)
        libTKHLR.so.7 => /usr/local/lib/libTKHLR.so.7 (0x808bce000)
        libTKBO.so.7 => /usr/local/lib/libTKBO.so.7 (0x808ce2000)
        libTKBool.so.7 => /usr/local/lib/libTKBool.so.7 (0x808f29000)
        libTKFeat.so.7 => /usr/local/lib/libTKFeat.so.7 (0x80930e000)
        libTKOffset.so.7 => /usr/local/lib/libTKOffset.so.7 (0x80944b000)
        libTKFillet.so.7 => /usr/local/lib/libTKFillet.so.7 (0x80962f000)
        libTKMesh.so.7 => /usr/local/lib/libTKMesh.so.7 (0x8098d5000)
        libTKXMesh.so.7 => /usr/local/lib/libTKXMesh.so.7 (0x8007d4000)
        libTKXSBase.so.7 => /usr/local/lib/libTKXSBase.so.7 (0x8099d2000)
        libTKService.so.7 => /usr/local/lib/libTKService.so.7 (0x809c66000)
        libTKV3d.so.7 => /usr/local/lib/libTKV3d.so.7 (0x809d8b000)
        libTKOpenGl.so.7 => /usr/local/lib/libTKOpenGl.so.7 (0x80a0b7000)
        libTKMeshVS.so.7 => /usr/local/lib/libTKMeshVS.so.7 (0x80a221000)
        libTKBin.so.7 => /usr/local/lib/libTKBin.so.7 (0x8007d9000)
        libTKBinL.so.7 => /usr/local/lib/libTKBinL.so.7 (0x80a285000)
        libTKBinTObj.so.7 => /usr/local/lib/libTKBinTObj.so.7 (0x80a2e3000)
        libTKCAF.so.7 => /usr/local/lib/libTKCAF.so.7 (0x80a2f8000)
        libTKCDF.so.7 => /usr/local/lib/libTKCDF.so.7 (0x80a393000)
        libTKLCAF.so.7 => /usr/local/lib/libTKLCAF.so.7 (0x80a3f2000)
        libTKStd.so.7 => /usr/local/lib/libTKStd.so.7 (0x80a4db000)
        libTKStdL.so.7 => /usr/local/lib/libTKStdL.so.7 (0x80a604000)
        libTKTObj.so.7 => /usr/local/lib/libTKTObj.so.7 (0x80a67a000)
        libTKVCAF.so.7 => /usr/local/lib/libTKVCAF.so.7 (0x80a6c3000)
        libTKXml.so.7 => /usr/local/lib/libTKXml.so.7 (0x80a704000)
        libTKXmlL.so.7 => /usr/local/lib/libTKXmlL.so.7 (0x80a739000)
        libTKXmlTObj.so.7 => /usr/local/lib/libTKXmlTObj.so.7 (0x80a7ac000)
        libTKIGES.so.7 => /usr/local/lib/libTKIGES.so.7 (0x80a7c0000)
        libTKSTEP.so.7 => /usr/local/lib/libTKSTEP.so.7 (0x80abc3000)
        libTKSTEP209.so.7 => /usr/local/lib/libTKSTEP209.so.7 (0x80af11000)
        libTKSTEPAttr.so.7 => /usr/local/lib/libTKSTEPAttr.so.7 (0x80afba000)
        libTKSTEPBase.so.7 => /usr/local/lib/libTKSTEPBase.so.7 (0x80b0c3000)
        libTKSTL.so.7 => /usr/local/lib/libTKSTL.so.7 (0x80b34f000)
        libTKXDESTEP.so.7 => /usr/local/lib/libTKXDESTEP.so.7 (0x80b369000)
        libTKXCAF.so.7 => /usr/local/lib/libTKXCAF.so.7 (0x80b43b000)
        libTKBinXCAF.so.7 => /usr/local/lib/libTKBinXCAF.so.7 (0x80b514000)
        libTKXmlXCAF.so.7 => /usr/local/lib/libTKXmlXCAF.so.7 (0x80b53b000)
        libTKVRML.so.7 => /usr/local/lib/libTKVRML.so.7 (0x80b56a000)
        libTKRWMesh.so.7 => /usr/local/lib/libTKRWMesh.so.7 (0x80b5fb000)
        libTKIVtk.so.7 => /usr/local/lib/libTKIVtk.so.7 (0x80b68c000)
        libvtkFiltersCore-9.0.so.1 => /usr/local/lib/libvtkFiltersCore-9.0.so.1 (0x80b6cd000)
        libvtkCommonExecutionModel-9.0.so.1 => /usr/local/lib/libvtkCommonExecutionModel-9.0.so.1 (0x80bc25000)
        libvtkCommonDataModel-9.0.so.1 => /usr/local/lib/libvtkCommonDataModel-9.0.so.1 (0x80bcf0000)
        libvtkCommonTransforms-9.0.so.1 => /usr/local/lib/libvtkCommonTransforms-9.0.so.1 (0x80c160000)
        libvtkCommonMisc-9.0.so.1 => /usr/local/lib/libvtkCommonMisc-9.0.so.1 (0x80c192000)
        libvtkCommonMath-9.0.so.1 => /usr/local/lib/libvtkCommonMath-9.0.so.1 (0x80c1af000)
        libvtkCommonCore-9.0.so.1 => /usr/local/lib/libvtkCommonCore-9.0.so.1 (0x80c1d1000)
        libvtksys-9.0.so.1 => /usr/local/lib/libvtksys-9.0.so.1 (0x80c655000)
        libexecinfo.so.1 => /usr/lib/libexecinfo.so.1 (0x80c69c000)
[... 89 lines cut ...]
        libvtkgl2ps-8.2.so.1 => /usr/local/lib/vtk-8.2/libvtkgl2ps-8.2.so.1 (0x813c4a000)
        libGLU.so.1 => /usr/local/lib/libGLU.so.1 (0x813c66000)
        libvtkglew-8.2.so.1 => /usr/local/lib/vtk-8.2/libvtkglew-8.2.so.1 (0x813cfc000)
[... cut ...]
        libintl.so.8 => /usr/local/lib/libintl.so.8 (0x81431a000)
        libvtkFiltersStatistics-8.2.so.1 => /usr/local/lib/vtk-8.2/libvtkFiltersStatistics-8.2.so.1 (0x814328000)
        libvtkImagingFourier-8.2.so.1 => /usr/local/lib/vtk-8.2/libvtkImagingFourier-8.2.so.1 (0x8143f3000)
        libvtkImagingCore-8.2.so.1 => /usr/local/lib/vtk-8.2/libvtkImagingCore-8.2.so.1 (0x814420000)`

So, the question is: why is this symbol missing? Am I missing a library, or is there a mismatch between opencascade and OCP?

adam-urbanczyk, 2021-04-15:
Now it starts to make sense, check #46 for possible solution. TLDR: use the following cmake flag when compiling OCCT:
```
-D USE_GLES2=OFF
```

nealie, 2021-04-16:
Well, I found the option to turn off USE_GLES2 in the FreeBSD opencascade port, and now that symbol doesn't seem to be  a problem.

So, the next problem is as follows:

`ImportError: /usr/local/lib/libTKIVtk.so.7: Undefined symbol "_Z41vtkFiltersParallelDIY2_AutoInit_Constructv"`

which translates to: `vtkFiltersParallelDIY2_AutoInit_Construct()`

Looking at the ldd output from two comments ago I can see that OCP is actually being liked against both vtk 8.2 and vtk 9.0.1. Unfortunately everything else on my system seems to be linked to vtk 8.2, so maybe this is the problem. Will OCP work with vtk 8.2?

adam-urbanczyk, 2021-04-16:
My suggestion would be to link everything against vtk9 (it is BTW interesting how you link against two version of VTK at once). You'll need to patch OCCT (only build related files, not code itself https://github.com/conda-forge/occt-feedstock/blob/master/recipe/vtk.patch) .

I never tried with vtk8.2 , but I assume it will also work. You'll need to change the version here: https://github.com/CadQuery/OCP/blob/e1df3469a39228ce2553aacc3a0238e334d4dd9a/templates/CMakeLists.j2#L20

nealie, 2021-06-22:
I haven't given up on getting this working, but I did get sidetracked by the FreeBSD ports system migration from Subversion to git. I now have a less convenient workflow that I can use, so I can get back to getting this working.

I tried to link OCP against VTK 8.2, but I believe that the port is built without python wrapping, so there's no vtkWrappingPythonCore available. I think this means that the only way forward is to get opencascade working with VTK 9.

I have talked to the FreeBSD ports maintainer for opencascade and he is unable to build against VTK 9, so I was wondering how it's done for OCP? The current version is 7.5.0 and I notice that you are, or at least were, using 7.5.1.

adam-urbanczyk, 2021-06-22:
The lates version of OCCT is 7.5.2 actually. I think you should be able to build VTK8 with python. Regarding linking against VTK9 you'll need a patch (only on build files AFAIR) - you can find it in the conda-forge recipe used for building OCCT.

nealie, 2021-07-05:
I can at last build a functioning OCP. The last problem was VTK and after talking to the FreeBSD cad/opencascade port maintainer, he upgraded to using VTK 9, which was not altogether straightforward as it does not work out of the box. Efforts have been undertaken to ensure that FreeCAD is not broken by this change, which apparently also needed some work, but I am hopeful that those changes will be committed which will allow me to commit my changes and we will finally have a working CadQuery on FreeBSD again. I've had to write six new ports because of new dependencies since my last port, which alas have not been already ported. This means that it will take some time to get everything committed in dependency order, but I am hopeful that it will work again.-

Now all I have to do it get CQ-editor working, but that's another story.

adam-urbanczyk, 2021-07-05:
Good to hear! Out of curiosity what was needed regarding VTK9 and OCCT (other than the patch mentioned above)?

nealie, 2021-07-06:
I'm not totally sure, but here's the proposed patches and subsequent discussion: https://reviews.freebsd.org/D30934

rsmith-nl, 2021-10-03:
> I can at last build a functioning OCP. 

Can you perhaps share your work up till now? I'm on FreeBSD as well, and I'd like to try to get a basic cadquery working (without cq-editor).

rsmith-nl, 2021-10-31:
It works!

There were a couple of minor issues;

* the cadquery port requires the math/nlopt port.
* the distinfo in the cadquery port references CadQuery-cadquery-2.1-20097a9_GH0.tar.gz. This should be CadQuery-cadquery-2.1.1-20097a9_GH0.tar.gz
* the port Makefile for py-ocp references libclang.so from LLVM 9. But the port uses LLVM10, AFAICT.
* nptyping requires https://github.com/ramonhagenaars/typish

I've added these as comments to the relevant bugs in the FreeBSD bugzilla.

rsmith-nl, 2021-10-31:
> I haven't yet tackled CQ-editor, but once this is all done, then I will.

Honestly (and without meaning disrespect), I'm not the biggest fan of cq-editor and IDE's in general; it pulls in the whole of spyder just for syntax highlighting of the python code. :-( And the cadquery examples in the distribution are kind of useless without it.
(Although I understand the attraction for newer users or those migrating for openscad.)

As a long-time UNIX user I favor [DOTADIW](https://en.wikipedia.org/wiki/Unix_philosophy#Do_One_Thing_and_Do_It_Well).
That means using my preferred editor to edit python scripts.
So in my cadquery scripts I export the result to a STEP file.
Then I use a separate viewer for that.

Most of the existing viewers are windows-only or quite heavyweight.
Looking at what we already need for cadquery anyway, I found DRAWEXE that comes with OpenCASCADE. So for the time being I've wrote a small shell-script (https://github.com/rsmith-nl/scripts/blob/master/stepviewer.sh) that uses DRAWEXE to display that STEP-file.
(In the future I might end up porting https://github.com/fougue/mayo.)

Although I haven't explored it properly yet, one could use DRAWEXE to e.g. automatically generate 2D views. Ideal for automated workflows.

adam-urbanczyk, 2021-10-31:
Discussing CQ-editor is a little bit off-topic here but note that you can use external editors with it. Also note that CQ has SVG exporting capabilities for "2D views".

adam-urbanczyk, 2021-10-31:
@nealie given what @rsmith-nl wrote above, can this issue be closed (i.e. it sounds like OCP can be built on FreeBSD)? Or am I missing something?

closed 2021-11-01 by nealie

jmwright, 2022-01-10:
@nealie What's the status is this port? Is there any chance of CadQuery running on FreeBSD on an ARM 64 system?

nealie, 2022-01-17:
Unfortunately the committing of all the ports has stalled due to many, many dependencies and in the meantime I am no longer able to build OCP on FreeBSD. I have temporarily lost to will to continue banging my head against this, but hopefully I can try again once the versions of the required libraries in ports match those required by OCP.