CadQuery/OCP #35: Compiling 7.5 from sources

bragostin, 2020-12-21:
I was trying to compile from sources following the same procedure as in #19 using 
`cmake -B build_dir -S src_dir -G Ninja && cmake --build build_dir`
but there does not seem to be a `CMakeLists.txt` in OCP-7.5-RC1.
I tried using that of OCP_src_Ubuntu-16.04 in 7.4-RC2 but it does not work.
I imagine the compiling procedure has changed?

adam-urbanczyk, 2020-12-21:
Nothing changed AFAIK. What are the steps to reproduce this?

bragostin, 2020-12-21:
From the OCP-7.5-RC1 directory and after creating the `build` directory
`cmake -B build -S . -G Ninja && cmake --build build
CMake Error: The source directory "/home/bruno/Downloads/CadQuery/OCP_7.5/OCP-7.5-RC1" does not appear to contain CMakeLists.txt.`
With 7.4 `CMakeLists.txt` was present in the source directory and compiling worked fine after modifying some paths to required libraries to suit my local installation.

adam-urbanczyk, 2020-12-21:
OK I understand what you mean now. Nothing changed but I did not attach manually the generated sources (there is one manually added file in 7.4 but non in 7.5). If you want to build right away you need generate them using pywrap.

adam-urbanczyk, 2020-12-21:
Here are some pointers: https://github.com/CadQuery/OCP/issues/27#issuecomment-705101291

bragostin, 2020-12-25:
I close this one because I ran into the exact same issues as #27 and was not able to solve them. 
However I finally installed a miniconda environment (which can be done through Debian apt package manager nowadays) and I now have OCP 7.5 installed and it did not break my native Debian python environment. So hurrah!

closed 2020-12-25 by bragostin

adam-urbanczyk, 2020-12-25:
Glad to hear. For my info, what was blocking for you regarding compilation?

bragostin, 2020-12-25:
In the end I get the same error message than in #27 
```
[367/368] Building CXX object CMakeFiles/OCP.dir/math_pre.cpp.o
ninja: build stopped: cannot make progress due to previous errors.
```

bragostin, 2020-12-25:
Although, I just noticed, installing OCP 7.5 conda downgraded cadquery master to cadquery 2.0...
Looks like I can have cadquery master only with OCP 7.4.
Is that to be expected?

adam-urbanczyk, 2020-12-26:
Yes, for now CQ does not work with 7.5. You 'd have to edit it by hand here and there.

bragostin, 2020-12-26:
That's right. For the record I did the following steps to have OCP 7.5 working with Cadquery master:
```
conda install --no-deps -c conda-forge -c cadquery ocp=7.5
conda install --no-deps -c conda-forge -c cadquery occt=7.5
```
and comment line 151 of shapes.py (`Font_BRepTextBuilder,`)

adam-urbanczyk, 2020-12-26:
Yes, AFAIR text related functionality needs fixing.