CadQuery/OCP #45: OCP not Pyside2 compatible on Windows

EmJay276, 2021-03-08:
I am currently porting my old CadQuery 2.0 to the new CadQuery 2.1 using OCP instead of pythonOCC
I have implemented a Viewer in my UI using the OCCT Widget from the CQ-Editor (using PySide2 instead of PyQt5).

On Linux it works fine, but on Windows I cannot get it working.
The problem is probably this line here:

`        return WNT_Window(wid.ascapsule())`
from
https://github.com/CadQuery/CQ-editor/blob/bb61e901e32133e26558dca5984819d1504f757b/cq_editor/widgets/occt_widget.py#L161

which is using the WNT_Window from OCP
https://github.com/CadQuery/OCP/blob/756fc449a3c756e320b255288a7a77b6a474329b/opencascade/WNT_Window.hxx#L64

The issue is, that the used `self.winId()` from 
https://github.com/CadQuery/CQ-editor/blob/bb61e901e32133e26558dca5984819d1504f757b/cq_editor/widgets/occt_widget.py#L153
is not the same for PySide2 and PyQT5. The PyQT5 (pointer?) object has a build in function `.ascapsule()` the PySide2 return value is just an `int`

WNT_Window is expecting a "Aspect_Handle" which is just a `unsigned long` see
https://dev.opencascade.org/doc/occt-6.9.0/refman/html/_aspect___handle_8hxx.html#a7c3dff1b0c57281d595c83c2636d6a44

It is possible to use a int as input for the WNT_Window constructor and just cast it to an "Aspect_Handle", see
https://github.com/tpaviot/pythonocc-core/blob/7e7f74bca95995da89a4f944b50bb939361c6a7b/src/Visualization/Display3d.cpp#L187

Is it possible to add support for a WNT_Window construtor with just an "int" as input that's casted to "Aspect_Handle"?
This would enable using OCP with PySide2 on Windows.

adam-urbanczyk, 2021-03-09:
OCP doe not specifically depend on PyQT or PySide and I think that you can just use the Python C API to crate the appropriate capsule object. `ctypes.pythonapi.PyCapsule_New` and `ctypes.pythonapi.PyCapsule_SetPointer` to be specific

EmJay276, 2021-03-09:
I already tried this, but all I get as a return value from `ctypes.pythonapi.PyCapsule_New` is an int and I can't get `ctypes.pythonapi.PyCapsule_SetPointer` working, I always get a `OSError: exception: access violation reading 0x0000000060329CF8` 
Sorry I'm not familiar with ctypes, can you perhaps provide a example?

adam-urbanczyk, 2021-03-09:
Something along those lines:
```python
import ctypes
PyCapsule_New = ctypes.pythonapi.PyCapsule_New
PyCapsule_New.restype = ctypes.py_object
PyCapsule_New.argtypes = (ctypes.c_void_p,
                          ctypes.c_char_p,
                          ctypes.c_void_p)

capsule = PyCapsule_New( YOURWINIDVALUE, None, None)
```

EmJay276, 2021-03-09:
Wow that works perfectly, thanks a lot!
I didn't get I have to set the restype and the argtypes

If anyone has the same problem, this here is my working version

```python
def _get_window_win(self, wid):

    from OCP.WNT import WNT_Window

    import ctypes
    PyCapsule_New = ctypes.pythonapi.PyCapsule_New
    PyCapsule_New.restype = ctypes.py_object
    PyCapsule_New.argtypes = (ctypes.c_void_p,
                              ctypes.c_char_p,
                              ctypes.c_void_p)

    return WNT_Window(PyCapsule_New(wid, None, None))
```