CadQuery/OCP #46: undefined symbol: _ZN19OpenGl_VertexBuffer16unbindFixedColorERKN11opencascade6handleI14OpenGl_ContextEE

greyltc, 2021-03-27:
I'm having some trouble building e1df346 today. 
Well actually, it builds, It's just unusable:
```python
import OCP
ImportError: ./OCP.cpython-39-x86_64-linux-gnu.so: undefined symbol: _ZN19OpenGl_VertexBuffer16unbindFixedColorERKN11opencascade6handleI14OpenGl_ContextEE
```

I've got opencascade 7.5.1 and vtk 9.0.1. Any ideas?
Maybe the builds of either one of those is missing a feature?

adam-urbanczyk, 2021-03-27:
How are you compiling OCCT? See: https://github.com/Open-Cascade-SAS/OCCT/blob/f7ad1e7e3729c128014a23f18c028c55f50db301/src/OpenGl/OpenGl_VertexBuffer.hxx#L370

greyltc, 2021-03-28:
> How are you compiling OCCT?

Like this: https://aur.archlinux.org/cgit/aur.git/plain/PKGBUILD?h=opencascade-rc

I figured it out. If you build occt with `-D USE_GLES2=ON` you don't get the symbol in the title and you learn you need it when you try to `import OCP`.

closed 2021-03-28 by greyltc

adam-urbanczyk, 2021-03-28:
Great! Note that `USE_GLES2` is OFF by default.