CadQuery/OCP #48: Pickle OCP objects

Jojain, 2021-05-12:
Currently it is not possible to serialize OCP objects using pickle.
```python
>>> a = cq.Workplane().box(5,3,5).val()
>>> with open ("test", "w") as t:
...     pickle.dump(a,t)
...
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
TypeError: cannot pickle 'OCP.TopoDS.TopoDS_Solid' object
```

This could find some use in specific application like in https://github.com/CadQuery/cadquery-plugins/tree/main/plugins/cq_cache
or in exchanging data across python interpreters.

I don't know much about the feasability of this but has this already been considered ?

marcus7070, 2021-05-13:
I expected to find a method `DumpJson` on `OCP.TopoDS.TopoDS_Shape` and `TopoDS_Solid`, but it's not there. 
https://old.opencascade.com/doc/occt-7.5.0/refman/html/class_topo_d_s___shape.html#a009204ad884a120d654c4ab2c071ec5f

@adam-urbanczyk, do you know where it went?

adam-urbanczyk, 2021-05-13:
Pickling is not implemented. You can (de)serialize TopoDS_Shape objects using the native (and lossless) brep format of OCP. Take a look at this PR: https://github.com/CadQuery/cadquery/pull/735 .

DumpJson is not what you want.

Jojain, 2021-09-04:
Closing this as importBrep and exportBrep is now implemented in cadquery

closed 2021-09-04 by Jojain