CadQuery/OCP #49: Error on compiling, some hints for me?

sosyco, 2021-05-17:
Hi,
fantastic work!
To learn more about the used technique I try to build OCP myself.
Environment: Ubuntu 20.4
Installed packages:
`apt-get -y install g++-8 mesa-common-dev libgl1-mesa-dev libglu1-mesa-dev freeglut3-dev libclang-11-dev libocct-draw-dev libocct-data-exchange-dev libocct-foundation-dev libocct-ocaf-dev libocct-visualization-dev occt-draw libfreeimage-dev libfreeimage3 libvtk7-dev rapidjson-dev`

- clone ocp and integrated pywrap: check
- build an install pywrap: check
- using pywrap for OCP... some failure after correct building.
  I doing something wrong!?
  `pywrap -i /usr/include/vtk-7.1/ -i /usr/include/rapidjson/ -i /usr/lib/x86_64-linux-gnu/ -D /usr/lib/x86_64-linux-gnu/libvtkWrappingPython38Core-7.1.so -l /usr/lib/llvm-11/lib/libclang.so all ocp.toml

...
  [W 210517 00:08:35 translation_unit:48] dummy.cxx:157:67: error: expected ';' after expression
100%|██████████████████████████████████████████████████████████████████████████████████████████████████| 313/313 [54:55<00:00, 10.53s/it]
...

100%|██████████████████████████████████████████████████████████████████████████████████████████████████| 313/313 [04:56<00:00,  1.05it/s]
[I 210517 01:34:42 __init__:275] sorting
[I 210517 01:34:42 __init__:293] removing duplicate typedefs

...
Processing module Standard                                                                                                               
Processing module gp                                                                                                                     
Processing module math                                                                                                                   
Processing module BVH                                                                                                                    
  1%|▉                                                                                                   | 3/313 [00:00<00:36,  8.55it/s]
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
...
  File "/usr/local/lib/python3.8/dist-packages/pywrap-0.1.dev0-py3.8.egg/bindgen/macros.j2", line 16, in template
    {% for super in cls.superclass if super in classes or super in typedefs %}, {{super}} {% endfor %}
  File "/usr/local/lib/python3.8/dist-packages/pywrap-0.1.dev0-py3.8.egg/bindgen/macros.j2", line 16, in template
    {% for super in cls.superclass if super in classes or super in typedefs %}, {{super}} {% endfor %}
jinja2.exceptions.UndefinedError: parameter 'typedefs' was not provided
`

thanks for every help

adam-urbanczyk, 2021-05-18:
Are you using correct versions of all the deps?

greyltc, 2021-05-19:
@sosyco what version of jinja do you have there?

sosyco, 2021-05-22:
I think so.
Maybe I misunderstand the sequence of building the packets!?
after building/installing opencascade 7.5.1:
1. cloning OCP
2. cloning pywrap in OCP in subfolder pywrap
3. building pywrap with ```python setup.py build && python setup.py install```
4. jumping back in parent-folder OCP
5. ```pywrap -i /usr/include/vtk-7.1/ -i /usr/include/rapidjson/ -i /usr/include/clang/10/include/ -l /usr/lib/llvm-10/lib/libclang.so all ocp.toml```
  -> ERROR

jmwright, 2021-05-23:
I've been able to replicate this same error on Ubuntu 20.04 using a Python venv.

```
jinja2.exceptions.UndefinedError: parameter 'typedefs' was not provided
```

I probably won't have time to dig into it more this weekend, but I will probably compare version numbers between a working conda environment and the venv next week.

sosyco, 2021-05-23:
Cool. I will dig too.

greyltc, 2021-05-23:
I've noticed that building with jinja version 3.x or later causes this error. You might try rolling jinja back to a 2.x release.

jmwright, 2021-05-24:
Pinning the jinja2 version to 2.11.3 as @adam-urbanczyk did [here](https://github.com/CadQuery/OCP/pull/50) fixed the problem for me.

sosyco, 2021-05-26:
One problem fixed... one left.. 
on last build step i get the following error:
```cmake --build build```
... 
```[ 99%] Building CXX object CMakeFiles/OCP.dir/math.cpp.o
[ 99%] Building CXX object CMakeFiles/OCP.dir/math_pre.cpp.o
gmake[2]: *** No rule to make target 'OPENCASCADE_LIB_TKIVtk-NOTFOUND', needed by 'OCP.cpython-39-x86_64-linux-gnu.so'.  Stop.
gmake[1]: *** [CMakeFiles/Makefile2:79: CMakeFiles/OCP.dir/all] Error 2
gmake: *** [Makefile:87: all] Error 2
```

adam-urbanczyk, 2021-05-26:
Looks like your OCCT is compiled without VTK support. You can either recompile it and enable VTK or change the configuration to not wrap the OCCT-VTK modules (`ocpt.toml`) and not link against them (`FindOpenCascade.cmake`).

chris-morgan, 2021-06-03:
It’s strongly desirable to update this to support jinja2 version 3, because platforms will prefer to use their own packages of the libraries, and those will all be being updated to the latest version. Pinning to an old version is burying your head in the sand.

The AUR package python-ocp-git is failing to build on Arch Linux for this reason.

Fortunately the patch is very simple, and I’ve submitted https://github.com/CadQuery/pywrap/pull/34. python-ocp-git builds with that patch applied.

sosyco, 2021-06-11:
From my site i get a first running system on "bare-docker".
Thanks to everybody!
to try the system:
```docker run --name cadquery-test --restart=always -p 3000:8000 -d sosyco/jupyterhub-cadquery```
Add user:
```docker exec -it cadquery-test bash```
```> useradd -m -s /bin/bash``` 
```> exit```

login as test on http://localhost:3000/user/test/lab
via jupyter-git you can clone and check the examples from https://github.com/bernhard-42/jupyter-cadquery.git 

@chris-morgan:
Thanks for your patch.
I will close this issue after your patch is merged.

PR #56 (2021-07-27): Jinja fix
Will resolve #49
closed 2021-07-27 by adam-urbanczyk