CadQuery/OCP #51: Is OCP suitable for third-party use yet?

leycec, 2021-06-08:
Our crack team of Pythonistas is contemplating which of several Python OCCT bindings to require in an unrelated third-party project. Rejoice, for OCP is one of those! We couldn't help but notice the phenomenal job the CadQuery community has done here. :+1: 

So, let's hear it for our finalists:

* [**pyOCCT.**](https://github.com/trelau/pyOCCT)
* [**pythonocc-core.**](https://github.com/tpaviot/pythonocc-core)
* **OCP,** which goes without saying.

Of the three, OCP superficially seems to be the most bleeding-edge, the most stable, and the least documented – and at least one of those things is good.

The [underlying clang-based `pywrap` binding generator internally leveraged by OCP](https://github.com/CadQuery/pywrap) looks to be well ahead of the [comparable `pybind11` binding generator internally leveraged by `pyOCCT`](https://pybind11.readthedocs.io). Whereas `pywrap` just defers to clang, `pybind11` implements its own ad-hoc C[++] header macro system requiring changes to source code. That makes my hackles rise.

From prior work, I'm intimately familiar with the [similar clang-based Shiboken2 binding generator internally leveraged by PySide2](https://wiki.qt.io/Qt_for_Python/Shiboken). Shiboken2 worked perfectly, which makes me suspicious of binding generators that "go their own way" rather than reuse standard off-the-shelf C[++] parsers like clang. But what do I know? I'm just the disreputable backend guy.

Given how fresh and shiny OCP is, would you advise third-party Python packages to begin requiring OCP as a mandatory runtime dependency... *or would that be a premature act of job suicide?* I suspect the latter but am hoping for the former.

In any case, thanks for all the tremendous outpouring of CadQuery volunteerism. Whether we decide to ditch OCP for something substantially worse or not, this is an incredibly inspiring effort. Let's all change the CAD industry for the better.

adam-urbanczyk, 2021-06-09:
OCP actually uses pybind11 and is inspired by pyOCCT. The difference is the binding generator and some technical details (vtk interop, iostream handling and maybe some additional things - I have not looked at pyOCCT for quite some time). I don't think there are many changes coming, so in principle you could start using it. Depending on the use case you might be better off with a subset of OCP.

leycec, 2021-06-10:
> OCP actually uses pybind11 and is inspired by pyOCCT.

That... makes sense. I was under the foolish misapprehension that PyWrap only leveraged clang; clearly, [these reading glasses need a checkup](https://github.com/CadQuery/pywrap):

> PyWrap is a C++ binding generator using `pybind11`, `libclang` and `jinja`.

Since I'll probably be packaging some or all of this for Gentoo Linux in the near future, it's great to have the heads-up on these dependency trees. :+1: 

> vtk interop

**...this is of extreme interest to us.**

Our scientific stack currently leverages VTK through [PyVista, the official VTK bindings for Python](https://docs.pyvista.org). If OCP has native support for VTK objects (e.g., b-rep and mesh data), that would constitute a joyful game changer for us.

> I don't think there are many changes coming, so in principle you could start using it.

Yay! *cue dancing unicorns* :unicorn: 

> Depending on the use case you might be better off with a subset of OCP.

Right. That was sorta our assumption to. The initial use case is [de]serialization of inscrutable CAD formats from the seventh plane of Hell. So, mostly STEP.

In fact, that's exactly how we discovered OCP! We were:

1. ~~Dangerous obssessed~~ curious about how CadQuery consumed STEP files.
2. Rapidly discovered [the `cadquery.occ_impl.importers.importStep() loader`](https://github.com/CadQuery/cadquery/blob/7b1a99ca8a901e3a76d27743ee98fc76344b362a/cadquery/occ_impl/importers.py#L53).
3. Realized that we'd struck solid open-source gold.

Happiness and this thread ensued.

marcus7070, 2021-06-10:
> **...this is of extreme interest to us.**
>
> Our scientific stack currently leverages VTK through [PyVista, the official VTK bindings for Python](https://docs.pyvista.org). If OCP has native support [...]

You will probably be interested in this PR: https://github.com/CadQuery/cadquery/pull/735/files#diff-f8ccd5a78e524f286864e37bc789e53a7f79d65eb3ad8b9a08ef5d8331944af0R1104 and this comment chain: https://github.com/CadQuery/cadquery/issues/705#issuecomment-816822399

leycec, 2021-06-10:
...this is magical. This also answers everything I've always wanted to know:

* Yes, the OCP API is stable.
* Yes, the OCP API is created with clang and other things.
* Yes, the OCP API has tight VTK integration.
* Yes, the OCP API is wonderful, so everyone should use it.

Thanks for the deep insights, @marcus7070. Go-go, CadQuery ecosystem! <sup>Also, Australia rocks. New Zealand permanent resident, so can confirm.  :australia: </sup>

closed 2021-06-10 by leycec

leycec, 2021-07-17:
**OMG.** I just revisited this conversation after stumbling through Open Cascade documentation on `TopoDS_Shape` objects and realizing I was profoundly out of my depth. In  a late-night frenzy of desperation, I re-clicked through the [amazing `git` changeset you so helpfully linked to](https://github.com/CadQuery/cadquery/pull/735/files#diff-f8ccd5a78e524f286864e37bc789e53a7f79d65eb3ad8b9a08ef5d8331944af0R1104) – only to realize that CadQuery's genius-tier `cadquery.occ_impl.shapes.Shape.toVtkPolyData()` method leveraging OCP's god-tier `OCP.IVtkOCC.IVtkOCC_Shape` class does everything anyone sane could ever want to do with STEP files in Python.

*such... a... relief...*

Welp. Apparently choosing OCP was the best decision I have ever made. Thanks so much yet again for saving the wreckage of our codebase, @marcus7070.

whophil, 2021-10-30:
Wow am I glad I found this thread. @leycec - I am at a similar crossroads as you, and am also interested in OCP -> VTK (and PyVista of course). Has your project progressed? If so have you run into any roadblocks with your OCP + VTK project?