CadQuery/OCP #53: no matches converting function ‘Append’ to type ‘void (class AIS_ManipulatorObjectSequence::*)(const int&)’

MarcWeber, 2021-07-01:
Hi, I try packaging latest version of CadQuery for Nixos.
However I am running into this issue:

```

In file included from /nix/store/m0fb47gw6naidy0zc9c6nsqd3y81h7n8-opencascade-occt-7.5.1/include/opencascade/AIS_DataMapOfIOStatus.hxx:20,
                 from /nix/store/m0fb47gw6naidy0zc9c6nsqd3y81h7n8-opencascade-occt-7.5.1/include/opencascade/AIS_InteractiveContext.hxx:20,
                 from /no-backup/OCP/OCP/AIS.cpp:41:
/nix/store/m0fb47gw6naidy0zc9c6nsqd3y81h7n8-opencascade-occt-7.5.1/include/opencascade/AIS_InteractiveObject.hxx:136:24: note: declared here
  136 |   Standard_EXPORT void SetAspect (const Handle(Prs3d_BasicAspect)& anAspect);
      |                        ^~~~~~~~~
/no-backup/OCP/OCP/AIS.cpp:1912:181: error: no matches converting function ‘Append’ to type ‘void (class AIS_ManipulatorObjectSequence::*)(const int&)’
 1912 |              (void (AIS_ManipulatorObjectSequence::*)( const int &  ) ) static_cast<void (AIS_ManipulatorObjectSequence::*)( const int &  ) >(&AIS_ManipulatorObjectSequence::Append),
      |                                                                                                                                                                                     ^
In file included from /nix/store/m0fb47gw6naidy0zc9c6nsqd3y81h7n8-opencascade-occt-7.5.1/include/opencascade/TColgp_HSequenceOfPnt.hxx:22,
                 from /nix/store/m0fb47gw6naidy0zc9c6nsqd3y81h7n8-opencascade-occt-7.5.1/include/opencascade/PrsDim_Dimension.hxx:38,
                 from /nix/store/m0fb47gw6naidy0zc9c6nsqd3y81h7n8-opencascade-occt-7.5.1/include/opencascade/PrsDim_AngleDimension.hxx:18,
                 from /nix/store/m0fb47gw6naidy0zc9c6nsqd3y81h7n8-opencascade-occt-7.5.1/include/opencascade/AIS_AngleDimension.hxx:17,
                 from /no-backup/OCP/OCP/AIS.cpp:72:
/nix/store/m0fb47gw6naidy0zc9c6nsqd3y81h7n8-opencascade-occt-7.5.1/include/opencascade/NCollection_DefineHSequence.hxx:43:9: note: candidates are: ‘template<class T> void AIS_ManipulatorObjectSequence::Append(const opencascade::handle<TheTransientType>&, typename std::enable_if<std::is_base_of<AIS_ManipulatorObjectSequence, T>::value>::type*)’
   43 |    void Append (const Handle(T)& theOther,      
```

```
    static_cast<py::class_<AIS_ManipulatorObjectSequence ,opencascade::handle<AIS_ManipulatorObjectSequence>  , Standard_Transient >>(klass)
    // constructors
        .def(py::init<  >()  )
        .def(py::init< const NCollection_Sequence<opencascade::handle<AIS_InteractiveObject>> & >()  , py::arg("theOther") )
    // custom constructors
    // methods
        .def("Append",
             /// the failing line is the following
             (void (AIS_ManipulatorObjectSequence::*)( const int &  ) ) static_cast<void (AIS_ManipulatorObjectSequence::*)( const int &  ) >(&AIS_ManipulatorObjectSequence::Append),
             R"#(None)#"  , py::arg("theItem")
          )
        .def("Append",
             (void (AIS_ManipulatorObjectSequence::*)( NCollection_Sequence<opencascade::handle<AIS_InteractiveObject>> &  ) ) static_cast<void (AIS_ManipulatorObjectSequence::*)( NCollection_Sequence<opencascade::handle<AIS_InteractiveObject>> &  ) >(&AIS_ManipulatorObjectSequence::Append),
             R"#(None)#"  , py::arg("theSequence")
          )

```

The biggest problem I face is that I don't know what versions to use.
Eg another issue said vtk_7, however latest seem to use _9.
I tried using occt (opencascade) latest version.

Does anybody have a docker script which compiles this all from source so that I can lookup the exact same versions ?
Or is there a simple distro I can just use where it's packaged ?

marcus7070, 2021-07-03:
Sorry Marc, I missed this. Check out my repo cq-flake, it's a bit ugly but it works. I'll try to answer more later.

marcus7070, 2021-07-04:
> Eg another issue said vtk_7, however latest seem to use _9.

At present VTK 9.0.1, with build flags and patches from https://github.com/conda-forge/vtk-feedstock/tree/fe196aa7e36db84841df1a94271922c054697e79/recipe. But OCP is a rapidly moving target, one of it's goals is to quickly react to new releases of OCCT.

> I tried using occt (opencascade) latest version.

We need 7.5.1 at the moment, with build flags and patches from here: https://github.com/conda-forge/occt-feedstock/tree/c7cba7d3edccf9e874c6170e496992324b85317a/recipe. Note that without those build flags you will experience segfaults during normal operation.

> Does anybody have a docker script which compiles this all from source so that I can lookup the exact same versions ?
> Or is there a simple distro I can just use where it's packaged ?

The only supported install is Conda. I have an Ubuntu VM with Miniconda for when I need to check things in a standard install. @greyltc maintains an AUR package which you might be interested in: https://aur.archlinux.org/packages/python-ocp/

To figure out what Conda does (so you can replicate it in Nix), you need to start with OCP's `environment.yml` (assuming OCP is the end goal you want to install), look up the Conda-forge feedstock for each package, rewind git history until you find the correct version and latest build number. At least, that's how I do https://github.com/marcus7070/cq-flake.

greyltc, 2021-07-04:
@MarcWeber, if you'd like to look at how I've packaged this stuff for Arch, you're probably better off looking at the -git ones right now:
https://aur.archlinux.org/packages/opencascade-rc/
https://aur.archlinux.org/packages/vtk9-java/
https://aur.archlinux.org/packages/python-ocp-git/
https://aur.archlinux.org/packages/python-cadquery-git/
https://aur.archlinux.org/packages/python-cq-editor-git/

MarcWeber, 2021-07-04:
Marcus has done it so the only question is whether it can be merged into official nixpkgs tree somewhen. Closing for now cause there are multiple references which work.

closed 2021-07-04 by MarcWeber