CadQuery/OCP #62: Discussion: conda-forge

whophil, 2021-11-02:
I am trying out OCP in a non-CadQuery project. For this project, we need a conda-forge compatible version of OCP - i.e., it is built with Anaconda compilers and linked against CDT (or otherwise CF-compatinle) libraries.

Is there any interest/progress in building OCP on conda-forge? 

I've noticed that previous versions of CadQuery are on conda-forge, but not the latest ones using OCP.

adam-urbanczyk, 2021-11-02:
What is incompatible at the moment? AFAICT `conda install -c conda-forge -c cadquery OCP` works as expected. OCP builds  depend on the conda-forge build of OCCT.

whophil, 2021-11-02:
This is likely an uncommon use case, but I am _extending_ the bindings of OCP in my project with additional pybind11 code.

This only works if the original binding library (`OCP.so`) and extension library (`my_project.so`) are compiled with the same compiler versions (see https://github.com/pybind/pybind11/issues/2056#issuecomment-570835972). My project is compiled with latest conda-forge Anaconda compilers (which is GCC 11 on Linux), while the CI for the `cadquery` channel's `ocp` package is on G++-8

adam-urbanczyk, 2021-11-02:
I see, interesting. Is 8 with 11 actually giving you issues? I'd worry more about using compatible pybind11 versions.

whophil, 2021-11-02:
Yes, I am seeing issues when trying to use the GCC8 OCP with my GCC11 project (pybind11 incompatible function arguments error). The GCC8 OCP does work when I compile my project with GCC9. I haven't tried GCC10.

On pybind version - I am building my project with pybind 2.6.1 whereas the OCP library is built with pybind 2.4.3. No issues there so far - at least not on Windows/Mac, where I can confirm that my extensions work correctly, because my Windows and Mac CI machines use the same compiler versions as those on the OCP CI machines.

adam-urbanczyk, 2021-11-03:
Regarding conda-forge: I guess yes, but help would be very welcome. NB: not all versions of pybind11 are compatible - your use case might need a strategy how to track that (see https://github.com/pybind/pybind11/blob/6de30d317277c26a72ed1929e09f225b74f23b7a/include/pybind11/detail/internals.h#L29).

whophil, 2021-11-03:
@adam-urbanczyk I can help out with the conda-forge stuff. Which version should I start on - 7.4.0 or one of the 7.5.x RC tags?

whophil, 2021-11-06:
In-progress conda-forge recipe here: https://github.com/conda-forge/staged-recipes/pull/16814

adam-urbanczyk, 2021-12-13:
Regarding the conda-forge building: would this help:

* https://github.com/CadQuery/OCP/blob/occt7.6/CMakeLists.txt
* https://github.com/CadQuery/OCP/blob/occt7.6/env.yml

It should be relatively trivial to use those files from within a conda recipe. I could backport them for 7.5.

whophil, 2021-12-18:
@adam-urbanczyk I can certainly give this a shot on the conda-forge builders.

However I still think it's a little easier to use pre-generated binding code if that can be made available in the releases. I haven't been able to devote a lot of time to the conda-forge recipe recently, but I do intend to get back to it.

adam-urbanczyk, 2021-12-22:
If add me, I can also work on that PR.

adam-urbanczyk, 2022-02-18:
@whophil can you pass `/FORCE:MULTIPLE` to the linker? I think it can be done via `-DCMAKE_MODULE_LINKER_FLAGS` or `-DCMAKE_SHARED_LINKER_FLAGS`.

whophil, 2022-02-18:
@adam-urbanczyk thanks for the suggestion. Seems like `lld-link.exe` didn't like it:

https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=461753&view=logs&j=240f1fee-52bc-5498-a14a-8361bde76ba0&t=06d62fd2-ab49-5add-9ee5-83b6882d7d2b&l=3196

```
[628/628] cmd.exe /C "cd . && %BUILD_PREFIX%\Library\bin\cmake.exe -E vs_link_dll --intdir=CMakeFiles\OCP.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100220~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100220~1.0\x64\mt.exe --manifests  -- C:\PROGRA~1\LLVM\bin\lld-link.exe /nologo @CMakeFiles\OCP.rsp  /out:OCP.cp39-win_amd64.pyd /implib:OCP.lib /pdb:OCP.pdb /dll /version:0.0 /FORCE:MULTIPLE /INCREMENTAL:NO  && cd ."
The syntax of the command is incorrect.
```

It's not clear to me _which_ executable is complaining about bad syntax.

adam-urbanczyk, 2022-02-18:
Yes, I did run out of memory. Maybe it is not a problem on this particular toolchain.

adam-urbanczyk, 2022-02-18:
AFAICT lld-link 13 does support the `/force:multiple` option.

whophil, 2022-02-19:
> Yes, I did run out of memory. Maybe it is not a problem on this particular toolchain.

It is indeed a problem on the conda-forge builders. (Out of memory when using default linker). I will check the LLVM linker version.

adam-urbanczyk, 2022-02-19:
Sill the same cryptic failure. I guess this needs to be checked locally.

whophil, 2022-02-21:
@adam-urbanczyk the conda-forge build is passing now, please take a look when you have a chance. I am not fully convinced that the passing build is repeatable, but if it passed once we are definitely close.

adam-urbanczyk, 2022-02-22:
Great, I'll take a look this week. What was the trick to make it work on win?

whophil, 2022-02-22:
> Great, I'll take a look this week. What was the trick to make it work on win?

Unfortunately, I don't have a single answer. I did a number of things including:
- Adding `/machine:x64` to the linker flags - this linker flag was there when I didn't specify `/FORCE:MULTIPLE`, but for some reason disappeared when specifying it.
- Removing `-j %CPU_COUNT%` - in local testing, it seems to run parallel regardless of if `-j` is specified. It seems that `-j` isn't a flag `nmake` understands, anyway
- Moving the source to a separate folder prior to invoking `cmake`
- Running the `cmake --build build` command twice (on the second pass, it won't do anything it doesn't need to do). But it doesn't seem like the second invocation is actually doing anything at the moment.

After the PR is merged and converted to a real feedstock, I'll do some experimentation to pare the recipe down to the absolute minimum.

In the interest of moving things along - would you leave a note on the PR stating that you are OK being listed as a maintainer? Once that is done, we can get it merged in, and I can start doing some experimentation on the real feedstock.

adam-urbanczyk, 2022-02-22:
Done

whophil, 2022-02-28:
Here is the converted feedstock https://github.com/conda-forge/ocp-feedstock

Thanks @adam-urbanczyk ! Closing this one out.

closed 2022-02-28 by whophil