CadQuery/OCP #63: AIS_DimensiotionMode not in OCCT 7.5.x?

wavexx, 2021-11-03:
I'm trying to build CadQuery from scratch on debian using the system occt 7.5.1.
After having to fiddle a bit to select the appropriate python version, paths and whatnot, I was able to get pywrap to run.

However, I'm getting stuck during the actual build.
What I'm getting very early on is:

```
../src/CadQuery/OCP/pywrap/OCP/AIS.cpp:125:10: fatal error: AIS_DimensionSelectionMode.hxx: No such file or directory
```

As far as I can see, this was removed in OCCT 7.5.0.
Can this be a result of an issue on my end somewhere?

Thanks.

adam-urbanczyk, 2021-11-09:
I don't see this file in the master. Which OCP version are you using?

wavexx, 2021-11-09:
Turns out I was running pywrap from pywrap itself (which also contains ocp.toml/opencascade), not from ocp. After trying again from the correct path though, I'm still running into build issues.
I have many "unnamed" types, such as:

```
OCP/OCP/AIS_pre.cpp:218:16: error: ‘unnamed’ was not declared in this scope
  218 |     py::enum_<(unnamed enum at dummy.cxx:34:1)>(m, "(unnamed enum at dummy.cxx:34:1)",R"#(None)#")
      |                ^~~~~~~
```

adam-urbanczyk, 2021-11-10:
Are you sure that you are using exctly the same configuration as in this repo?

wavexx, 2021-11-10:
Definitely not, since I'm rebuilding on debian unstable. Definitely part of the problem: I wish I could manage to make an official package out of this, which is why I'm insisting on the system's version of the packages (no pip, no conda, full unbundling). The other parts are quite trivial, but OCP isn't, and there's not much documentation.

Can I ask what could be the cause of this?

I tried both pybind11 2.7 and 2.8, however if I understand the process right, this might be type info coming from clang. Current version of clang is debian is already at 10.x, with 13.x being the experimental option (which is what I tried first, since I'm already using it for development).

Are the headers in opencascade modified/hand-picked compared to upstream, or can we use the system's opencascade headers for example in the same major.minor set (7.5.*) directly? Since the system version is fully available (and vtk is fully external anyway), maybe using those every time would be more robust.

Do we need to regenerate the symbols?

adam-urbanczyk, 2021-11-11:
You'll need to share the intermediate files if you need some debugging help.

Note that OCCT <7.5.3 has a serious STEP export bug. If your goal is to package CQ, I'd be against using 7.5.2.

wavexx, 2021-11-11:
Thanks for the heads-up.

Meanwhile I also tried to use clang 10, but it does result in the same. I'll post something soon.

Can I ask a brief overview of how is OCP is normally updated for a newer release?
My understanding is:

- copy new headers to opencascade/ (my question here is, why is this needed? are these hand-modified or is there another reason why this is copied instead of being referenced like for vtk ?)
- update symbols
- eventually adjust ocp.toml
- run pywrap to regenerate OCP itself
- build OCP

Regarding the last step, when actually building, which include path is being expected? System's occt, or this project's opencascade dir? Again, this is because if we have OCCT headers already, why having the local copy, as above.

Also FindOpenCascade itself on my system is throwing warnings and fails to find OCCT in the first place, as well as having trouble setting up the correct python paths. #61 improves quite a bit on that matter.

Grawp, 2021-11-19:
> Thanks for the heads-up.
>
> Meanwhile I also tried to use clang 10, but it does result in the same. I'll post something soon.
>
> Can I ask a brief overview of how is OCP is normally updated for a newer [...]

In my fork https://github.com/Grawp/pywrap/tree/dev , https://github.com/Grawp/OCP/tree/dev , etc.. I live happily without the duplicated headers and other things. I forked them with primary goal of porting CQ to **Nix** (other Nix related effort is https://github.com/marcus7070/cq-flake) but also for cleaning the code a little.

Perhaps you could find some inspiration in it (or in the other Nix related effort) to help you to build it on Debian. Btw. I also use OCCT 7.5.1 but I had to apply some of the build parameters @marcus7070 (kudos to him) is using.

adam-urbanczyk, 2021-11-19:
@wavexx the flow is correct. When I have time I'll add something to the README / wiki here.
@wavexx @Grawp you have been warned about the STEP export issue. Please inform your users that you are distributing crippled versions of CQ.

closed 2021-11-19 by adam-urbanczyk

greyltc, 2021-11-26:
> Turns out I was running pywrap from pywrap itself (which also contains ocp.toml/opencascade), not from ocp. After trying again from the correct path though, I'm still running into build issues. I [...]

@wavexx, I'm pretty sure https://github.com/CadQuery/pywrap/pull/43 solves this.

wavexx, 2021-11-26:
> @wavexx @Grawp you have been warned about the STEP export issue. Please inform your users that you are distributing crippled versions of CQ.

@adam-urbanczyk it's not about using this specific version of OCCT, I'm trying to make something that would build in general under the platform where OCCT/VTK is unbundled. I can always add back version constraints later.

> @wavexx, I'm pretty sure [CadQuery/pywrap#43](https://github.com/CadQuery/pywrap/pull/43) solves this.

:heart: the build is still running, but looks like that does solve the issue indeed! Thanks for looking into this.