CadQuery/OCP #64: Release to include binding code?

whophil, 2021-11-08:
Would it be possible to have the binding code included in the releases? Either bundled or as a separate asset? I believe this would really simplify the conda-forge process, because generating the bindings on the pared-down conda-forge builders has been giving me a lot of trouble.

Relates to #62

adam-urbanczyk, 2021-11-09:
That should be possible. I'll go the separate asset way though.

whophil, 2021-11-09:
Great, thanks @adam-urbanczyk! Is this something you would be able to do for the latest release `7.5.2-beta`?

One other minor point while we're on the topic of releases - the `7.5.2-beta` release zips (and perhaps other versions) are missing the `pywrap` submodule contents.

adam-urbanczyk, 2021-11-28:
I left a comment in #16814, though it seems to be lost. What is exactly the issue with running pywrap in the recipe? I've seen something about `compiler('CXX')` and the location of libclang. I don't see/understand the connection between those two things.

whophil, 2021-11-28:
> I left a comment in #16814, though it seems to be lost. What is exactly the issue with running pywrap in the recipe? I've seen something about `compiler('CXX')` and the location of libclang. I don't see/understand the connection between those two things.

Here is a link to the comment https://github.com/conda-forge/staged-recipes/pull/16814#discussion_r757929286

I ran into a number of issues with running pywrap in the recipe, but I can't say I know the exact problem. Broadly, I was never able to get pywrap to find all the correct includes - the ones from the conda build prefix - and have it generate the bindings correctly.

Initially I tried to use `clang` only, as is done in the CI of this project, and supply a whole bunch of include paths manually (e.g., the `libstdc++` and `clang` include paths from inside the conda prefix), but I couldn't get this to work on all platforms. See commits around https://github.com/conda-forge/staged-recipes/pull/16814/commits/4a373e43f61ba108543c39f298c44cf950805ed4.

Using `compiler('cxx')` in the pywrap step was a hopeful attempt to see if the activation scripts would magically set up the include paths in a way that everything would just work - it did not.

Given these issues, and the fact that you are amenable to bundling the binding code as part of the release, I figured it would be a reasonable workaround to just fetch the binding code and compile it - at least for the initial conda-forge package. Running `pywrap` inside the recipe is always possible for the future.

adam-urbanczyk, 2022-02-10:
7.5.3 has the sources attached. Are you still planning to something with the recipe @whophil ?

whophil, 2022-02-17:
Thanks @adam-urbanczyk ! I will give the conda-forge recipe another shot, but I am short on time at the moment.

closed 2022-02-17 by whophil