CadQuery/OCP #67: 7.5.3p1 seems fine

greyltc, 2021-11-25:
Not an issue.

I've tested master with occt's tag/release [V_7_5_3p1](https://git.dev.opencascade.org/gitweb/?p=occt.git;a=commit;h=b08cd044f97635a40cceb86769b2399527090001) and it seems to work alright. The `V_7_6_0` tag/release doesn't work though.

`pytest` output for cadquery looks like this:
```
========================================= test session starts =========================================
platform linux -- Python 3.9.7, pytest-6.2.5, py-1.10.0, pluggy-0.13.1                                 
rootdir: src/cadquery                                 
plugins: cov-2.12.1, hypothesis-6.17.0, anyio-3.3.4                                                    
collected 438 items                                                                                    
                                                                                                       
tests/test_assembly.py ........................................                                 [  9%] 
tests/test_cad_objects.py ...............................                                       [ 16%] 
tests/test_cadquery.py ........................................................................ [ 32%] 
............................................................................................... [ 54%] 
....                                                                                            [ 55%] 
tests/test_cqgi.py ................                                                             [ 58%] 
tests/test_examples.py ........................................................................ [ 75%] 
..............                                                                                  [ 78%] 
tests/test_exporters.py ..........                                                              [ 80%] 
tests/test_hull.py ..                                                                           [ 81%] 
tests/test_importers.py ....                                                                    [ 82%] 
tests/test_jupyter.py ..                                                                        [ 82%] 
tests/test_selectors.py ...........................................                             [ 92%] 
tests/test_sketch.py ...............                                                            [ 95%] 
tests/test_workplanes.py ..................                                                     [100%] 
                                                                                                       
========================================== warnings summary ===========================================
../../../../../../../../usr/lib/python3.9/site-packages/nptyping/types/_unicode.py:30                  
  /usr/lib/python3.9/site-packages/nptyping/types/_unicode.py:30: DeprecationWarning: `np.unicode` is a
 deprecated alias for `np.compat.unicode`. To silence this warning, use `np.compat.unicode` by itself. 
In the likely event your code does not need to work on Python 2 you can use the builtin `str` for which
 `np.compat.unicode` is itself an alias. Doing this will not modify any behaviour and is safe. If you s
pecifically wanted the numpy scalar type, use `np.str_` here.                                          
  Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-not
es.html#deprecations                                                                                   
    class Unicode(NPType, numpy.unicode, metaclass=_UnicodeMeta):                                      
                                                                                                       
../../../../../../../../usr/lib/python3.9/site-packages/packaging/version.py:127: 2416 warnings        
  /usr/lib/python3.9/site-packages/packaging/version.py:127: DeprecationWarning: Creating a LegacyVersi
on has been deprecated and will be removed in the next major release                                   
    warnings.warn(                                                                                     
                                                                                                       
tests/test_cad_objects.py::TestCadObjects::testCompoundCenter                                          
tests/test_cadquery.py::TestCadQuery::testCombineSolidsInLoop                                          
tests/test_cadquery.py::TestCadQuery::testCubePlugin                                                   
tests/test_cadquery.py::TestCadQuery::testCylinderPlugin                                               
tests/test_cadquery.py::TestCadQuery::testDoubleTwistedLoft                                            
  /home/grey/.cache/paru/clone/python-cadquery-git/src/cadquery/cadquery/utils.py:34: FutureWarning: co
mbineSolids will be removed in the next release.                                                       
    warn(f"{f.__name__} will be removed in the next release.", FutureWarning)                          
                                                                                                       
tests/test_cadquery.py::TestCadQuery::testCutBlindUntilFace                                            
  src/cadquery/tests/test_cadquery.py:3333: Deprecatio
nWarning: Please use assertAlmostEqual instead.                                                        
    self.assertAlmostEquals(wp_ref_regular_cut.val().Volume(), wp.val().Volume())                      
                                                                                                       
tests/test_cadquery.py::TestCadQuery::testExtrudeUntilFace                                             
  src/cadquery/tests/test_cadquery.py:3158: Deprecatio
nWarning: Please use assertAlmostEqual instead.                                                        
    self.assertAlmostEquals(wp_ref_extrude.val().Volume(), wp.val().Volume())                          
                                                                                                       
tests/test_cadquery.py::TestCadQuery::testFindFace                                                     
  src/cadquery/cadquery/utils.py:34: FutureWarning: fi
ndFace will be removed in the next release.
    warn(f"{f.__name__} will be removed in the next release.", FutureWarning)

tests/test_exporters.py::TestExporters::testAMF
tests/test_exporters.py::TestExporters::testSTEP
tests/test_exporters.py::TestExporters::testSTL
tests/test_exporters.py::TestExporters::testSVG
tests/test_exporters.py::TestExporters::testSVGOptions
tests/test_exporters.py::TestExporters::testTJS
  src/cadquery/cadquery/utils.py:34: FutureWarning: ex
portShape will be removed in the next release.
    warn(f"{f.__name__} will be removed in the next release.", FutureWarning)

tests/test_exporters.py::TestExporters::testSTEP
tests/test_exporters.py::TestExporters::testSTL
  src/cadquery/cadquery/utils.py:34: FutureWarning: re
adAndDeleteFile will be removed in the next release.
    warn(f"{f.__name__} will be removed in the next release.", FutureWarning)

-- Docs: https://docs.pytest.org/en/stable/warnings.html
============================ 438 passed, 2433 warnings in 86.72s (0:01:26) ============================
```

closed 2021-11-25 by greyltc

adam-urbanczyk, 2021-11-25:
7.6 does not work == you cannot compile OCP with it? If so, I'm not surprised. There is usually some work needed to make eery rel work

greyltc, 2021-11-26:
yeah, the build fails. I'll make a new issue for that with details in a handful of hours.

whophil, 2021-11-28:
Is the plan to cut a release for OCP 7.5.3? I'm running into some issues getting OCP packaged on conda-forge (https://github.com/conda-forge/staged-recipes/pull/16814), and I'm interested to see if OCP 7.5.3 would resolve.

adam-urbanczyk, 2021-11-28:
I'm currently working on 7.6, so not really. I don't think any of the issues would be related to 7.5.2 vs 7.5.3.

whophil, 2021-11-28:
> I'm currently working on 7.6, so not really. I don't think any of the issues would be related to 7.5.2 vs 7.5.3.

Would you be opposd to releasing 7.5.3 in the interim? If I understand the process correctly it seems like a very small amount of work. It would also be nice to have the binding headers in the release zips as discussed in https://github.com/CadQuery/OCP/issues/64