CadQuery/OCP #68: Compiling OCP for Apple M1 Max (arm64)

bernhard-42, 2021-11-28:
Trying to get CadQuery running on an arm64 CPU based Macbook Pro (M1 Max):
 
I have set up conda via [MambaForge](https://github.com/conda-forge/miniforge) which provides Python and conda packages for  Mac arm64 CPU ([installer](https://github.com/conda-forge/miniforge/releases/latest/download/Mambaforge-MacOSX-x86_64.sh)):

```shell
(base) $ mamba activate cq
(cq) $ file $(which python)
/opt/mamba/envs/cq/bin/python: Mach-O 64-bit executable arm64```
```

# Issue

I came pretty far, however stuck with one final compilation error (the single `pywrap` warning, see below, seems to be unrelated):

```shell
$ cmake --build build -j 8 -- -k 0

[1/2] Building CXX object CMakeFiles/OCP.dir/RWGltf.cpp.o
FAILED: CMakeFiles/OCP.dir/RWGltf.cpp.o
/usr/bin/c++ -DOCP_EXPORTS -I/Users/bernhard/Development/cadquery/OCP/OCP -I/opt/mamba/envs/cq/include/opencascade -isystem /opt/mamba/envs/cq/include -isystem /opt/mamba/envs/cq/include/python3.8 -isystem /opt/mamba/envs/cq/include/vtk-9.0 -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk -fPIC -fpermissive -fvisibility=hidden -fvisibility-inlines-hidden -std=gnu++17 -MD -MT CMakeFiles/OCP.dir/RWGltf.cpp.o -MF CMakeFiles/OCP.dir/RWGltf.cpp.o.d -o CMakeFiles/OCP.dir/RWGltf.cpp.o -c /Users/bernhard/Development/cadquery/OCP/OCP/RWGltf.cpp
In file included from /Users/bernhard/Development/cadquery/OCP/OCP/RWGltf.cpp:6:
In file included from /opt/mamba/envs/cq/include/pybind11/pybind11.h:17:
/opt/mamba/envs/cq/include/pybind11/detail/init.h:63:68: error: no matching constructor for initialization of 'RWGltf_GltfOStreamWriter'
inline Class *construct_or_initialize(Args &&...args) { return new Class{std::forward<Args>(args)...}; }
                                                                   ^    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/mamba/envs/cq/include/pybind11/detail/init.h:178:31: note: in instantiation of function template specialization 'pybind11::detail::initimpl::construct_or_initialize<RWGltf_GltfOStreamWriter, int &, 0>' requested here
            v_h.value_ptr() = construct_or_initialize<Cpp<Class>>(std::forward<Args>(args)...);
                              ^
/opt/mamba/envs/cq/include/pybind11/pybind11.h:1408:14: note: in instantiation of function template specialization 'pybind11::detail::initimpl::constructor<int &>::execute<pybind11::class_<RWGltf_GltfOStreamWriter, shared_ptr<RWGltf_GltfOStreamWriter>>, pybind11::arg, 0>' requested here
        init.execute(*this, extra...);
             ^
/Users/bernhard/Development/cadquery/OCP/OCP/RWGltf.cpp:473:10: note: in instantiation of function template specialization 'pybind11::class_<RWGltf_GltfOStreamWriter, shared_ptr<RWGltf_GltfOStreamWriter>>::def<int &, pybind11::arg>' requested here
        .def(py::init< int & >()  , py::arg("theOStream") )
         ^
/opt/mamba/envs/cq/include/opencascade/RWGltf_GltfOStreamWriter.hxx:21:7: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'int' to 'const RWGltf_GltfOStreamWriter' for 1st argument
class RWGltf_GltfOStreamWriter : public rapidjson::Writer<rapidjson::OStreamWrapper>
      ^
/opt/mamba/envs/cq/include/opencascade/RWGltf_GltfOStreamWriter.hxx:25:3: note: candidate constructor not viable: no known conversion from 'int' to 'rapidjson::OStreamWrapper &' (aka 'BasicOStreamWrapper<basic_ostream<char>> &') for 1st argument
  RWGltf_GltfOStreamWriter (rapidjson::OStreamWrapper& theOStream)
  ^
1 error generated.
ninja: build stopped: cannot make progress due to previous errors.
```

# The steps I used to get there:

```bash
brew install cmake ninja

# Install Mambaforge https://github.com/conda-forge/miniforge
# https://github.com/conda-forge/miniforge/releases/latest/download/Mambaforge-MacOSX-x86_64.sh

mamba create -n cq python=3.8
mamba activate cq
mamba install pybind11 occt=7.5.2 pandas

git clone https://github.com/CadQuery/OCP.git
cd OCP/

git submodule update --init
cd pywrap/
pip install .
cd ..

PREFIX=/Library/Developer/CommandLineTools

SDK=SDKs/MacOSX12.sdk

C_INC=$PREFIX/$SDK/usr/include/
CPP_INC=$PREFIX/$SDK/usr/include/c++/v1/
GL_INC=$PREFIX/$SDK/System/Library/Frameworks/OpenGL.framework/Versions/A/Headers/
VTK_INC=/opt/mamba/envs/cq/include/vtk-9.0/

LIBCLANG=$PREFIX/usr/lib/libclang.dylib

mkdir -p include
rm -f ./include/OpenGL
ln -s $GL_INC ./include/OpenGL
pywrap -i $CPP_INC -i $C_INC -i ./include -i $VTK_INC -l $LIBCLANG all ocp.toml

cmake -B build -S ./OCP -G Ninja -DCMAKE_BUILD_TYPE=Release -DPython3_FIND_STRATEGY=LOCATION -DPython3_ROOT_DIR=$CONDA_PREFIX
cmake --build build -j 8 -- -k 0
```

## pywrap issue

The pywrap run only had one issue:

Since this error appeared in a dump JSON routine, I thought to ignore and compile nevertheless.

```text
 72%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▌                                                          | 225/313 [04:09<02:20,  1.60s/it]
[W 211128 14:04:44 translation_unit:47] ./opencascade/Graphic3d_WorldViewProjState.hxx
[W 211128 14:04:44 translation_unit:48] dummy.cxx:154:5: error: use of undeclared identifier 'OCCT_DUMP_FIELD_VALUE_NUMERICAL'
[W 211128 14:04:44 translation_unit:48] dummy.cxx:154:60: error: expected ';' after expression
[W 211128 14:04:44 translation_unit:48] dummy.cxx:155:5: error: use of undeclared identifier 'OCCT_DUMP_FIELD_VALUE_POINTER'
[W 211128 14:04:44 translation_unit:48] dummy.cxx:155:57: error: expected ';' after expression
[W 211128 14:04:44 translation_unit:48] dummy.cxx:156:5: error: use of undeclared identifier 'OCCT_DUMP_FIELD_VALUE_NUMERICAL'
[W 211128 14:04:44 translation_unit:48] dummy.cxx:156:68: error: expected ';' after expression
[W 211128 14:04:44 translation_unit:48] dummy.cxx:157:5: error: use of undeclared identifier 'OCCT_DUMP_FIELD_VALUE_NUMERICAL'
[W 211128 14:04:44 translation_unit:48] dummy.cxx:157:67: error: expected ';' after expression
```

adam-urbanczyk, 2021-11-28:
Could you share `RWGltf.cpp` and `tmp_filtered.pkl`?

bernhard-42, 2021-11-28:
`tmp_filtered.pkl` has 253M. Shall I load it and extract some smaller parts to share?
Or what would be a good place to share such a big file?

bernhard-42, 2021-11-28:
And now that I think of it, I do have an "OCP" folder with cpp files in the OCP repo, but I actually don't remember/know where it came from. Is this something the process build creates? 
I just started on a clone of OCP that I created some time ago.

bernhard-42, 2021-11-28:
> Is this something the process build creates?

To answer my own question, the OCP folder gets created by the process. Just repeated the steps in a fresh clone.

bernhard-42, 2021-11-28:
btw. I tried it on OCP master branch - if that info helps

bernhard-42, 2021-11-28:
To be able to continue, I commented out  the section of `klass = m.attr("RWGltf_GltfOStreamWriter");` in `OCP/RWGltf.cpp`
Afterwards I was able to create the library:

```shell
$ ls -lh ../OCP/OCP.cpython-38-darwin.so
-rwxr-xr-x  1 bernhard  staff   268M Nov 28 17:31 ../OCP/OCP.cpython-38-darwin.so
$ file ../OCP/OCP.cpython-38-darwin.so
../OCP/OCP.cpython-38-darwin.so: Mach-O 64-bit bundle arm64
```

I then did the following to create a cadquery environment:

```
mamba create -n cq2 python=3.8 pybind11 occt=7.5.2 pandas scipy ezdxf nptyping typing_extensions jupyterlab multimethod nlopt
mamba activate cq2

# "install" OCP
cp ./OCP/OCP.cpython-38-darwin.so $CONDA_PREFIX/lib/python3.8/site-packages/

# install cadquery
git clone https://github.com/cadquery/cadquery.git
cd cadquery
pip install .
```

SUCCESS!

[image]

adam-urbanczyk, 2021-11-28:
You can provide just the part related to `RWGltf_GltfOStreamWriter`. I'll look up the exact location later.

bernhard-42, 2021-11-28:
So the only remaining problem is the Gltf Writer. I did some more research and now think the pywrap error (which seems to come from a missing `#include <Standard_Dump.hxx>`) and the compile error could belong together. But the whole OCP thing is to complex for a non C++ developer.
I am happy to have a working CadQuery environment on an arm64 Mac - and can happily ignore the missing Gltf writer for the time being

bernhard-42, 2021-11-28:
This is the section I commented out:

```c++
    // Class RWGltf_GltfOStreamWriter from ./opencascade/RWGltf_GltfOStreamWriter.hxx
    klass = m.attr("RWGltf_GltfOStreamWriter");

    
    // nested enums

    static_cast<py::class_<RWGltf_GltfOStreamWriter , shared_ptr<RWGltf_GltfOStreamWriter>  >>(klass)
    // constructors
        .def(py::init< int & >()  , py::arg("theOStream") )
    // custom constructors
    // methods
    // methods using call by reference i.s.o. return
    // static methods
    // static methods using call by reference i.s.o. return
    // operators
    // additional methods and static methods
    // properties
    // methods returning by ref wrapped as properties
;
```

The `.def` wants `int`, however the include file `OCP/opencascade/RWGltf_GltfOStreamWriter.hxx` says

```c++
// Copyright (c) 2019 OPEN CASCADE SAS
//
// This file is part of Open CASCADE Technology software library.
//
// This library is free software; you can redistribute it and/or modify it under
// the terms of the GNU Lesser General Public License version 2.1 as published
// by the Free Software Foundation, with special exception defined in the file
// OCCT_LGPL_EXCEPTION.txt. Consult the file LICENSE_LGPL_21.txt included in OCCT
// distribution for complete text of the license and disclaimer of any warranty.
//
// Alternatively, this file may be used under the terms of Open CASCADE
// commercial license or contractual agreement.

#ifndef _RWGltf_GltfOStreamWriter_HeaderFile
#define _RWGltf_GltfOStreamWriter_HeaderFile

#include <rapidjson/prettywriter.h>
#include <rapidjson/ostreamwrapper.h>

//! rapidjson::Writer wrapper for forward declaration.
class RWGltf_GltfOStreamWriter : public rapidjson::Writer<rapidjson::OStreamWrapper>
{
public:
  //! Main constructor.
  RWGltf_GltfOStreamWriter (rapidjson::OStreamWrapper& theOStream)
  : rapidjson::Writer<rapidjson::OStreamWrapper> (theOStream) {}
};

#endif // _RWGltf_GltfOStreamWriter_HeaderFile
```

adam-urbanczyk, 2021-11-28:
Thanks, I was referring to the parts of `tmp_filtered.pkl` though.

bernhard-42, 2021-11-29:
I compared a linux build with my setup and turns out that I had `pybind11` 2.8 instead of 2.6.
I switched to `pybind11` 2.6.2 and suddenly a new `pywrap` warning appeared (or I missed it yesterday):

```text
[W 211129 22:13:46 translation_unit:47] ./opencascade/RWGltf_GltfOStreamWriter.hxx
[W 211129 22:13:46 translation_unit:48] dummy.cxx:7:10: fatal error: 'rapidjson/rapidjson.h' file not found
```

I installed `rapidjson` via `mamba` and added the `mamba` env include path.

Now both pywrap and cmake are successful and the OCP package works.

The final build steps were:

```bash
brew install cmake ninja

# Install Mambaforge https://github.com/conda-forge/miniforge
# https://github.com/conda-forge/miniforge/releases/latest/download/Mambaforge-MacOSX-x86_64.sh

mamba create -n cq python=3.8
mamba activate cq
mamba install pybind11=2.6 occt=7.5.2 pandas rapidjson

git clone https://github.com/CadQuery/OCP.git
cd OCP/

git submodule update --init
cd pywrap/
pip install .
cd ..

PREFIX=/Library/Developer/CommandLineTools

SDK=SDKs/MacOSX12.sdk

C_INC=$PREFIX/$SDK/usr/include/
CPP_INC=$PREFIX/$SDK/usr/include/c++/v1/
GL_INC=$PREFIX/$SDK/System/Library/Frameworks/OpenGL.framework/Versions/A/Headers/
VTK_INC=$CONDA_PREFIX/include/vtk-9.0/
MAMBA_INC=$CONDA_PREFIX/include

LIBCLANG=$PREFIX/usr/lib/libclang.dylib

mkdir -p include
rm -f ./include/OpenGL
ln -s $GL_INC ./include/OpenGL
pywrap -i $CPP_INC -i $C_INC -i ./include -i $VTK_INC -i $MAMBA_INC -l $LIBCLANG all ocp.toml

cmake -B build -S ./OCP -G Ninja -DCMAKE_BUILD_TYPE=Release -DPython3_FIND_STRATEGY=LOCATION -DPython3_ROOT_DIR=$CONDA_PREFIX
cmake --build build -j 8 -- -k 0
```

adam-urbanczyk, 2021-11-29:
Interesting, rapidjson should be a dep of OCCT. Didn't you have it in the original env?

bernhard-42, 2021-11-30:
Yes, maybe it was, I didn't pay too much attention when I (re-) installed ...

However, I don't think the pywrap error message around rapidjson appeared with pybind11 2.8. 

Anyways, switching to pybind11 2.6 solved the problem for me. And it is the same version that you use for Linux.

bernhard-42, 2021-11-30:
Are you planning to build both libs for MacOS in the future?

And maybe for the intel library, would it be possible to switch to RELEASE (as I did above). The current DEBUG version has 1.3GB unpacked.

Finally, will you support the latest occt 7.6.0 soon or do you maybe prefer to wait for 7.6.1

adam-urbanczyk, 2021-11-30:
Pywrap parsing does not use pybind11 at all - sounds unlikely that the version has impact. 

Which intel library? OCP on mac is built with DEBUG - need to fix that.

Yes, I'm working on 7.6.

bernhard-42, 2021-11-30:
> Pywrap parsing does not use pybind11 at all - sounds unlikely that the version has impact.

OK, then I don't know why it suddenly worked. Anyways, it works :-)

> Which intel library?

yes, I meant OCP on Mac.

> Yes, I'm working on 7.6.

cool, thanks!

bernhard-42, 2022-02-16:
@adam-urbanczyk Don't want to open a separate thread now, just a question:

Would it be possible to build a Universal library for the Mac for 7.6.x (that works for ARM and Intel)?

A universal library looks like that:

```bash
$ file /usr/lib/zsh/5.8/zsh/termcap.so
/usr/lib/zsh/5.8/zsh/termcap.so: Mach-O universal binary with 2 architectures: [x86_64:Mach-O 64-bit bundle x86_64] [arm64e:Mach-O 64-bit bundle arm64e]
/usr/lib/zsh/5.8/zsh/termcap.so (for architecture x86_64):	Mach-O 64-bit bundle x86_64
/usr/lib/zsh/5.8/zsh/termcap.so (for architecture arm64e):	Mach-O 64-bit bundle arm64e
```

I made a test and as soon as I add `-arch arm64 -arch x86_64` to the linker step of an example, It creates the universal (fat) binary.

bernhard-42, 2022-02-17:
fyi: https://dev.opencascade.org/content/testing-occt-apple-m1-arm64

bernhard-42, 2022-02-18:
Got it. I now ran pytest against OCP 7.5.3.

Again a segmentation fault for `tests/test_cadquery.py::TestCadQuery::testBrepImportExport`

```
Fatal Python error: Segmentation fault in occ_impl/shapes.py", line 467 in exportBrep:

    BRepTools.Write_s(self.wrapped, f)
```

The rest is working (fine with some warnings)

adam-urbanczyk, 2023-01-25:
conda-forge cross-compiles for osx-arm64 now

closed 2023-01-25 by adam-urbanczyk