CadQuery/OCP #79: Expose OCC_VERSION

bernhard-42, 2022-02-20:
I currently try to ensure that my newest version of _Jupyter-Cadquery_ will work with both OCCT 7.5 and 7.6. 
Unfortunately, `Poly_PolygonOnTriangulation` which I use to discretise edges has a breaking change:

7.5 only has `Nodes()` and 7.6 doesn't have `Nodes()`, but `Node()` (see https://dev.opencascade.org/doc/occt-7.5.0/refman/html/class_poly___polygon_on_triangulation.html and https://dev.opencascade.org/doc/occt-7.6.0/refman/html/class_poly___polygon_on_triangulation.html)

Currently I have to catch an Exception when using `Nodes()` to see whether it is 7.6 or 7.5

For the future, would it be possible to expose OCC_VERSION or OCC_VERSION_COMPLETE in a way that one can access it from Python? That would be a much cleaner way to branch the code.

(btw. currently this is necessary, since in CadQuery it seems to be a best practice of users to install `master`, e.g. to leverage newest developments like `Sketch`. So, a soon as master is on 7.6, my code would break)

bernhard-42, 2022-02-20:
Upon recheck it turns out I missed that `Nodes()` actually exists in both "7.5" and "7.6".
However, having OCC_VERSION exposed would nevertheless be interesting to catch other breaking changes for clients of OCP ...

bernhard-42, 2022-02-20:
Finally, I have to distinguish between OCCT 7.5 and 7.6

```python 
   if hasattr(poly, "Node"):
        # much faster in OCCT 7.6, but doesn't exist in 7.5
        indices = [poly.Node(i) for i in range(1, poly.NbNodes() + 1)]
    else:
        indices = [index for index in poly.Nodes()]
```

I have seen that in the past, the `poly.Node(i)` variant is  about 5-10x faster than `[index for index in poly.Nodes()]`. Since this is called very often for large objects, it is an improvement I clearly want to have with the new API in 7.6.

`hasattr(poly, "Node")` does the trick for now, however, I still feel OCC_VERSION would be nice to have.

adam-urbanczyk, 2023-01-12:
Fixed in `7.7.0`. Check `OCP.__version__`

closed 2023-01-12 by adam-urbanczyk