CadQuery/OCP #83: BinTools.Read_s BRepTools.Read_s broken on Mac for io stream

bernhard-42, 2022-03-07:
I use OCP 7.5.3 on an Intel Mac, where OCP got installed via `conda install ... cadquery=master`

```bash
$ ls -l $CONDA_PREFIX/lib/libTKBRep.*
-rwxrwxr-x  2 bernhard  wheel  994716 16 Okt 21:33 /opt/miniconda/envs/jcq3/lib/libTKBRep.7.5.3.dylib
```

Note, this is unrelated to https://github.com/CadQuery/OCP/issues/82 which was an issue on Linux and is resolved

## BinTools

`BinTools.Read_s` crashes with a segmentation fault.

```python
(jcq3) bernhard@Atiyah:~ $ ipython
Python 3.9.7 (default, Sep 16 2021, 08:50:36)
Type 'copyright', 'credits' or 'license' for more information
IPython 8.1.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import io
   ...: from OCP.BinTools import BinTools
   ...: from OCP.TopoDS import TopoDS_Shape
   ...: from OCP.BRepPrimAPI import BRepPrimAPI_MakeBox
   ...: from OCP.gp import gp_Pnt, gp_Dir, gp_Ax2

In [2]: c = gp_Pnt(0,0,0)
   ...: d = gp_Dir(0,0,1)
   ...: shape = BRepPrimAPI_MakeBox(gp_Ax2(c,d), 1, 1, 1).Shape()
   ...: bio = io.BytesIO()
   ...: BinTools.Write_s(shape, bio)
   ...: buffer = bio.getvalue()
   ...: print(buffer[:40])
b'\nOpen CASCADE Topology V3 (c)\nLocations '

In [3]: bio2 = io.BytesIO(buffer)
   ...: shape2 = TopoDS_Shape()
   ...: BinTools.Read_s(shape2, bio2)
   ...: shape2
Segmentation fault: 11
```

The same code works under Linux.

And `BinTools.Read_s` works on the Intel Mac when reading from file:

```python
In [21]: BinTools.Write_s(shape, "shape.brep")
    ...: ! head -n 2 shape.brep
    ...:

Open CASCADE Topology V3 (c)

In [22]: shape2 = TopoDS_Shape()
    ...: BinTools.Read_s(shape2, "shape.brep")
    ...: shape2
Out[22]: <OCP.TopoDS.TopoDS_Shape at 0x7faf6a804430>
```

## BRepTools

Similar issue with `BRepTools`, however no segmentation fault, but a `Standard_Failure`:

```python
In [43]: bio = io.BytesIO()
    ...: BRepTools.Write_s(shape, bio)
    ...: buffer = bio.getvalue()
    ...: print(buffer[:40])
b'\nCASCADE Topology V1, (c) Matra-Datavisi'

In [44]: bio2 = io.BytesIO(buffer)
    ...: shape2 = TopoDS_Shape()
    ...: builder = BRep_Builder()
    ...: BRepTools.Read_s(shape2, bio2, builder)
    ...: shape2
---------------------------------------------------------------------------
Standard_Failure                          Traceback (most recent call last)
Input In [44], in <cell line: 4>()
      2 shape2 = TopoDS_Shape()
      3 builder = BRep_Builder()
----> 4 BRepTools.Read_s(shape2, bio2, builder)
      5 shape2

Standard_Failure: NCollection_IndexedMap::FindKey
```

This code also works under Linux

## M1 Mac

I tested it under OCP 7.5.3, 7.6.0 and 7.6.1, the io stream version doesn't work while the file version does.

whophil, 2022-03-08:
`ocp` is available on conda-forge now, could you please verify if your envs have `ocp` from the `cadquery` channel or `conda-forge` channel?

adam-urbanczyk, 2022-03-08:
@whophil I don't think that 7.6 is available via conda-forge.

@bernhard-42 can you debug this - provide a backtrace and a diff of the stream vs file?

bernhard-42, 2022-03-08:
@whophil Since 7.5.3 is installed with conda, I of course used -c with both channels. So it is the official version
However, all Mac M1 are my builds, but the issue is about 7.5.3 on Intel

bernhard-42, 2022-03-08:
@adam-urbanczyk Happy to help, however I am not so familiar with lldb.

I would run python, load all libs, then attach lldb to python, continue and run the failing code in python. 
Would ‘thread backtrace’ after crash be the right thing then? 
Would it help when OCP lib is in Release mode?
And how to get backtrace in the second case when it doesn’t crash?

bernhard-42, 2022-03-09:
hmm, this is pretty frustrating. I have two environments, one created with conda on March 5th (at least the file timestamps suggest that) and one today. 

`jcq3` on the left was created in the usual cadquery way 

```
conda install -c conda-forge -c cadquery python=3.9 cadquery=master
```

`test-ocp` on the right was created via 

```
conda create -n test-ocp -c conda-forge python=3.9 ipython OCP=7.5.3 occt=7.5.3 vtk=9.0.1
```

[image]

These are really subtle differences and I have no idea why the environments are different. As a user I see the same versions for OCP, 7.5.3.0. Nevertheless, one works the other doesn't:

[image]

I then forced conda to install ocp from cadquery channel:

```bash
conda create -n test-ocp-2 -c conda-forge python=3.9 ipython occt=7.5.3 vtk=9.0.1
conda activate test-ocp-2
conda install -c cadquery ocp
```
And it still works:

[image]

No idea why my first environment is broken ...

whophil, 2022-03-09:
The broken environment `jcq3` has a different VTK _build_ than the other two working environments (but same version). I don't know if this is causing your segfault, though.

I occasionally end up with "bad" environments when installing new packages to or upgrading an existing environment, where problems can usually be fixed by just creating a brand new environment.

A few things which may be helpful:
- `ocp` hit the conda-forge channel on Feb 27 or so.
- You can choose the channel for a specific package with this syntax, which says to get `ocp` from the `cadquery` channel
  ```
  conda create -n cadquery-test -c cadquery cadquery cadquery::ocp
  ```

whophil, 2022-03-15:
@bernhard-42 FWIW, the`cadquery` project is still tested against the `ocp` package which comes from the `cadquery` channel.

adam-urbanczyk, 2023-01-25:
@bernhard-42 is this solved with 7.6 or 7.7?

bernhard-42, 2023-01-25:
I wanted to test it with the latest OCP/cadquery, however, still some issues with installation (see at the end).
I solved it by using

```shell
mamba create -n cadquery python=3.10
mamba activate cadquery
mamba install -c conda-forge -c cadquery OCP
mamba install -c conda-forge -c cadquery casadi nlopt typish nptyping multimethod
env -u CONDA_PREFIX_1 pip install --no-deps git+https://github.com/cadquery/cadquery.git
```
I needed both `env -u` and `--no-deps`, leading to 
```shell
cadquery                  2.2.dev564+g4a79b31          pypi_0    pypi
ocp                       7.6.3.0         py310had5d204_3    conda-forge
vtk                       9.2.2           qt_py310hc6e6bcd_205    conda-forge
```
being installed

Using this environment with OCP 7.6.3.0 works without segfault:

```python
In [1]: import io
   ...: from OCP.BinTools import BinTools
   ...: from OCP.TopoDS import TopoDS_Shape
   ...: from OCP.BRepPrimAPI import BRepPrimAPI_MakeBox
   ...: from OCP.gp import gp_Pnt, gp_Dir, gp_Ax2

In [2]: c = gp_Pnt(0,0,0)
   ...: d = gp_Dir(0,0,1)
   ...: shape = BRepPrimAPI_MakeBox(gp_Ax2(c,d), 1, 1, 1).Shape()

In [3]: bio = io.BytesIO()
   ...: BinTools.Write_s(shape, bio)
   ...: buffer = bio.getvalue()
   ...: print(buffer[:40])
b'\nOpen CASCADE Topology V4, (c) Open Casc'

In [4]: bio2 = io.BytesIO(buffer)
   ...: shape2 = TopoDS_Shape()
   ...: BinTools.Read_s(shape2, bio2)
   ...: shape2
Out[4]: <OCP.TopoDS.TopoDS_Shape at 0x107754870>

In [5]: shape2.ShapeType()
Out[5]: <TopAbs_ShapeEnum.TopAbs_SOLID: 2>
```

So I'd say, this issue is resolved.

= = = = = = = = = = = 8< = = = = = = = = = = = 8< = = = = = = = = = = = 8< = = = = = = = = = = = 8< = = = = = = = = = = = 

Should I raise an issue about this:

Error when using standard CadQuery installation commands with `conda`:

```shell
mamba create -n cadquery python=3.10
mamba activate cadquery
mamba install -c conda-forge -c cadquery cadquery=master

Looking for: ['cadquery=master']

conda-forge/osx-arm64                                       Using cache
conda-forge/noarch                                          Using cache
pkgs/main/noarch                                              No change
pkgs/r/osx-arm64                                              No change
pkgs/main/osx-arm64                                           No change
pkgs/r/noarch                                                 No change
cadquery/osx-arm64                                            No change
cadquery/noarch                                               No change

Pinned packages:
  - python 3.10.*

Could not solve for environment specs
Encountered problems while solving:
  - nothing provides requested cadquery master**

The environment can't be solved, aborting the operation
```

adam-urbanczyk, 2023-01-25:
There is no 7.7 osx-arm64 build yet.

closed 2023-01-25 by adam-urbanczyk

bernhard-42, 2023-01-25:
Got it, I wasn't aware that cadquery is already using OCP 7.7.0, thanks