CadQuery/OCP #85: 7.6 Release?

whophil, 2022-04-12:
Is there work to be done before an official 7.6 release?

whophil, 2022-05-25:
@adam-urbanczyk any thoughts on this? The conda-forge OCP feedstock depends on an official GitHub release on this repo

adam-urbanczyk, 2022-05-25:
My plan was to release this after CQ 2.2 release + some additional time for testing.

adam-urbanczyk, 2022-11-23:
@whophil plans do change, so there you go: https://github.com/CadQuery/OCP/releases/tag/7.6.3.0

closed 2022-11-23 by adam-urbanczyk

whophil, 2022-11-23:
Thanks @adam-urbanczyk ! If you could provide the binding code as release artifacts as you did with 7.5.3 (e.g., [OCP_src_ubuntu-18.04.zip](https://github.com/CadQuery/OCP/releases/download/7.5.3.0/OCP_src_ubuntu-18.04.zip), I can start the conda-forge build.

adam-urbanczyk, 2022-11-23:
Done