CadQuery/OCP #87: segfault on loft

crides, 2022-05-25:
I don't have an minimal example yet, but this is what caused the issue:

```python
import cadquery as cq, math

m2_hole = 2
bar_len, bar_ang, bar_w = 85, 35, 4
nub_dist, nub_hole_off, nub_len, nub_w, nub_h, nub_thick = 7, 2.5, 14, 6.25, 6.3, 2
board_dist, board_w, board_thick = 17, 6, 2.5
board_bar_off = 2 + bar_w / 2

trans = cq.Matrix()
trans.rotateX(math.radians(10))
trans.rotateY(math.radians(-5))
trans.rotateZ(math.radians(15))
poses = [(-board_dist / 2, board_dist / 2), (board_dist / 2, -board_dist / 2)]
board_side = (cq.Workplane().sketch()
    .push(poses).circle(board_w / 2).wires().hull()
    .push(poses).circle(m2_hole / 2, "s")
    .finalize()
    .extrude(-board_thick)
    .transformed((0, 0, 45)).rect(board_w, nub_len).extrude(0.1)
    .val().transformShape(trans))
board_side = cq.Workplane().add(board_side)

bar_top = 5 + board_bar_off, -bar_len * math.cos(math.radians(bar_ang)) + board_bar_off, bar_len * math.sin(math.radians(bar_ang))
nub_side = (cq.Workplane().transformed(offset=(bar_top[0], bar_top[1], bar_top[2] + nub_thick))
    .box(nub_w, nub_len, nub_h, (True, True, False))
    .faces(">Z").shell(nub_thick)
    .faces("<X").workplane()
    .pushPoints([(-nub_dist / 2, nub_h - nub_hole_off), (nub_dist / 2, nub_h - nub_hole_off)])
    .hole(m2_hole))
bar = (cq.Workplane().add(board_side.faces("<<Z[2]").wires().vals()).toPending()
    .add(nub_side.faces("<Z").wires().vals()).toPending().loft())
```

Right before the `bar` creation: 
[image]

Right before the `.loft` call:
[image]

Core & backtrace: 

[cq-core.zst](http://0x0.st/oakJ.zst)

```
#0  0x00007f82b1bb0796 in ?? () from /home/steven/.local/lib/python3.10/site-packages/ocpvtk.libs/libTKTopAlgo-a16b7be4.so.7.5.3
#1  0x00007f82b1bb139f in BRepLib::EncodeRegularity(TopoDS_Shape const&, double) () from /home/steven/.local/lib/python3.10/site-packages/ocpvtk.libs/libTKTopAlgo-a16b7be4.so.7.5.3
#2  0x00007f829ba33339 in BRepOffsetAPI_ThruSections::Build() () from /home/steven/.local/lib/python3.10/site-packages/ocpvtk.libs/libTKOffset-b9ab1e2e.so.7.5.3
#3  0x00007f829d5320df in pybind11::cpp_function::initialize<pybind11::cpp_function::initialize<void, BRepOffsetAPI_ThruSections, , pybind11::name, pybind11::is_method, pybind11::sibling, char [5]>(void (BRepOffsetAPI_ThruSections::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [5])::{lambda(BRepOffsetAPI_ThruSections*)#1}, void, BRepOffsetAPI_ThruSections*, pybind11::name, pybind11::is_method, pybind11::sibling, char [5]>(pybind11::cpp_function::initialize<void, BRepOffsetAPI_ThruSections, , pybind11::name, pybind11::is_method, pybind11::sibling, char [5]>(void (BRepOffsetAPI_ThruSections::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [5])::{lambda(BRepOffsetAPI_ThruSections*)#1}&&, void (*)(BRepOffsetAPI_ThruSections*), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [5])::{lambda(pybind11::detail::function_call&)#3}::_FUN(pybind11::detail::function_call&) () from /home/steven/.local/lib/python3.10/site-packages/OCP.cpython-310-x86_64-linux-gnu.so
#4  0x00007f829c992b3f in pybind11::cpp_function::dispatcher(_object*, _object*, _object*) () from /home/steven/.local/lib/python3.10/site-packages/OCP.cpython-310-x86_64-linux-gnu.so
#5  0x00007f82ba2dc2d2 in ?? () from /usr/lib/libpython3.10.so.1.0
#6  0x00007f82ba2d67cb in _PyObject_MakeTpCall () from /usr/lib/libpython3.10.so.1.0
#7  0x00007f82ba2e823b in ?? () from /usr/lib/libpython3.10.so.1.0
#8  0x00007f82ba2d14f1 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.10.so.1.0
#9  0x00007f82ba2e7fc4 in ?? () from /usr/lib/libpython3.10.so.1.0
#10 0x00007f82ba2d14f1 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.10.so.1.0
#11 0x00007f82ba2dc73c in _PyFunction_Vectorcall () from /usr/lib/libpython3.10.so.1.0
#12 0x00007f82ba2cca9b in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.10.so.1.0
#13 0x00007f82ba2cb392 in ?? () from /usr/lib/libpython3.10.so.1.0
#14 0x00007f82ba383c14 in PyEval_EvalCode () from /usr/lib/libpython3.10.so.1.0
#15 0x00007f82ba38b671 in ?? () from /usr/lib/libpython3.10.so.1.0
#16 0x00007f82ba2dc940 in ?? () from /usr/lib/libpython3.10.so.1.0
#17 0x00007f82ba2cc769 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.10.so.1.0
#18 0x00007f82ba2e7fc4 in ?? () from /usr/lib/libpython3.10.so.1.0
#19 0x00007f82ba2d14f1 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.10.so.1.0
#20 0x00007f82ba2e81cd in ?? () from /usr/lib/libpython3.10.so.1.0
#21 0x00007f82b514fd31 in ?? () from /usr/lib/python3.10/site-packages/PyQt5/QtCore.abi3.so
#22 0x00007f82b51556a0 in ?? () from /usr/lib/python3.10/site-packages/PyQt5/QtCore.abi3.so
#23 0x00007f82b5133015 in ?? () from /usr/lib/python3.10/site-packages/PyQt5/QtCore.abi3.so
#24 0x00007f82b8373224 in ?? () from /usr/lib/libQt5Core.so.5
#25 0x00007f82b514eac0 in ?? () from /usr/lib/python3.10/site-packages/PyQt5/QtCore.abi3.so
#26 0x00007f82ba2c77b9 in ?? () from /usr/lib/libpython3.10.so.1.0
#27 0x00007f82ba2cca9b in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.10.so.1.0
#28 0x00007f82ba2e81cd in ?? () from /usr/lib/libpython3.10.so.1.0
#29 0x00007f82b514fd31 in ?? () from /usr/lib/python3.10/site-packages/PyQt5/QtCore.abi3.so
#30 0x00007f82b5150320 in ?? () from /usr/lib/python3.10/site-packages/PyQt5/QtCore.abi3.so
#31 0x00007f82b8373224 in ?? () from /usr/lib/libQt5Core.so.5
#32 0x00007f82b837569f in QTimer::timeout(QTimer::QPrivateSignal) () from /usr/lib/libQt5Core.so.5
#33 0x00007f82b5033a74 in ?? () from /usr/lib/python3.10/site-packages/PyQt5/QtCore.abi3.so
#34 0x00007f82b8366766 in QObject::event(QEvent*) () from /usr/lib/libQt5Core.so.5
#35 0x00007f82b502d7e4 in ?? () from /usr/lib/python3.10/site-packages/PyQt5/QtCore.abi3.so
#36 0x00007f82b8db81c6 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
#37 0x00007f82b96b2507 in ?? () from /usr/lib/python3.10/site-packages/PyQt5/QtWidgets.abi3.so
#38 0x00007f82b83425aa in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/libQt5Core.so.5
#39 0x00007f82b838ddd5 in QTimerInfoList::activateTimers() () from /usr/lib/libQt5Core.so.5
#40 0x00007f82b838e3ba in ?? () from /usr/lib/libQt5Core.so.5
#41 0x00007f82b747bee3 in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#42 0x00007f82b74d20f9 in ?? () from /usr/lib/libglib-2.0.so.0
#43 0x00007f82b7479455 in g_main_context_iteration () from /usr/lib/libglib-2.0.so.0
#44 0x00007f82b838e57a in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Core.so.5
#45 0x00007f82b833a88b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Core.so.5
#46 0x00007f82b8345fd7 in QCoreApplication::exec() () from /usr/lib/libQt5Core.so.5
#47 0x00007f82b96b4d80 in ?? () from /usr/lib/python3.10/site-packages/PyQt5/QtWidgets.abi3.so
#48 0x00007f82ba2dc2f8 in ?? () from /usr/lib/libpython3.10.so.1.0
#49 0x00007f82ba2d67cb in _PyObject_MakeTpCall () from /usr/lib/libpython3.10.so.1.0
#50 0x00007f82ba2d1e28 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.10.so.1.0
#51 0x00007f82ba2dc73c in _PyFunction_Vectorcall () from /usr/lib/libpython3.10.so.1.0
#52 0x00007f82ba2cc769 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.10.so.1.0
#53 0x00007f82ba2cb392 in ?? () from /usr/lib/libpython3.10.so.1.0
#54 0x00007f82ba383c14 in PyEval_EvalCode () from /usr/lib/libpython3.10.so.1.0
#55 0x00007f82ba394b54 in ?? () from /usr/lib/libpython3.10.so.1.0
#56 0x00007f82ba39055b in ?? () from /usr/lib/libpython3.10.so.1.0
#57 0x00007f82ba229fca in ?? () from /usr/lib/libpython3.10.so.1.0
#58 0x00007f82ba229c4a in _PyRun_SimpleFileObject () from /usr/lib/libpython3.10.so.1.0
#59 0x00007f82ba22a5df in _PyRun_AnyFileObject () from /usr/lib/libpython3.10.so.1.0
#60 0x00007f82ba3a130b in Py_RunMain () from /usr/lib/libpython3.10.so.1.0
#61 0x00007f82ba37469d in Py_BytesMain () from /usr/lib/libpython3.10.so.1.0
#62 0x00007f82b9faf310 in __libc_start_call_main () from /usr/lib/libc.so.6
#63 0x00007f82b9faf3c1 in __libc_start_main_impl () from /usr/lib/libc.so.6
#64 0x000055950136404e in _start ()
```

crides, 2022-07-10:
Somehow I have forgotten to revisit this, sorry.

It seems that the issue occurs if I put another `toPending()` before adding the second wire. Not needing a second call to that seems.. interesting, though adding it anyway shouldn't cause an outright segfault either?

adam-urbanczyk, 2026-09-06:
Your examples do not segfault anymore.

closed 2026-09-06 by adam-urbanczyk