CadQuery/OCP #104: IVtkTools_ShapeDataSource ititialization

BlankuApp, 2023-02-25:
I would like to present an OCCT shape in VTK viewer using your package.

According to the OCCT documentation, the C++ code looks like the following:
```C++
TopoDS_Shape aShape;
IVtkOCC_Shape::Handle aShapeImpl = new IVtkOCC_Shape(aShape);

vtkSmartPointer<IVtkTools_ShapeDataSource> aDS = vtkSmartPointer<IVtkTools_ShapeDataSource>::New();
aDS->SetShape(aShapeImpl);

vtkSmartPointer<vtkPolyDataMapper> Mapper = vtkSmartPointer<vtkPolyDataMapper>::New();
Mapper->SetInputConnection(aDS->GetOutputPort());
vtkSmartPointer<vtkActor> Actor = vtkSmartPointer<vtkActor>::New();
Actor->SetMapper(Mapper);
```

I'm wondering how to make `aDS` since in python `IVtkTools_ShapeDataSource` has no `__init__` function.
What is the equivalent of the above code in python?

Sorry for my absolute beginner question.

whophil, 2023-02-26:
```
>>> OCP.IVtkTools.IVtkTools_ShapeDataSource.New_s()
<OCP.IVtkTools.IVtkTools_ShapeDataSource object at 0x1427f1e70>
```

BlankuApp, 2023-02-26:
Thank you.
It works.
How about the next step: `aDS->GetOutputPort()`.
In C++, `IVtkTools_ShapeDataSource` has no `GetOutputPort()` originally. `GetOutputPort()` method is inherited from `vtkAlgorithm` class.
However, in OCP, `IVtkTools_ShapeDataSource` has no parent classes.
[https://github.com/CadQuery/OCP-stubs/blob/e838ff400d5ee2f4a0579d2a713b19311855288f/OCP-stubs/IVtkTools/__init__.pyi#L83](https://github.com/CadQuery/OCP-stubs/blob/e838ff400d5ee2f4a0579d2a713b19311855288f/OCP-stubs/IVtkTools/__init__.pyi#L83)

adam-urbanczyk, 2023-02-26:
This would likely require writing some custom converter in OCP. So for now, I'm afraid you can't do this.

whophil, 2023-02-26:
Hi @BlankuApp , there are other ways to get OpenCascade shapes into VTK viewers without using that class. As an example, see this method which takes in an OpenCascade shape, meshes it, and returns a vtkPolyData.

BlankuApp, 2023-02-27:
Hey @whophil, thanks for suggesting a solution!
It looks like you might have forgotten to include the method you had in mind.
Could you please share it again?
Thanks!

whophil, 2023-02-27:
@BlankuApp https://github.com/CadQuery/cadquery/blob/4c2a0383495862c71aaf90da2240dea9d40f99bf/cadquery/occ_impl/shapes.py#L1287

BlankuApp, 2023-03-03:
@whophil It worked very well.
thank you so much.

closed 2023-03-03 by BlankuApp