CadQuery/OCP #105: build failure: invalid ''static_cast'

GenericNerdyUsername, 2023-03-02:
I'm trying to build the 7.7.0.0 release on nixos using the
```
pywrap [includes and libclang] all ocp.toml
cmake
cd build
cmake --build .
```
method, but get an error during the last command bit.

versions:
```
python: 3.10.10
vtk: 9.2.5
rapidjson: 1.1.0
libgl: 1.6.0
libclang: 11.1.0
```

full error:
```
/build/source/OCP/ChFi3d.cpp:125:155: error: invalid ‘static_cast’ from type ‘Standard_Boolean (*)(const TopoDS_Edge&, const TopoDS_Face&, const TopoDS_Face&, GeomAbs_Shape)’ {aka ‘bool (*)(const TopoDS_Edge&, const TopoDS_>
  125 |                     (Standard_Boolean (*)( const TopoDS_Edge & ,  const TopoDS_Face & ,  const TopoDS_Face & ,  Standard_Real ,  const GeomAbs_Shape  ) ) static_cast<Standard_Boolean (*)( const TopoDS_Edge & ,  cons>
      |                                                                                                                                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~>

```

adam-urbanczyk, 2023-03-06:
That does not occur in CI, nor have I seen it locally. You'll have to figure it out on your own.

GenericNerdyUsername, 2023-03-09:
Im getting the same result in a ubuntu vm, what distro/software versions are you using?

GenericNerdyUsername, 2023-03-09:
Also how is the CI working when the cmake version for ubuntu 18 is too old for this project?

closed 2023-03-21 by GenericNerdyUsername

GenericNerdyUsername, 2023-08-23:
Did you regenerate symbols? On nix doing so looks like this:
```
ln -s ${opencascade-occt} ./lib_linux
python3 -m dump_symbols .
cp symbols_mangled_linux.dat pywrap/symbols_mangled_linux.dat
rm {.,pywrap}/symbols_mangled_{win,mac}.dat
```

akoen, 2023-08-25:
I solved this by deleting a spurious definition in the bindings discovered here: https://github.com/CadQuery/OCP/issues/115#issuecomment-1637894716, although I should try your solution.