CadQuery/OCP #106: No hash found for file 'LICENSES_bundled.txt'

sethfischer, 2023-03-05:
cadquery-ocp 7.7.0a0 has a missing or invalid hash for `LICENSES_bundled.txt`.

This can be demonstrated with the [wheel](https://pypi.org/project/wheel/) command line tool:

```
$ wheel unpack ../cadquery_ocp-7.6.3-cp310-cp310-manylinux_2_31_x86_64.whl 
Unpacking to: cadquery_ocp-7.6.3...No hash found for file 'LICENSES_bundled.txt'
```

And it also presents as an error when attempting to install with Poetry 1.4.0:

```
$ poetry add --allow-prereleases cadquery-ocp
Using version ^7.7.0a0 for cadquery-ocp

Updating dependencies
Resolving dependencies... (0.1s)

Writing lock file

Package operations: 1 install, 0 updates, 0 removals

  • Installing cadquery-ocp (7.7.0a0): Failed

  AssertionError

  In cadquery_ocp-7.7.0a0-cp39-cp39-manylinux_2_31_x86_64.whl, LICENSES_bundled.txt is not mentioned in RECORD

  at ~/.local/share/pypoetry/venv/lib/python3.8/site-packages/installer/sources.py:158 in get_contents
      154│             if item.filename[-1:] == "/":  # looks like a directory
      155│                 continue
      156│ 
      157│             record = record_mapping.pop(item.filename, None)
    → 158│             assert record is not None, "In {}, {} is not mentioned in RECORD".format(
      159│                 self._zipfile.filename,
      160│                 item.filename,
      161│             )  # should not happen for valid wheels
      162│ 
```

Poetry 1.4.0 introduces a "modern installer" which verifies hashes. The modern installer can be disabled with:

```
poetry config --local installer.modern-installation false
```

which writes to `poetry.toml` in the project directory:

```toml
[installer]
modern-installation = false
```

Pip does not check hashes in wheel RECORD contents during installation, see pypa/pip#4705.

Poetry is checking hashes in compliance with [PEP 0427](https://peps.python.org/pep-0427/#the-dist-info-directory):

> During extraction, wheel installers verify all the hashes in RECORD against the file contents. Apart from RECORD and its signatures, installation will fail if any file in the archive is not both mentioned and correctly hashed in RECORD.

adam-urbanczyk, 2023-03-05:
This repo does not provide any wheels. I think ocp build system does

closed 2023-03-05 by adam-urbanczyk

sethfischer, 2023-03-05:
See CadQuery/ocp-build-system#12