CadQuery/OCP #108: How to install cadquery-ocp on Linux arm64 ?

fungiboletus, 2023-04-18:
Hi,

I'm trying to install cadquery-ocp from PIP. On amd64 it works very well, but it doesn't seem to be available on arm64:

> # pip install cadquery-ocp
> ERROR: Could not find a version that satisfies the requirement cadquery-ocp (from versions: none)
> ERROR: No matching distribution found for cadquery-ocp

Do you have any plan to support linux on arm64 ? Thanks.

jmwright, 2023-04-18:
@fungiboletus The cadquery-ocp wheel is maintained separately from this repo.

https://github.com/CadQuery/ocp-build-system

There is an issue for building for other platforms there, but the main bottleneck right now is that GitHub hasn't released the GitHub-hosted Linux ARM64 Runner yet.

The Conda installs should now work with Linux ARM64, so you could use that.

fungiboletus, 2023-04-18:
Oh sorry about the wrong repository.

A workaround could be about using docker buildx with qemu. It works relatively well in github actions, though you have to build things inside a container and not directly in the action.

I'm trying to avoid conda whenever I can, but I will consider the alternative. Thanks for the help!

closed 2023-04-18 by fungiboletus