CadQuery/OCP #119: Installation Order - OCP with VTK and PyVista

akaszynski, 2023-08-31:
Ran into an issue recently where tessellation to VTK wasn't working:

```py
import cadquery as cq
import OCP

shape = cq.Workplane().box(1, 1, 1).faces(">Z").hole(0.7).edges().fillet(0.03).val()

vs = OCP.IVtkOCC.IVtkOCC_Shape(shape.wrapped)
sd = OCP.IVtkVTK.IVtkVTK_ShapeData()
sm = OCP.IVtkOCC.IVtkOCC_ShapeMesher()
sm.Build(vs, sd)

pdata = sd.getVtkPolyData()
print(pdata)
```
Where `pdata` was output as `None`. Same issue using the more concise:
```py
import cadquery as cq

shape = cq.Workplane().box(1, 1, 1).faces(">Z").hole(0.7).edges().fillet(0.03).val()
pdata = shape.toVtkPolyData()
```

After a bit of debugging, turns out the issue will occur if you install `vtk` with:

`pip install vtk==9.2.5`.

Since OCP installs the `vtkmodules` directory, this is silently overwritten when installing `vtk`.

This is an issue on my end since I'm using `cadquery` and I'd like to also plot the resulting CAD using `pyvista`, which depends on the `vtk` package. The only way I can think to work around this is to install `pyvista` with no dependencies. If anyone wishes to plot cadquery shapes using `pyvista`, here's a minimum working example:

```
pip install cadquery
```
Followed by installing pyvista without VTK
```
pip install pyvista --no-deps
pip install scooby matplotlib pooch
```

This works and the plots look great!

```py
import pyvista as pv
import cadquery as cq

shape = cq.Workplane().box(1, 1, 1).faces(">Z").hole(0.7).edges().fillet(0.03).val()
pdata = pv.wrap(shape.toVtkPolyData())
pdata.plot()
```

[image]

---
Ask for the maintainers: Is there any way for `OCP` to use a pre-compiled version of VTK? Is there any other way of working around this issue?

whophil, 2023-08-31:
@akaszynski - are you installing OCP via conda?

akaszynski, 2023-08-31:
This is all just from good old `pip`.

whophil, 2023-08-31:
@akaszynski I see. This may not be an adequate solution for you, but this should "just work" if you use `conda`. I certainly understand the many reasons you might not want to use conda.

This seems to be mainly an issue with the wheels, whose builds are handled in a separate project - https://github.com/CadQuery/ocp-build-system/.

Cross-referencing your issue there (https://github.com/CadQuery/ocp-build-system/issues/14)

akaszynski, 2023-08-31:
Yep, I recall opening that there, but reposted here so that others could know there's a solution.

Sadly, conda isn't an option for me at this point due to issues repackaging with `pyinstaller`. One solution I thought of was creating a simple fork of `pyvista` called `pyvista-no-vtk` that installs `pyvista` without installing vtk. PyVista (for my use case) still works out of the box since we use only `vtkmodules`.

adam-urbanczyk, 2023-08-31:
> Yep, I recall opening that there, but reposted here so that others could know there's a solution.

Thanks for sharing, noted. Still, the advised installation method is conda.

> Sadly, conda isn't an option for me at this point due to issues repackaging with `pyinstaller`. One solution I thought of was creating a simple fork of `pyvista` called `pyvista-no-vtk` that [...]

There should not be any fundamental issue pyinstaller and conda. BTW: recently for CQ-editor, we switched to constructor. Maybe something to try?

adam-urbanczyk, 2026-09-06:
AFAICT fixed downstream (https://github.com/CadQuery/ocp-build-system/issues/14)

closed 2026-09-06 by adam-urbanczyk