CadQuery/OCP #121: How to update math_Matrix

mnesarco, 2023-09-08:
Hi Friends,
I need to use `math_Matrix` in an algorithm in python, but the c++ object is updated using non const references returned by `auto operator(int, int) -> double&`.

In C++:
```cpp
void something(math_Matrix& m) {
  m(1,5) = 0.5;
}
```

In python this has no sense:
```python
from OCP.math import math_Matrix

def something(m: math_Matrix):
  m(1,5) = 0.5
```
Same problem with `math_Vector`

I appreciate any advice.

Thanks.

whophil, 2023-09-08:
@mnesarco are you just trying to set the value? Can you use `m.Set(1, 1, 5, 5, 0.5)`? (https://dev.opencascade.org/doc/refman/html/classmath___matrix.html#aa997d5e92e66768d98697f2b658b46fa)

adam-urbanczyk, 2023-09-08:
Not ideal, but try `m.Set(i,i,j,j, OCP.math.math_Matrix(1,1,1,1, value))`.

closed 2024-10-30 by mnesarco