CadQuery/OCP #132: pywrap: AttributeError: 'NoneType' object has no attribute 'name'

l29ah, 2023-12-11:
I'm packaging CadQuery for Gentoo, so i'm building OCP from sources. Running `pywrap --verbose --njobs 9 --libclang /usr/lib/llvm/17/lib64/libclang.so --include /usr/lib/clang/17/include --include /usr/include/vtk-9.2 all ocp.toml` on the `7.7.2.0` tag results in
```
  0% 0/314 [00:00<?, ?it/s][D 231211 14:02:49 module:41] Processing headers
[D 231211 14:02:49 module:44] ./opencascade/Standard_UUID.hxx
[D 231211 14:02:49 module:41] Processing headers
[D 231211 14:02:49 module:44] ./opencascade/math_GaussSetIntegration.hxx
[D 231211 14:02:49 module:41] Processing headers
[D 231211 14:02:49 module:44] ./opencascade/gp_Circ2d.hxx
[D 231211 14:02:49 module:41] Processing headers
[D 231211 14:02:49 module:44] ./opencascade/Quantity_NameOfColor.hxx
[D 231211 14:02:49 module:41] Processing headers
[D 231211 14:02:49 module:44] ./opencascade/TColgp_HSequenceOfXY.hxx
[D 231211 14:02:49 module:41] Processing headers
[D 231211 14:02:49 module:44] ./opencascade/OSD_ErrorList.hxx
[D 231211 14:02:49 module:41] Processing headers
[D 231211 14:02:49 module:44] ./opencascade/NCollection_BaseAllocator.hxx
[D 231211 14:02:49 module:41] Processing headers
[D 231211 14:02:49 module:44] ./opencascade/BVH_Sorter.hxx
[D 231211 14:02:49 module:41] Processing headers
[D 231211 14:02:49 module:44] ./opencascade/Storage_HArrayOfCallBack.hxx
[D 231211 14:02:50 module:41] Processing headers
[D 231211 14:02:50 module:44] ./opencascade/FSD_Base64.hxx
loky.process_executor._RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/loky/process_executor.py", line 463, in _process_worker
    r = call_item()
        ^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/loky/process_executor.py", line 291, in __call__
    return self.fn(*self.args, **self.kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/joblib/parallel.py", line 589, in __call__
    return [func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/joblib/parallel.py", line 589, in <listcomp>
    return [func(*args, **kwargs)
            ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bindgen/__init__.py", line 246, in _process_module
    return ModuleInfo(name,path,files,module_names,settings)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bindgen/module.py", line 45, in __init__
    self.headers.append(process_header(p,prefix,settings,name))
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bindgen/header.py", line 752, in process_header
    hi.parse(path,input_folder,settings,module_name)
  File "/usr/lib/python3.11/site-packages/bindgen/header.py", line 707, in parse
    self.dependencies = [el.location.file.name for el in tr_unit.get_includes()]
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bindgen/header.py", line 707, in <listcomp>
    self.dependencies = [el.location.file.name for el in tr_unit.get_includes()]
                         ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'name'
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib/python-exec/python3.11/pywrap", line 33, in <module>
    sys.exit(load_entry_point('pywrap==9999', 'console_scripts', 'pywrap')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bindgen/__main__.py", line 101, in all
    ctx.invoke(parse, configuration=configuration, output=tmp_parsed, platform=platform)
  File "/usr/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/click/decorators.py", line 45, in new_func
    return f(get_current_context().obj, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bindgen/__main__.py", line 40, in parse
    result = parse_modules(obj.verbose,obj.njobs,settings,module_mapping,module_settings,platform)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bindgen/__init__.py", line 248, in parse_modules
    modules = Parallel(prefer='processes',n_jobs=n_jobs)\
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/joblib/parallel.py", line 1952, in __call__
    return output if self.return_generator else list(output)
                                                ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/joblib/parallel.py", line 1595, in _get_outputs
    yield from self._retrieve()
  File "/usr/lib/python3.11/site-packages/joblib/parallel.py", line 1699, in _retrieve
    self._raise_error_fast()
  File "/usr/lib/python3.11/site-packages/joblib/parallel.py", line 1734, in _raise_error_fast
    error_job.get_result(self.timeout)
  File "/usr/lib/python3.11/site-packages/joblib/parallel.py", line 736, in get_result
    return self._return_or_raise()
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/joblib/parallel.py", line 754, in _return_or_raise
    raise self._result
AttributeError: 'NoneType' object has no attribute 'name'
```
What's wrong with it?

adam-urbanczyk, 2023-12-12:
IDK, sounds like some include has no file name associated with it according to libclang.

If you want to build from sources, please use the released [sources](https://github.com/CadQuery/OCP/releases/tag/7.7.2.0) and not generate them yourself.

l29ah, 2023-12-12:
> If you want to build from sources, please use the released sources and not generate them yourself.

That wouldn't be "sources" then.

adam-urbanczyk, 2023-12-24:
Those are the official released sources. There is nothing you will gain by running the generator unless you want modify the `ocp.toml` configuration file.

l29ah, 2024-04-30:
Could you clarify how to use these? I couldn't find it in the repo documentation, and conda is not a great option for packaging in Gentoo, as it's not packaged itself, and would likely present issues with dependency bundling, among other things.

adam-urbanczyk, 2024-05-09:
Just download the `*src` files from e.g. here: https://github.com/CadQuery/OCP/releases/download/7.7.2.0/OCP_src_stubs_ubuntu-20.04.zip . You'll likely hit this issue though: https://github.com/conda-forge/occt-feedstock/issues/108 , so I'd wait for #148 being merged and released as `7.7.2.1`.

The zip comes with `CMakeLists.txt` so just use cmake to compile and that is it.