CadQuery/OCP #133: PyInstaller + CadQuery issue

kutraj87, 2023-12-20:
Hi 

Firstly, thanks a lot for the wonderful library, CQ is absolutely a dream to use!

I'm writing about a packaging query using PyInstaller (Windows 11, 64-bit). I've tried looking on the pyinstaller man pages to see if any additional options will work but haven't found much yet, so am posting here hoping more experienced folks can help.

When I combine my application (currently uses CQ from PIP, with OpenGL/GLFW and imgui) using PyInstaller, I first got the following error:

File "cadquery\occ_impl\geom.py", line 5, in <module>
ModuleNotFoundError: No module named 'OCP'
[10140] Failed to execute script 'main' due to unhandled exception!

Based on this, I installed OCP (via Conda) and tried again after inserting a dummy import call to the OCP library. Now I get a VTK error. 

 File "cadquery\occ_impl\shapes.py", line 19, in <module>
ImportError: Failed to load vtkCommonDataModel: No module named vtkmodules.vtkCommonMath
[10452] Failed to execute script 'main' due to unhandled exception!

Now I'm at the precipice of having to install VTK which I'd like to avoid for 2 reasons - my application is written in native OpenGL including all the shaders etc. and secondly, from what I understand, I'd need to compile VTK from source on Windows (which I'm not very familiar with unlike Linux). 

Is there an alternative route that I can take for packaging Cadquery on Windows? Are there any folks out there who have tried this previously?

Thanks for your time! Any pointers will be much appreciated.

Best wishes,
KV

adam-urbanczyk, 2023-12-20:
Take a look here https://github.com/CadQuery/CQ-editor/blob/master/pyinstaller.spec , but this is not used anymore and likley not working. For CQ-editor conda-constructor is currently in use: https://github.com/CadQuery/CQ-editor/blob/master/conda/construct.yaml

trunglebao, 2024-01-29:
I used auto-py-to-exe (because i'm lazy) and this provide an easy to use GUI to pyinstaller. Need to collect-all these 3 packages for cadquery to work on exe: casadi; cadquery-ocp; vtkmodules.all

The pyinstaller command should looks like this:
pyinstaller --noconfirm --onefile --windowed --collect-data "casadi" --collect-data "cadquery-ocp" --collect-data "vtkmodules.all"

[image]

closed 2024-01-30 by adam-urbanczyk