CadQuery/OCP #138: Released 7.7.2.0 stub source contains binding for non-existent method

danderson, 2024-02-10:
I'm trying to use  OCP_src_stubs_ubuntu-20.04.zip from the 7.7.2.0 release to compile OCP. Compilation fails with:

```
/build/OCP/OCP/ChFi3d.cpp: In function 'void register_ChFi3d(pybind11::module&)':
/build/OCP/OCP/ChFi3d.cpp:125:323: error: invalid static_cast from type 'Standard_Boolean (*)(const TopoDS_Edge&, const TopoDS_Face&, const TopoDS_Face&, GeomAbs_Shape)' {aka 'bool (*)(const TopoDS_Edge&, const TopoDS_Face&, const TopoDS_Face&, GeomAbs_Shape)'} to type 'Standard_Boolean (*)(const TopoDS_Edge&, const TopoDS_Face&, const TopoDS_Face&, Standard_Real, GeomAbs_Shape)' {aka 'bool (*)(const TopoDS_Edge&, const TopoDS_Face&, const TopoDS_Face&, double, GeomAbs_Shape)'}
                     (Standard_Boolean (*)( const TopoDS_Edge & ,  const TopoDS_Face & ,  const TopoDS_Face & ,  Standard_Real ,  const GeomAbs_Shape  ) ) static_cast<Standard_Boolean (*)( const TopoDS_Edge & ,  const TopoDS_Face & ,  const TopoDS_Face & ,  Standard_Real ,  const GeomAbs_Shape  ) >(&ChFi3d::IsTangentFaces),
```

The compile error is hard to read, but the core of the issue is that `register_ChFi3d` is trying to create bindings for 2 variants of `ChFi3d::IsTangentFaces`: one with 4 arguments, and one with 5 arguments:

```
        .def_static("IsTangentFaces_s",
                    (Standard_Boolean (*)( const TopoDS_Edge & ,  const TopoDS_Face & ,  const TopoDS_Face & ,  const GeomAbs_Shape  ) ) static_cast<Standard_Boolean (*)( const TopoDS_Edge & ,  const TopoDS_Face & ,  const TopoDS_Face & ,  const GeomAbs_Shape  ) >(&ChFi3d::IsTangentFaces),
                    R"#(Returns true if theEdge between theFace1 and theFace2 is tangent)#"  , py::arg("theEdge"),  py::arg("theFace1"),  py::arg("theFace2"),  py::arg("Order")=static_cast<const GeomAbs_Shape>(GeomAbs_G1)
          )
        .def_static("IsTangentFaces_s",
                    (Standard_Boolean (*)( const TopoDS_Edge & ,  const TopoDS_Face & ,  const TopoDS_Face & ,  Standard_Real ,  const GeomAbs_Shape  ) ) static_cast<Standard_Boolean (*)( const TopoDS_Edge & ,  const TopoDS_Face & ,  const TopoDS_Face & ,  Standard_Real ,  const GeomAbs_Shape  ) >(&ChFi3d::IsTangentFaces),
                    R"#(None)#"  , py::arg("theEdge"),  py::arg("theFace1"),  py::arg("theFace2"),  py::arg("G1Tol"),  py::arg("Order")=static_cast<const GeomAbs_Shape>(GeomAbs_G1)
          )
```

However, in the OpenCASCADE 7.7.2 header files, there is only one form of `ChFi3d::IsTangentFaces`, the 4-argument version:

```
  //! Returns true if theEdge between theFace1 and theFace2 is tangent
  Standard_EXPORT static Standard_Boolean IsTangentFaces (const TopoDS_Edge& theEdge,
                                                          const TopoDS_Face& theFace1,
                                                          const TopoDS_Face& theFace2,
                                                          const GeomAbs_Shape Order = GeomAbs_G1);
```

This is the case in both the upstream OpenCASCADE 7.7.2 source code, and in the copy of the header in this repo at `opencascade/ChFi3d.hxx`.

I scanned all releases of OpenCASCADE from 7.5.0 to 7.8.0, and none of them have a 5-argument IsTangentFaces. And yet, clearly, the published stubs were generated in an environment where OpenCASCADE _does_ have such a definition.

Is it possible the released stubs were generated against a development copy of OpenCASCADE, rather than a clean 7.7.2? I'm struggling to find any other explanation for the discrepancy.

This is also the root cause of #115 and #105, which were worked around by manually deleting the spurious definition from the release stubs.

adam-urbanczyk, 2024-04-01:
See: https://github.com/conda-forge/occt-feedstock/issues/108

danderson, 2024-04-01:
Thanks for the reference, and sorry for my apparently triggering quite a lot of pain :grimacing: . Now that I know about occt-feedstock I'll make sure to track that in order to match your build environment. Thank you!

adam-urbanczyk, 2026-09-06:
Blobfish patch is not applied anymore,

closed 2026-09-06 by adam-urbanczyk