CadQuery/OCP #139: build failure with gcc: NCollection_tmpl.hxx:838:60: error: expected unqualified-id before * token

efferre79, 2024-03-07:
I am using the CMakeLists.txt file from the repository, I have successfully created the bindings using clang 17, pywrap from master branch and pybind 2.11.1 but later I get many errors when starting to build. I am here reporting just the first lines from the build failure:

```
ninja -v -j8 -l16
[1/630] /usr/bin/x86_64-pc-linux-gnu-g++ -DHAVE_FFMPEG -DHAVE_FREETYPE -DHAVE_Inspector -DHAVE_OPENGL_EXT -DHAVE_QT -DHAVE_RAPIDJSON -DHAVE_TBB -DHAVE_TK -DH
AVE_VTK -DHAVE_XLIB -DOCC_CONVERT_SIGNALS -DOCP_EXPORTS -DVTK_OPENGL2_BACKEND -Dkiss_fft_scalar=double -I/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/
work/OCP-7.7.2.0/OCP -isystem /usr/include/opencascade -isystem /usr/include/vtk-9.2 -isystem /usr/include/freetype2 -isystem /usr/include/SDL2 -isystem /usr
/include/python3.11 -isystem /usr/include/python3.12  -march=core2 -O2 -pipe -std=gnu++17 -fPIC -fpermissive -fvisibility=hidden -fvisibility-inlines-hidden 
-fopenmp -MD -MT OCP/CMakeFiles/OCP.dir/AdvApp2Var_pre.cpp.o -MF OCP/CMakeFiles/OCP.dir/AdvApp2Var_pre.cpp.o.d -o OCP/CMakeFiles/OCP.dir/AdvApp2Var_pre.cpp.o -c /var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/AdvApp2Var_pre.cpp
FAILED: OCP/CMakeFiles/OCP.dir/AdvApp2Var_pre.cpp.o 
/usr/bin/x86_64-pc-linux-gnu-g++ -DHAVE_FFMPEG -DHAVE_FREETYPE -DHAVE_Inspector -DHAVE_OPENGL_EXT -DHAVE_QT -DHAVE_RAPIDJSON -DHAVE_TBB -DHAVE_TK -DHAVE_VTK -DHAVE_XLIB -DOCC_CONVERT_SIGNALS -DOCP_EXPORTS -DVTK_OPENGL2_BACKEND -Dkiss_fft_scalar=double -I/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP -isystem /usr/include/opencascade -isystem /usr/include/vtk-9.2 -isystem /usr/include/freetype2 -isystem /usr/include/SDL2 -isystem /usr/include/python3.11 -isystem /usr/include/python3.12  -march=core2 -O2 -pipe -std=gnu++17 -fPIC -fpermissive -fvisibility=hidden -fvisibility-inlines-hidden -fopenmp -MD -MT OCP/CMakeFiles/OCP.dir/AdvApp2Var_pre.cpp.o -MF OCP/CMakeFiles/OCP.dir/AdvApp2Var_pre.cpp.o.d -o OCP/CMakeFiles/OCP.dir/AdvApp2Var_pre.cpp.o -c /var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/AdvApp2Var_pre.cpp
In file included from /var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/AdvApp2Var_pre.cpp:114:
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/NCollection_tmpl.hxx: In function \u2018void register_template_NCollection_Sequence(pybind11::object&, const char*)\u2019:
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/NCollection_tmpl.hxx:838:60: error: expected unqualified-id before \u2018*\u2019 token
              (iterator (NCollection_Sequence<TheItemType>::*)() const) &NCollection_Sequence<TheItemType>::begin,
                                                            ^
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/NCollection_tmpl.hxx:838:61: error: expected primary-expression before \u2018)\u2019 token
              (iterator (NCollection_Sequence<TheItemType>::*)() const) &NCollection_Sequence<TheItemType>::begin,
                                                             ^
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/NCollection_tmpl.hxx:838:15: warning: there are no arguments to \u2018iterator\u2019 that depend on a template parameter, so a declaration of \u2018iterator\u2019 must be available [-fpermissive]
              (iterator (NCollection_Sequence<TheItemType>::*)() const) &NCollection_Sequence<TheItemType>::begin,
               ^~~~~~~~
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/NCollection_tmpl.hxx:838:64: error: expected \u2018)\u2019 before \u2018const\u2019
              (iterator (NCollection_Sequence<TheItemType>::*)() const) &NCollection_Sequence<TheItemType>::begin,
              ~                                                 ^~~~~~
                                                                )
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/NCollection_tmpl.hxx: In function \u2018void register_template_NCollection_EBTree(pybind11::object&, const char*)\u2019:
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/NCollection_tmpl.hxx:970:21: warning: ISO C++ forbids declaration of \u2018type name\u2019 with no type [-fpermissive]
              (const TreeNode & (NCollection_EBTree<TheObjType,TheBndType>::*)( const TheObjType &  ) const) &NCollection_EBTree<TheObjType,TheBndType>::FindNode,
                     ^~~~~~~~
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/NCollection_tmpl.hxx:970:15: error: expected primary-expression before \u2018const\u2019
              (const TreeNode & (NCollection_EBTree<TheObjType,TheBndType>::*)( const TheObjType &  ) const) &NCollection_EBTree<TheObjType,TheBndType>::FindNode,
               ^~~~~

<OMITTED REMAINING PART OF BUILD LOG ERRORS>
```

I am using GCC 12 and opencascade 7.7.2. I have also tried using gcc-8 or clang to build the CXX code without success.

@adam-urbanczyk which are the versions of build tools you suggest to use?

adam-urbanczyk, 2024-03-09:
I actually doubt that this is a toolchain issue. Could you first verify that you can compile the released generated sources with your toolchain to confirm/reject my doubt?

efferre79, 2024-03-09:
I cannot find any repo with the pywrap generated C++ files in the OCP subdir, can you kindly point to where I can find it?

adam-urbanczyk, 2024-03-09:
https://github.com/CadQuery/OCP/releases/tag/7.7.2.0 , `OCP_src*`

efferre79, 2024-03-09:
By using the released stubs OCP_src_stubs_ubuntu-20.04.zip you suggested the building starts but I get another error:

`[146/630] /usr/bin/x86_64-pc-linux-gnu-g++ -DHAVE_FFMPEG -DHAVE_FREETYPE -DHAVE_Inspector -DHAVE_OPENGL_EXT -DHAVE_QT -DHAVE_RAPIDJSON -DHAVE_TBB -DHAVE_TK -DHAVE_VTK -DHAVE_XLIB -DOCC_CONVERT_SIGNALS -DOCP_EXPORTS -DVTK_OPENGL2_BACKEND -Dkiss_fft_scalar=double -I/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r2/work/OCP-7.7.2.0/OCP -isystem /usr/include/opencascade -isystem /usr/include/vtk-9.2 -isystem /usr/include/freetype2 -isystem /usr/include/SDL2 -isystem /usr/include/python3.11 -isystem /usr/include/python3.12  -march=core2 -O2 -pipe -std=gnu++17 -fPIC -fpermissive -fvisibility=hidden -fvisibility-inlines-hidden -fopenmp -MD -MT OCP/CMakeFiles/OCP.dir/ChFi3d.cpp.o -MF OCP/CMakeFiles/OCP.dir/ChFi3d.cpp.o.d -o OCP/CMakeFiles/OCP.dir/ChFi3d.cpp.o -c /var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r2/work/OCP-7.7.2.0/OCP/ChFi3d.cpp
FAILED: OCP/CMakeFiles/OCP.dir/ChFi3d.cpp.o 
/usr/bin/x86_64-pc-linux-gnu-g++ -DHAVE_FFMPEG -DHAVE_FREETYPE -DHAVE_Inspector -DHAVE_OPENGL_EXT -DHAVE_QT -DHAVE_RAPIDJSON -DHAVE_TBB -DHAVE_TK -DHAVE_VTK -DHAVE_XLIB -DOCC_CONVERT_SIGNALS -DOCP_EXPORTS -DVTK_OPENGL2_BACKEND -Dkiss_fft_scalar=double -I/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r2/work/OCP-7.7.2.0/OCP -isystem /usr/include/opencascade -isystem /usr/include/vtk-9.2 -isystem /usr/include/freetype2 -isystem /usr/include/SDL2 -isystem /usr/include/python3.11 -isystem /usr/include/python3.12  -march=core2 -O2 -pipe -std=gnu++17 -fPIC -fpermissive -fvisibility=hidden -fvisibility-inlines-hidden -fopenmp -MD -MT OCP/CMakeFiles/OCP.dir/ChFi3d.cpp.o -MF OCP/CMakeFiles/OCP.dir/ChFi3d.cpp.o.d -o OCP/CMakeFiles/OCP.dir/ChFi3d.cpp.o -c /var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r2/work/OCP-7.7.2.0/OCP/ChFi3d.cpp
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r2/work/OCP-7.7.2.0/OCP/ChFi3d.cpp: In function ‘void register_ChFi3d(pybind11::module&)’:
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r2/work/OCP-7.7.2.0/OCP/ChFi3d.cpp:125:155: error: invalid ‘static_cast’ from type ‘Standard_Boolean (*)(const TopoDS_Edge&, const TopoDS_Face&, const TopoDS_Face&, GeomAbs_Shape)’ {aka ‘bool (*)(const TopoDS_Edge&, const TopoDS_Face&, const TopoDS_Face&, GeomAbs_Shape)’} to type ‘Standard_Boolean (*)(const TopoDS_Edge&, const TopoDS_Face&, const TopoDS_Face&, Standard_Real, GeomAbs_Shape)’ {aka ‘bool (*)(const TopoDS_Edge&, const TopoDS_Face&, const TopoDS_Face&, double, GeomAbs_Shape)’}
  125 |                     (Standard_Boolean (*)( const TopoDS_Edge & ,  const TopoDS_Face & ,  const TopoDS_Face & ,  Standard_Real ,  const GeomAbs_Shape  ) ) static_cast<Standard_Boolean (*)( const TopoDS_Edge & ,  const TopoDS_Face & ,  const TopoDS_Face & ,  Standard_Real ,  const GeomAbs_Shape  ) >(&ChFi3d::IsTangentFaces),
      |                                                                                                                                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
`

Furthermore by comparing the OCP sources between the released archive and the pywrap generated ones, they look quite different, for instance in the NCollection_tmpl.hxx file all the iterator related methods in register_template_NCollection_Sequence() are not present (inOCP_src_stubs_ubuntu-20.04.zip)

efferre79, 2024-03-09:
It looks like that shipped bindings OCP_src_stubs_ubuntu-20.04.zip were generated from an unpublished opencascade version (same error reported in #138). The function [isTangentFaces()](https://git.dev.opencascade.org/gitweb/?p=occt.git;a=blob;f=src/ChFi3d/ChFi3d.cxx;h=702b83b1de6d32edb2f22c3b005393efc28bcfe1;hb=cec1ecd0c9f3b3d2572c47035d11949e8dfa85e2#l137) for opencascade-7.7.2 has only four arguments

adam-urbanczyk, 2024-03-09:
Interesting, in the version that was used (occt 7.7.2 available in in the conda-forge channel) I see this:
```c++
//! creation of spatial fillets on a solid.
class ChFi3d 
{
public:

  DEFINE_STANDARD_ALLOC

  //! Defines the type of concavity in the edge of connection of two faces
  Standard_EXPORT static ChFiDS_TypeOfConcavity DefineConnectType (const TopoDS_Edge&     E,
                                                                   const TopoDS_Face&     F1,
                                                                   const TopoDS_Face&     F2,
                                                                   const Standard_Real    SinTol,
                                                                   const Standard_Boolean CorrectPoint);

  //! Returns true if theEdge between theFace1 and theFace2 is tangent
  Standard_EXPORT static Standard_Boolean IsTangentFaces (const TopoDS_Edge& theEdge,
                                                          const TopoDS_Face& theFace1,
                                                          const TopoDS_Face& theFace2,
                                                          const GeomAbs_Shape Order = GeomAbs_G1);
  
  Standard_EXPORT static Standard_Boolean IsTangentFaces (const TopoDS_Edge& theEdge,
                                                          const TopoDS_Face& theFace1,
                                                          const TopoDS_Face& theFace2,
                                                          Standard_Real G1Tol,
                                                          const GeomAbs_Shape Order = GeomAbs_G1);
```

So two overloads. I do not understand what happened here.

efferre79, 2024-03-09:
> Interesting, in the version that was used (occt 7.7.2 available in in the conda-forge channel) I see this:
>
> ```c++
> //! creation of spatial fillets on a solid.
> class ChFi3d
> {
> public: [...]

The change has been introduced [here](https://github.com/conda-forge/occt-feedstock/commit/ad5e2f0519ca536d33b592f2f98db794880985f9), maybe the author @Michsior14 can explain while the occt code in conda-forge is diverging from upstream. If that is really important because it solves an important issue, than I would strongly suggest to integrate the patch upstream!

efferre79, 2024-03-09:
@adam-urbanczyk the building error of this issue is related to an incomplete parsing of the API by pywrap.

Here I am guessing because I am far from being a C++ expert. If the class methods are using local types (e.g. the `iterator` typedef defined in the template class `NCollection_Vector<TheItemType>`), pywrap should use fully qualified identifiers (prepending the class name and the scope resolution operator).

For instance:

```
        .def("begin",
             (iterator (NCollection_Vector<TheItemType>::*)() const) &NCollection_Vector<TheItemType>::begin,
             R"#(Returns an iterator pointing to the first element in the vector.)#" )
        .def("end",
```

should become

```
        .def("begin",
             (typename NCollection_Vector<TheItemType>::iterator (NCollection_Vector<TheItemType>::*)() const) &NCollection_Vector<TheItemType>::begin,
             R"#(Returns an iterator pointing to the first element in the vector.)#" )
        .def("end",
```

in the output file `OCP/NCollection_tmpl.hxx` generated by pywrap. At least this solves the error from gcc, I was not able to complete the building process because there are many other errors like that. 

Another similar error, this time

```
/usr/bin/x86_64-pc-linux-gnu-g++ -DHAVE_FFMPEG -DHAVE_FREETYPE -DHAVE_Inspector -DHAVE_OPENGL_EXT -DHAVE_QT -DHAVE_RAPIDJSON -DHAVE_TBB -DHAVE_TK -DHAVE_VTK 
-DHAVE_XLIB -DOCC_CONVERT_SIGNALS -DOCP_EXPORTS -DVTK_OPENGL2_BACKEND -Dkiss_fft_scalar=double -I/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0/work/OCP-7.
7.2.0/OCP -isystem /usr/include/opencascade -isystem /usr/include/vtk-9.2 -isystem /usr/include/freetype2 -isystem /usr/include/SDL2 -isystem /usr/include/py
thon3.11 -isystem /usr/include/python3.12  -O2 -pipe -march=znver3 -ggdb -std=gnu++17 -fPIC -fpermissive -fvisibility=hidden -fvisibility-inlines-hidden -fop
enmp -MD -MT CMakeFiles/OCP.dir/AppDef_pre.cpp.o -MF CMakeFiles/OCP.dir/AppDef_pre.cpp.o.d -o CMakeFiles/OCP.dir/AppDef_pre.cpp.o -c /var/tmp/portage/dev-pyt
hon/cadquery-ocp-7.7.2.0/work/OCP-7.7.2.0/OCP/AppDef_pre.cpp
In file included from /var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0/work/OCP-7.7.2.0/OCP/AppDef_pre.cpp:267:
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0/work/OCP-7.7.2.0/OCP/NCollection_tmpl.hxx: In function ?void register_template_NCollection_UBTree(pybind11::
object&, const char*)?:
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0/work/OCP-7.7.2.0/OCP/NCollection_tmpl.hxx:292:32: error: expected primary-expression before ?(? token
  292 |              (Standard_Integer (NCollection_UBTree<TheObjType,TheBndType>::*)( Selector &  ) const) &NCollection_UBTree<TheObjType,TheBndType>::Select,
      |                                ^
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0/work/OCP-7.7.2.0/OCP/NCollection_tmpl.hxx:292:76: error: expected unqualified-id before ?*? token
  292 |              (Standard_Integer (NCollection_UBTree<TheObjType,TheBndType>::*)( Selector &  ) const) &NCollection_UBTree<TheObjType,TheBndType>::Select,
      |                                                                            ^
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0/work/OCP-7.7.2.0/OCP/NCollection_tmpl.hxx:292:77: error: expected primary-expression before ?)? token
  292 |              (Standard_Integer (NCollection_UBTree<TheObjType,TheBndType>::*)( Selector &  ) const) &NCollection_UBTree<TheObjType,TheBndType>::Select,
      |                                                                             ^

```
here the argument to `Select()` method should be qualified to `NCollection_UBTree<TheObjType,TheBndType>::Selector`.

adam-urbanczyk, 2024-03-10:
Yes, that sounds more like it. If this one overload is blocking, I can update `ocp.toml` to wrap it manually and make a new release. Otherwise, please share the stdout of your pywrap call. Note that I'll have limited time to help with this kind of issue. CI is green, so you should be able to reproduce the build.

Michsior14, 2024-03-10:
> The change has been introduced [here](https://github.com/conda-forge/occt-feedstock/commit/ad5e2f0519ca536d33b592f2f98db794880985f9), maybe the author @Michsior14 can explain while the occt code in [...]

You would need to ask [bloblfish](https://gitlab.com/blobfish/occt) or `occt-feedstock` maintainers. I've only updated the patch to v7.8.0 (it's been introduced some versions ago already).

efferre79, 2024-03-10:
> You would need to ask [bloblfish](https://gitlab.com/blobfish/occt) or [occt-feedstock](https://github.com/conda-forge/occt-feedstock?tab=readme-ov-file#feedstock-maintainers) maintainers. I've only [...]

already opened an [issue](https://github.com/conda-forge/occt-feedstock/issues/108)

adam-urbanczyk, 2024-03-14:
@efferre79 any thoughts/updates?

> [...] I can update `ocp.toml` to wrap it manually and make a new release. Otherwise, please share the stdout of your pywrap call. Note that I'll have limited time to help with this kind of issue. CI is green, so you should be able to reproduce the build.

efferre79, 2024-03-14:
@adam-urbanczyk sorry for the latency but up to now I was trying to patch pywrap to improve the generated code from the current ocp.toml. [Here](https://gist.github.com/efferre79/077b947f5ef413d39f21bd61159ff51e) you can find the log I get during the source configuration done by cmake. Let me know if that is what you asked for.

efferre79, 2024-03-17:
@adam-urbanczyk I have created a starting point to modify pywrap and fixing at least partially this issue, if you have time have a look at my branch [innertypes](https://github.com/efferre79/pywrap.git).

Right now I am stuck because I am facing something weird probably depending probably on a libclang bug, the error I get is the following:

```
ninja -v -j16 -l16
[1/630] /usr/bin/x86_64-pc-linux-gnu-g++ -DHAVE_FFMPEG -DHAVE_FREETYPE -DHAVE_Inspector -DHAVE_OPENGL_EXT -DHAVE_QT -DHAVE_RAPIDJSON -DHAVE_TBB -DHAVE_TK -DHAVE_VTK -DHAVE_XLIB -DOCC_CONVERT_SIGNALS -DOCP_EXPORTS -DVTK_OPENGL2_BACKEND -Dkiss_fft_scalar=double -I/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP -isystem /usr/include/opencascade -isystem /usr/include/vtk-9.2 -isystem /usr/include/freetype2 -isystem /usr/include/SDL2 -isystem /usr/include/python3.11 -isystem /usr/include/python3.12  -O2 -pipe -march=znver3 -ggdb -std=gnu++17 -fPIC -fpermissive -fvisibility=hidden -fvisibility-inlines-hidden -fopenmp -MD -MT OCP/CMakeFiles/OCP.dir/Adaptor2d_pre.cpp.o -MF OCP/CMakeFiles/OCP.dir/Adaptor2d_pre.cpp.o.d -o OCP/CMakeFiles/OCP.dir/Adaptor2d_pre.cpp.o -c /var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/Adaptor2d_pre.cpp
[2/630] /usr/bin/x86_64-pc-linux-gnu-g++ -DHAVE_FFMPEG -DHAVE_FREETYPE -DHAVE_Inspector -DHAVE_OPENGL_EXT -DHAVE_QT -DHAVE_RAPIDJSON -DHAVE_TBB -DHAVE_TK -DHAVE_VTK -DHAVE_XLIB -DOCC_CONVERT_SIGNALS -DOCP_EXPORTS -DVTK_OPENGL2_BACKEND -Dkiss_fft_scalar=double -I/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP -isystem /usr/include/opencascade -isystem /usr/include/vtk-9.2 -isystem /usr/include/freetype2 -isystem /usr/include/SDL2 -isystem /usr/include/python3.11 -isystem /usr/include/python3.12  -O2 -pipe -march=znver3 -ggdb -std=gnu++17 -fPIC -fpermissive -fvisibility=hidden -fvisibility-inlines-hidden -fopenmp -MD -MT OCP/CMakeFiles/OCP.dir/AppCont_pre.cpp.o -MF OCP/CMakeFiles/OCP.dir/AppCont_pre.cpp.o.d -o OCP/CMakeFiles/OCP.dir/AppCont_pre.cpp.o -c /var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/AppCont_pre.cpp
[3/630] /usr/bin/x86_64-pc-linux-gnu-g++ -DHAVE_FFMPEG -DHAVE_FREETYPE -DHAVE_Inspector -DHAVE_OPENGL_EXT -DHAVE_QT -DHAVE_RAPIDJSON -DHAVE_TBB -DHAVE_TK -DHAVE_VTK -DHAVE_XLIB -DOCC_CONVERT_SIGNALS -DOCP_EXPORTS -DVTK_OPENGL2_BACKEND -Dkiss_fft_scalar=double -I/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP -isystem /usr/include/opencascade -isystem /usr/include/vtk-9.2 -isystem /usr/include/freetype2 -isystem /usr/include/SDL2 -isystem /usr/include/python3.11 -isystem /usr/include/python3.12  -O2 -pipe -march=znver3 -ggdb -std=gnu++17 -fPIC -fpermissive -fvisibility=hidden -fvisibility-inlines-hidden -fopenmp -MD -MT OCP/CMakeFiles/OCP.dir/AppBlend_pre.cpp.o -MF OCP/CMakeFiles/OCP.dir/AppBlend_pre.cpp.o.d -o OCP/CMakeFiles/OCP.dir/AppBlend_pre.cpp.o -c /var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/AppBlend_pre.cpp
[4/630] /usr/bin/x86_64-pc-linux-gnu-g++ -DHAVE_FFMPEG -DHAVE_FREETYPE -DHAVE_Inspector -DHAVE_OPENGL_EXT -DHAVE_QT -DHAVE_RAPIDJSON -DHAVE_TBB -DHAVE_TK -DHAVE_VTK -DHAVE_XLIB -DOCC_CONVERT_SIGNALS -DOCP_EXPORTS -DVTK_OPENGL2_BACKEND -Dkiss_fft_scalar=double -I/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP -isystem /usr/include/opencascade -isystem /usr/include/vtk-9.2 -isystem /usr/include/freetype2 -isystem /usr/include/SDL2 -isystem /usr/include/python3.11 -isystem /usr/include/python3.12  -O2 -pipe -march=znver3 -ggdb -std=gnu++17 -fPIC -fpermissive -fvisibility=hidden -fvisibility-inlines-hidden -fopenmp -MD -MT OCP/CMakeFiles/OCP.dir/AdvApprox_pre.cpp.o -MF OCP/CMakeFiles/OCP.dir/AdvApprox_pre.cpp.o.d -o OCP/CMakeFiles/OCP.dir/AdvApprox_pre.cpp.o -c /var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/AdvApprox_pre.cpp
[5/630] /usr/bin/x86_64-pc-linux-gnu-g++ -DHAVE_FFMPEG -DHAVE_FREETYPE -DHAVE_Inspector -DHAVE_OPENGL_EXT -DHAVE_QT -DHAVE_RAPIDJSON -DHAVE_TBB -DHAVE_TK -DHAVE_VTK -DHAVE_XLIB -DOCC_CONVERT_SIGNALS -DOCP_EXPORTS -DVTK_OPENGL2_BACKEND -Dkiss_fft_scalar=double -I/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP -isystem /usr/include/opencascade -isystem /usr/include/vtk-9.2 -isystem /usr/include/freetype2 -isystem /usr/include/SDL2 -isystem /usr/include/python3.11 -isystem /usr/include/python3.12  -O2 -pipe -march=znver3 -ggdb -std=gnu++17 -fPIC -fpermissive -fvisibility=hidden -fvisibility-inlines-hidden -fopenmp -MD -MT OCP/CMakeFiles/OCP.dir/Adaptor3d_pre.cpp.o -MF OCP/CMakeFiles/OCP.dir/Adaptor3d_pre.cpp.o.d -o OCP/CMakeFiles/OCP.dir/Adaptor3d_pre.cpp.o -c /var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/Adaptor3d_pre.cpp
[6/630] /usr/bin/x86_64-pc-linux-gnu-g++ -DHAVE_FFMPEG -DHAVE_FREETYPE -DHAVE_Inspector -DHAVE_OPENGL_EXT -DHAVE_QT -DHAVE_RAPIDJSON -DHAVE_TBB -DHAVE_TK -DHAVE_VTK -DHAVE_XLIB -DOCC_CONVERT_SIGNALS -DOCP_EXPORTS -DVTK_OPENGL2_BACKEND -Dkiss_fft_scalar=double -I/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP -isystem /usr/include/opencascade -isystem /usr/include/vtk-9.2 -isystem /usr/include/freetype2 -isystem /usr/include/SDL2 -isystem /usr/include/python3.11 -isystem /usr/include/python3.12  -O2 -pipe -march=znver3 -ggdb -std=gnu++17 -fPIC -fpermissive -fvisibility=hidden -fvisibility-inlines-hidden -fopenmp -MD -MT OCP/CMakeFiles/OCP.dir/AppBlend.cpp.o -MF OCP/CMakeFiles/OCP.dir/AppBlend [... cut ...]
/usr/include/opencascade/AppDef_HArray1OfMultiPointConstraint.hxx:24:1: note: candidate: ‘AppDef_HArray1OfMultiPointConstraint::AppDef_HArray1OfMultiPointConstraint(AppDef_HArray1OfMultiPointConstraint&&)’
/usr/include/opencascade/AppDef_HArray1OfMultiPointConstraint.hxx:24:1: note:   candidate expects 1 argument, 3 provided

```
I have written a stupid python [script](https://gist.github.com/efferre79/07871d4277bd1a8fcfcb4848c40f341f) to demonstrate the problem, the parsing of the args of AppDef_HArray1OfMultiPointConstraint class constructors is not correct and lead to the error above.

adam-urbanczyk, 2024-03-17:
In the log dump that you shared you have this:
```
  0%|          | 0/314 [00:00<?, ?it/s][W 240314 09:52:14 translation_unit:52] ./opencascade/Standard_DefineException.hxx
[W 240314 09:52:14 translation_unit:53] /usr/lib/gcc/x86_64-pc-linux-gnu/12/include/g++-v12/cstddef:50:10: fatal error: 'stddef.h' file not found
```
You'll need to specify correct/additional (I'm not sure what they are) include dirs for your env.

adam-urbanczyk, 2024-03-17:
Maybe #140 solves the issue, let's see

efferre79, 2024-03-17:
> Maybe #140 solves the issue, let's see

Almost, on my system I had to further patch the `CMakeLists.txt` your _implicit-includes_ branch as follows (otherwise -i takes the separating space as first character of the path):

```
--- ./CMakeLists.txt    2024-03-17 22:02:52.569280098 +0100
+++ ./CMakeLists.txt.new    2024-03-17 22:05:24.544281844 +0100
@@ -34,7 +34,8 @@
 message( STATUS "Include dirs: ${CMAKE_CXX_IMPLICIT_INCLUDE_DIRECTORIES}")
 
 foreach( inc IN LISTS CMAKE_CXX_IMPLICIT_INCLUDE_DIRECTORIES)
-    list( APPEND CXX_INCLUDES "-i ${inc}/")
+    list( APPEND CXX_INCLUDES "-i" ) 
+    list( APPEND CXX_INCLUDES "${inc}/" )
 endforeach()
 
 if(${CMAKE_SYSTEM_NAME} STREQUAL "Windows")
```

But I would avoid forcing CXX_INCLUDES value with CMAKE_CXX_IMPLICIT_INCLUDE_DIRECTORIES (there is a reason they are not set, see https://cmake.org/cmake/help/latest/variable/CMAKE_LANG_IMPLICIT_INCLUDE_DIRECTORIES.html). When gcc is used as  compiler, it already knows them so it is not needed. I will try later with clang as CXX.

So I think it is better to provide them to pywrap execution only, defining a new PYWRAP_CXX_INCLUDES variable.

Furthermore your #140 patched as told above is solving my issue with AppDef_HArray1OfMultiPointConstraint class constructors described [here](https://github.com/CadQuery/OCP/issues/139#issuecomment-2002494460). I still get another error in the methods definition for non templates but probably that can be solved by further improving the patch of my pywrap branch. Working on it right now.

adam-urbanczyk, 2024-03-18:
AFAIK `CXX_INCLUDES` has no special meaning as a cmake variable and is not used. Are you maybe confusing it with an env variable?

efferre79, 2024-03-18:
> AFAIK `CXX_INCLUDES` has no special meaning as a cmake variable and is not used. Are you maybe confusing it with an env variable?

Yes, you are right. The name misled me, sorry for the noise.

efferre79, 2024-03-24:
hello @adam-urbanczyk, I have improved the pywrap code to almost successfully compile OCP. You can find the changes in my   [innertypes branch](https://github.com/efferre79/pywrap/tree/innertypes). I haven't created a pull request yet because only one missing error must be solved but I don't know what to do from the C++ point of view. The error I get is the following:

```
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/OpenGl.cpp:4269:15: error: \u2018template<class T, int N> class BVH_Set\u2019 used without tem
plate arguments
 4269 |              (BVH_Set::BVH_BoxNt (OpenGl_TriangleSet::*)() const) static_cast<BVH_Set::BVH_BoxNt (OpenGl_TriangleSet::*)() const>(&OpenGl_TriangleSet::Box),
      |               ^~~~~~~
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/OpenGl.cpp:4269:55: error: expected unqualified-id before \u2018*\u2019 token
 4269 |              (BVH_Set::BVH_BoxNt (OpenGl_TriangleSet::*)() const) static_cast<BVH_Set::BVH_BoxNt (OpenGl_TriangleSet::*)() const>(&OpenGl_TriangleSet::Box),
      |                                                       ^
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/OpenGl.cpp:4269:56: error: expected primary-expression before \u2018)\u2019 token
 4269 |              (BVH_Set::BVH_BoxNt (OpenGl_TriangleSet::*)() const) static_cast<BVH_Set::BVH_BoxNt (OpenGl_TriangleSet::*)() const>(&OpenGl_TriangleSet::Box),
      |                                                        ^
/var/tmp/portage/dev-python/cadquery-ocp-7.7.2.0-r1/work/OCP-7.7.2.0/OCP/OpenGl.cpp:4269:59: error: expected \u2018)\u2019 before \u2018const\u2019
 4269 |              (BVH_Set::BVH_BoxNt (OpenGl_TriangleSet::*)() const) static_cast<BVH_Set::BVH_BoxNt (OpenGl_TriangleSet::*)() const>(&OpenGl_TriangleSet::Box),
      |              ~                                            ^~~~~~
```
where of course the `BVH_Set::BVH_BoxNt` qualification comes out with my patch (the original code provides `BVH_BoxNt` which leads to a similar error shown at the beginning of this issue). Now the problem is that `BVH_Set` is a template class used by a typedef in the regular class `OpenGl_TriangleSet`. If I produce `BVH_Set<T,N>::BVH_BoxNt` it doesn't work as well because used in a non-template register function.

So currently I added `OpenGl_TriangleSet::Box` to the excluded methods list for the OpenGl module. I have seen that there are many blacklisted things in **ocp.toml** so the bindings would be incomplete anyway.

Moreover this is the difference in the shipped includes vs upstream files, wondering is something else should be added to **ocp.toml**:

```
--- /dev/fd/63	2024-03-24 16:15:37.525863873 +0100
+++ /dev/fd/62	2024-03-24 16:15:37.525863873 +0100
@@ -402,6 +402,9 @@
 BOPDS_VectorOfPoint.hxx
 BOPDS_VectorOfShapeInfo.hxx
 BOPDS_VectorOfVectorOfPair.hxx
+BOPTest.hxx
+BOPTest_DrawableShape.hxx
+BOPTest_Objects.hxx
 BOPTools_AlgoTools.hxx
 BOPTools_AlgoTools2D.hxx
 BOPTools_AlgoTools3D.hxx
@@ -889,6 +892,9 @@
 BRepSweep_Tool.hxx
 BRepSweep_Translation.hxx
 BRepSweep_Trsf.hxx
+BRepTest.hxx
+BRepTest_DrawableHistory.hxx
+BRepTest_Objects.hxx
 BRepToIGESBRep_Entity.hxx
 BRepToIGES_BREntity.hxx
 BRepToIGES_BRShell.hxx
@@ -1363,6 +1369,34 @@
 Convert_SequenceOfArray1OfPoles2d.hxx
 Convert_SphereToBSplineSurface.hxx
 Convert_TorusToBSplineSurface.hxx
+DBRep.hxx
+DBRep_DrawableShape.hxx
+DBRep_Edge.hxx
+DBRep_Edge.lxx
+DBRep_Face.hxx
+DBRep_Face.lxx
+DBRep_HideData.hxx
+DBRep_IsoBuilder.hxx
+DBRep_ListIteratorOfListOfEdge.hxx
+DBRep_ListIteratorOfListOfFace.hxx
+DBRep_ListIteratorOfListOfHideData.hxx
+DBRep_ListOfEdge.hxx
+DBRep_ListOfFace.hxx
+DBRep_ListOfHideData.hxx
+DBRep_Params.hxx
+DDF.hxx
+DDF_AttributeBrowser.hxx
+DDF_Browser.hxx
+DDF_Data.hxx
+DDF_ListIteratorOfTransactionStack.hxx
[... 245 lines cut ...]
+XSDRAWIGES.hxx
+XSDRAWSTEP.hxx
+XSDRAWSTLVRML.hxx
+XSDRAWSTLVRML_CoordsMap.hxx
+XSDRAWSTLVRML_DataMapIteratorOfCoordsMap.hxx
+XSDRAWSTLVRML_DataMapIteratorOfElemNodesMap.hxx
+XSDRAWSTLVRML_DataSource.hxx
+XSDRAWSTLVRML_DataSource3D.hxx
+XSDRAWSTLVRML_DrawableMesh.hxx
+XSDRAWSTLVRML_ElemNodesMap.hxx
+XSDRAW_Functions.hxx
+XSDRAW_Vars.hxx
 XmlDrivers.hxx
 XmlDrivers_DocumentRetrievalDriver.hxx
 XmlDrivers_DocumentStorageDriver.hxx
@@ -8138,6 +8323,7 @@
 gp_XY.hxx
 gp_XYZ.hxx
 igesread.h
+inspector
 math.hxx
 math_Array1OfValueAndWeight.hxx
 math_BFGS.hxx

```

adam-urbanczyk, 2024-03-24:
Too many changes at once, I'm afraid. What is the problem you are facing, can you generate code without warnings other than seen in the CI. If so, what is the delta of the generated code wrt to the released one [other than related to the patches of blobfish]?

efferre79, 2024-03-24:
[here](https://gist.github.com/efferre79/a96f914186a586c4174e615d205340d6) is the diff between the generated bindings (which don't compile) and the contents of **OCP_src_stubs_ubuntu-20.04.zip**

adam-urbanczyk, 2024-03-24:
How about warnings?

efferre79, 2024-03-24:
Here is the output from cmake

```
-- The C compiler identification is GNU 12.3.1
-- The CXX compiler identification is GNU 12.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/x86_64-pc-linux-gnu-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/x86_64-pc-linux-gnu-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAVE_FFI_CALL
-- Performing Test HAVE_FFI_CALL - Success
-- Found FFI: /usr/lib64/libffi.so  
-- Performing Test Terminfo_LINKABLE
-- Performing Test Terminfo_LINKABLE - Success
-- Found Terminfo: /usr/lib64/libtinfo.so  
-- Found ZLIB: /usr/lib64/libz.so (found version "1.3")  
-- Found LibXml2: /usr/lib64/libxml2.so (found version "2.12.5") 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found OGG: /usr/lib64/libogg.so  
-- Found THEORA: /usr/lib64/libtheora.so  
-- Found PkgConfig: /usr/bin/x86_64-pc-linux-gnu-pkg-config (found version "2.1.1") 
-- Found PEGTL: /usr/include/tao (found suitable version "2.8.3", minimum required is "2.0.0") 
-- Found FFMPEG: /usr/include (found suitable version "4.4.4", minimum required is "4.4") found components: avformat avcodec avutil swscale swresample 
-- Found GL2PS: /usr/lib64/libgl2ps.so (found suitable version "1.4.2", minimum required is "1.4.2") 
-- Found LibHaru: /usr/lib64/libhpdf.so (found suitable version "2.4.2", minimum required is "2.4.0") 
-- Found JsonCpp: /usr/lib64/libjsoncpp.so (found suitable version "1.9.5", minimum required is "0.7.0") 
-- Found NetCDF: /usr/include (found version "4.9.2") 
-- Found CGNS: /usr/include (found suitable version "4.30", minimum required is "4.10") 
-- Found HDF5: /usr/lib64/libhdf5.so;/usr/lib64/libz.so;/usr/lib64/libdl.a;/usr/lib64/libm.so (found version "1.14.3") found components: C HL 
-- Found Boost: /usr/lib64/cmake/Boost-1.84.0/BoostConfig.cmake (found suitable exact version "1.84.0")  
-- Found Boost: /usr/lib64/cmake/Boost-1.84.0/BoostConfig.cmake (found suitable exact version "1.84.0") found components: serialization 
-- Found LibPROJ: /usr/lib64/libproj.so (found version "9.2.1") 
-- Found SQLite3: /usr/include (found version "3.45.1") 
-- Found nlohmann_json: /usr/share/cmake/nlohmann_json/nlohmann_jsonConfig.cmake (found version "3.11.2") 
-- Found X11: /usr/include   
-- Looking for XOpenDisplay in /usr/lib64/libX11.so;/usr/lib64/libXext.so
-- Looking for XOpenDisplay in /usr/lib64/libX11.so;/usr/lib64/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found GLEW: /usr/lib64/libGLEW.so  
-- Found OpenGL: /usr/lib64/libOpenGL.so  found components: OpenGL GLX 
-- Found Python3: /usr/bin/python3.12 (found suitable version "3.12.2", minimum required is "3.11") found components: Interpreter Development.Module Development.Embed 
-- Found EXPAT: /usr/lib64/libexpat.so (found version "2.5.0") 
-- Found LZMA: /usr/lib64/liblzma.so (found version "5.4.6") 
-- Found LZ4: /usr/lib64/liblz4.so (found version "1.9.4") 
-- Found double-conversion: /usr/lib64/libdouble-conversion.so  
-- Found Eigen3: /usr/include/eigen3 (found version "3.4.0") 
-- Found TIFF: /usr/lib64/libtiff.so (found version "4.6.0")  
-- Found PNG: /usr/lib64/libpng.so (found version "1.6.43+apng") 
-- Found JPEG: /usr/lib64/libjpeg.so (found version "62") 
-- Found Freetype: /usr/lib64/libfreetype.so (found version "2.13.2") 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
[... 385 lines cut ...]
Processing module XCAFNoteObjects
Processing module XCAFPrs
Processing module XCAFView
Processing module BinXCAFDrivers
Processing module XmlXCAFDrivers
Processing module Vrml
Processing module VrmlAPI
Processing module RWGltf
Processing module IVtk
Processing module IVtkOCC
Processing module IVtkTools
Processing module IVtkVTK
Processing module DE
Processing module Xw
-- Found Python: /usr/bin/python3.11 (found version "3.11.8") found components: Interpreter Development Development.Module Development.Embed 
-- Python lib: /usr/lib64/libpython3.11.so
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Found pybind11: /usr/include (found version "2.11.1")
-- Found Boost: /usr/lib64/cmake/Boost-1.84.0/BoostConfig.cmake (found suitable exact version "1.84.0")  
-- Found Boost: /usr/lib64/cmake/Boost-1.84.0/BoostConfig.cmake (found suitable exact version "1.84.0") found components: serialization 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 

```

efferre79, 2024-03-30:
@adam-urbanczyk a quick update of what I discovered.

First of all I have finally managed to build OCP with clang 14 and 15. But this bug is still valid when using clang 16 or greater. See also below.

I am using a [Gentoo-based](https://blogs.gentoo.org/mgorny/2022/10/07/clang-in-gentoo-now-sets-default-runtimes-via-config-file/) system, they _partially_ backported the clang 16 functionality of system config files in clang15. So building OCP with clang 15 didn't work out of the box on my system because the system config files have some `-I` clang directives which are causing parsing errors with OCP build system. The option `--no-default-config` is not backported but I had to find another workaround by setting the CLANG_NO_DEFAULT_CONFIG env variable in the Gentoo build receipt. In the [commit ](https://github.com/CadQuery/pywrap/pull/53) I unconditionally added `--no-default-config` but that is wrong because it can be used only with clang 16 or later (causing errors when using clang 15). Of course `--no-default-config` may still be needed on a generic system when using newer versions of clang, so I created an [improved PR](https://github.com/CadQuery/pywrap/pull/56).

Then the bad news. Unfortunately, starting from clang version 16 the behaviour is changed (probably the reason is explained [here](https://github.com/llvm/llvm-project/issues/61810)) so it is not possible to get the same kind of information required for types qualification of function arguments/return values.

adam-urbanczyk, 2024-04-01:
Thanks, so for the time being the solution is to use the configuration specified in this repo (i.e. clang 15).

 I do not understand the story with `--no-default-config` and Gentoo. I'd rather remove this completely from pywrap to be honest, especially that you apparently found a way to specify the correct headers.

efferre79, 2024-04-02:
> Thanks, so for the time being the solution is to use the configuration specified in this repo (i.e. clang 15).
>
> I do not understand the story with `--no-default-config` and Gentoo. I'd rather remove [...]

The commit using `--no-default-config` is not Gentoo specific in the sense it might potentially address the same problem with other distributions. Basically if you do 
```
$ clang 2>&1 -v | grep -i config
Configuration file: /etc/clang/x86_64-pc-linux-gnu-clang.cfg
System configuration file directory: /etc/clang
```
you can see if clang supports system configuration files. This is enabled upstream from clang-16, on Gentoo since clang-15. On Gentoo, they addressed another [issue](https://bugs.gentoo.org/show_bug.cgi?id=870001) and since then one of the config files contains a clang `--include` directive passed implicitely to the compiler. Include files specified on command line lead to a `AttributeError: 'NoneType' object has no attribute 'name'` error in pywrap ([here](https://github.com/CadQuery/pywrap/blob/6503d68c113410584fc62c2eaca8c8d294560a2a/bindgen/header.py#L707)) because they don't have a valid Path() object associated in your code. It took a lot of time to understand the source of the problem and for that reason I proposed the change by using `--no-default-config` option to overcome that issue. 

If you decide to remove the previous commit, I would strongly suggest to make pywrap more robust. For instance you can filter out the invalid HeaderInfo() objects or printing a warning or in alternative, you could foresee an user defined clang command line option to be added when parsing the translation units.

adam-urbanczyk, 2024-04-03:
Alright, I'll update the README to say something about the deps (i.e. essentially take the versions from the devenv file or be on your own).

Would you happen to know why includes added via `--include` would not have a file name according to libclang? Sounds like a bug to me.

Last, but not least, shall I get rid of the problematic overload coming from the blobfish patch? AFAICT this way everyone will be able to use the generated sources directly? Does that fit your use case?

efferre79, 2024-04-05:
> Would you happen to know why includes added via `--include` would not have a file name according to libclang? Sounds like a bug to me.

I have no idea.

> Last, but not least, shall I get rid of the problematic overload coming from the blobfish patch? AFAICT this way everyone will be able to use the generated sources directly? Does that fit your use case?

I am using directly opencascade headers from my distribution and I don't use conda at all.  In my opinion having to ship the generated sources is quite limiting, it should be a backup option and considered as last option.

To be honest I don't see the reason of keeping blobfish patch, even the maintainers don't know what is its purpose. The most famous software using opencascade seems to be freecad and its build system can be used directly with upstream opencascade.

adam-urbanczyk, 2025-07-07:
Closing, on master at least clang 20 is required now btw.

closed 2025-07-07 by adam-urbanczyk