CadQuery/OCP #147: Problem with IVtkOCC_Shape at 7.7.2

BlankuApp, 2024-04-28:
When i install the OCP using `conda install -c conda-forge -c cadquery ocp` , I cannot run the following code:
```
        vtk_shape = IVtkOCC_Shape(self._ocp)
        shape_data = IVtkVTK_ShapeData()
        shape_mesher = IVtkOCC_ShapeMesher()

        drawer = vtk_shape.Attributes()
        drawer.SetUIsoAspect(Prs3d_IsoAspect(Quantity_Color(), Aspect_TOL_SOLID, 1, 0))
        drawer.SetVIsoAspect(Prs3d_IsoAspect(Quantity_Color(), Aspect_TOL_SOLID, 1, 0))
        drawer.SetFaceBoundaryDraw(True)

        shape_mesher.Build(vtk_shape, shape_data)

        rv = shape_data.getVtkPolyData()
```
it returnes `rv=None`

But when i install the older versions `conda install -c conda-forge -c cadquery ocp=7.7.0` the problem is solved.

adam-urbanczyk, 2024-04-28:
Use the `conda-forge` build: `conda install -c conda-forge ocp=7.7.2`.

dvc94ch, 2024-07-01:
ran into this issue too using poetry, will try with conda

dvc94ch, 2024-07-03:
python packages are still a mess, conda doesn't support nushell, arch linux doesn't have a proper conda package (only aur), but finally got it working with pixi to install the conda packages + pip and using pip to install build123d. (also tried nix and going on some other tangents). but the result seems like the simplest way to show a model without vscode:

```
# build model
import build123d as bd

length, width, thickness = 80.0, 60.0, 10.0
center_hole_dia = 22.0

ex2 = bd.Box(length, width, thickness)
ex2 -= bd.Cylinder(center_hole_dia / 2, height=thickness)

# convert to pyvista
import OCP
import pyvista as pv

vs = OCP.IVtkOCC.IVtkOCC_Shape(ex2.solid().wrapped)
sd = OCP.IVtkVTK.IVtkVTK_ShapeData()
sm = OCP.IVtkOCC.IVtkOCC_ShapeMesher()
sm.Build(vs, sd)
data = sd.getVtkPolyData()
mesh = pv.PolyData(data)

# plot mesh in pyvista
plotter = pv.Plotter()
plotter.add_mesh(mesh)
plotter.show()
```

adam-urbanczyk, 2024-07-03:
The conclusion here is that one needs to use abi compatible builds of OCP/OCCT/VTK. The most robust way is to install on any distro/shell is to directly download `micromamba`, not init the shell, but rather use `micromamba run`.  https://mamba.readthedocs.io/en/latest/user_guide/micromamba.html

BTW, you can use `show` from `cadquery.vis` for visualizing `TopoDS_Shape` with VTK only.

closed 2024-07-03 by adam-urbanczyk