CadQuery/OCP #149: tests/test_assembly.py::test_colors_assy0[chassis0_assy-expected0] Fatal Python error: Segmentation fault

l29ah, 2024-05-13:
I'm packaging CadQuery for Gentoo. I've built OCP with the pre-generated "ubuntu" bindings sources as suggested in https://github.com/CadQuery/OCP/issues/132#issuecomment-1851412570
CadQuery master pytest segfaults in (through?) OCP:
```
python3.11 -m pytest -vv -ra -l -Wdefault -Werror::pytest.PytestUnhandledCoroutineWarning --color=yes -o console_output_style=count -o tmp_path_retention_count=0 -o tmp_path_retention_policy=failed -p no:cov -p no:flake8 -p no:flakes -p no:pylint -p no:markdown -p no:sugar -p no:xvfb -p no:pytest-describe -p no:plus -p no:tavern -p no:salt-factories
===================================================== test session starts ======================================================
platform linux -- Python 3.11.9, pytest-8.2.0, pluggy-1.5.0 -- /var/tmp/portage/media-gfx/cadquery-9999/work/cadquery-9999_python3.11/install/usr/bin/python3.11
cachedir: .pytest_cache
rootdir: /var/tmp/portage/media-gfx/cadquery-9999/work/cadquery-9999
plugins: anyio-4.2.0, pkgcore-0.12.27
collected 576 items                                                                                                            

tests/test_assembly.py::test_metadata PASSED                                                                          [  1/576]
tests/test_assembly.py::test_color PASSED                                                                             [  2/576]
tests/test_assembly.py::test_assembly PASSED                                                                          [  3/576]
tests/test_assembly.py::test_assy_root_name[simple_assy-None] PASSED                                                  [  4/576]
tests/test_assembly.py::test_assy_root_name[nested_assy-TOP] PASSED                                                   [  5/576]
tests/test_assembly.py::test_step_export PASSED                                                                       [  6/576]
tests/test_assembly.py::test_native_export PASSED                                                                     [  7/576]
tests/test_assembly.py::test_vtkjs_export PASSED                                                                      [  8/576]
tests/test_assembly.py::test_vrml_export PASSED                                                                       [  9/576]
tests/test_assembly.py::test_toJSON PASSED                                                                            [ 10/576]
tests/test_assembly.py::test_save[step-args0] PASSED                                                                  [ 11/576]
tests/test_assembly.py::test_save[xml-args1] PASSED                                                                   [ 12/576]
tests/test_assembly.py::test_save[stp-args2] PASSED                                                                   [ 13/576]
tests/test_assembly.py::test_save[caf-args3] PASSED                                                                   [ 14/576]
tests/test_assembly.py::test_save[wrl-args4] PASSED                                                                   [ 15/576]
tests/test_assembly.py::test_save[stl-args5] PASSED                                                                   [ 16/576]
tests/test_assembly.py::test_save_stl_formats PASSED                                                                  [ 17/576]
tests/test_assembly.py::test_save_gltf PASSED                                                                         [ 18/576]
tests/test_assembly.py::test_exportGLTF PASSED                                                                        [ 19/576]
tests/test_assembly.py::test_save_gltf_boxes2 PASSED                                                                  [ 20/576]
tests/test_assembly.py::test_save_vtkjs PASSED                                                                        [ 21/576]
tests/test_assembly.py::test_save_raises PASSED                                                                       [ 22/576]
tests/test_assembly.py::test_leaf_node_count[simple_assy-3] PASSED                                                    [ 23/576]
tests/test_assembly.py::test_leaf_node_count[nested_assy-3] PASSED                                                    [ 24/576]
tests/test_assembly.py::test_leaf_node_count[empty_top_assy-1] PASSED                                                 [ 25/576]
tests/test_assembly.py::test_colors_assy0[chassis0_assy-expected0] Fatal Python error: Segmentation fault

Current thread 0x00007f8d195afe80 (most recent call first):
  File "/var/tmp/portage/media-gfx/cadquery-9999/work/cadquery-9999/cadquery/occ_impl/shapes.py", line 400 in clean
  File "/var/tmp/portage/media-gfx/cadquery-9999/work/cadquery-9999/cadquery/cq.py", line 3300 in <listcomp>
  File "/var/tmp/portage/media-gfx/cadquery-9999/work/cadquery-9999/cadquery/cq.py", line 3299 in _combineWithBase
  File "/var/tmp/portage/media-gfx/cadquery-9999/work/cadquery-9999/cadquery/cq.py", line 3149 in extrude
  File "/var/tmp/portage/media-gfx/cadquery-9999/work/cadquery-9999/tests/test_assembly.py", line 309 in chassis0_assy
  File "/usr/lib/python3.11/site-packages/_pytest/fixtures.py", line 884 in call_fixture_func
  File "/usr/lib/python3.11/site-packages/_pytest/fixtures.py", line 1122 in pytest_fixture_setup
  File "/usr/lib/python3.11/site-packages/pluggy/_callers.py", line 103 in _multicall
  File "/usr/lib/python3.11/site-packages/pluggy/_manager.py", line 120 in _hookexec
  File "/usr/lib/python3.11/site-packages/pluggy/_hooks.py", line 513 in __call__
  File "/usr/lib/python3.11/site-packages/_pytest/fixtures.py", line 1073 in execute
  File "/usr/lib/python3.11/site-packages/_pytest/fixtures.py", line 603 in _get_active_fixturedef
  File "/usr/lib/python3.11/site-packages/_pytest/fixtures.py", line 518 in getfixturevalue
  File "/var/tmp/portage/media-gfx/cadquery-9999/work/cadquery-9999/tests/test_assembly.py", line 804 in test_colors_assy0
  File "/usr/lib/python3.11/site-packages/_pytest/python.py", line 162 in pytest_pyfunc_call
  File "/usr/lib/python3.11/site-packages/pluggy/_callers.py", line 103 in _multicall
  File "/usr/lib/python3.11/site-packages/pluggy/_manager.py", line 120 in _hookexec
  File "/usr/lib/python3.11/site-packages/pluggy/_hooks.py", line 513 in __call__
  File "/usr/lib/python3.11/site-packages/_pytest/python.py", line 1627 in runtest
  File "/usr/lib/python3.11/site-packages/_pytest/runner.py", line 173 in pytest_runtest_call
  File "/usr/lib/python3.11/site-packages/pluggy/_callers.py", line 103 in _multicall
  File "/usr/lib/python3.11/site-packages/pluggy/_manager.py", line 120 in _hookexec
  File "/usr/lib/python3.11/site-packages/pluggy/_hooks.py", line 513 in __call__
  File "/usr/lib/python3.11/site-packages/_pytest/runner.py", line 241 in <lambda>
  File "/usr/lib/python3.11/site-packages/_pytest/runner.py", line 341 in from_call
  File "/usr/lib/python3.11/site-packages/_pytest/runner.py", line 240 in call_and_report
  File "/usr/lib/python3.11/site-packages/_pytest/runner.py", line 135 in runtestprotocol
  File "/usr/lib/python3.11/site-packages/_pytest/runner.py", line 116 in pytest_runtest_protocol
  File "/usr/lib/python3.11/site-packages/pluggy/_callers.py", line 103 in _multicall
  File "/usr/lib/python3.11/site-packages/pluggy/_manager.py", line 120 in _hookexec
  File "/usr/lib/python3.11/site-packages/pluggy/_hooks.py", line 513 in __call__
[... 76 lines cut ...]
#55 0x00007f8d190d34ff in _PyObject_MakeTpCall () from /usr/lib64/libpython3.11.so.1.0
#56 0x00007f8d190864e3 in _PyEval_EvalFrameDefault () from /usr/lib64/libpython3.11.so.1.0
#57 0x00007f8d191afbf0 in _PyEval_Vector () from /usr/lib64/libpython3.11.so.1.0
#58 0x00007f8d190805d9 in _PyEval_EvalFrameDefault () from /usr/lib64/libpython3.11.so.1.0
#59 0x00007f8d191afbf0 in _PyEval_Vector () from /usr/lib64/libpython3.11.so.1.0
#60 0x00007f8d190d36db in _PyObject_FastCallDictTstate () from /usr/lib64/libpython3.11.so.1.0
#61 0x00007f8d190d3960 in _PyObject_Call_Prepend () from /usr/lib64/libpython3.11.so.1.0
#62 0x00007f8d19139b34 in slot_tp_call () from /usr/lib64/libpython3.11.so.1.0
#63 0x00007f8d190d34ff in _PyObject_MakeTpCall () from /usr/lib64/libpython3.11.so.1.0
#64 0x00007f8d190864e3 in _PyEval_EvalFrameDefault () from /usr/lib64/libpython3.11.so.1.0
#65 0x00007f8d191afbf0 in _PyEval_Vector () from /usr/lib64/libpython3.11.so.1.0
#66 0x00007f8d191afcb1 in PyEval_EvalCode () from /usr/lib64/libpython3.11.so.1.0
[... cut ...]
#71 0x00007f8d191afbf0 in _PyEval_Vector () from /usr/lib64/libpython3.11.so.1.0
#72 0x00007f8d1921174a in pymain_run_module () from /usr/lib64/libpython3.11.so.1.0
#73 0x00007f8d19211f3a in Py_RunMain () from /usr/lib64/libpython3.11.so.1.0
#74 0x00007f8d1921272b in Py_BytesMain () from /usr/lib64/libpython3.11.so.1.0
#75 0x00007f8d18e5f2e0 in __libc_start_call_main () from /lib64/libc.so.6
#76 0x00007f8d18e5f399 in __libc_start_main () from /lib64/libc.so.6
#77 0x00005617a6166095 in _start ()
```
How do i address this?

adam-urbanczyk, 2024-05-14:
Would you be able to step through the failing test in python share the object (as `brep`) passed to `ShapeUpgrade_UnifySameDomain`?

l29ah, 2024-05-14:
I don't understand what you're asking me to provide. There're more than two segfaulting tests, as i'm commenting them out. Is your request about `testBrepImportExport`, or about the first crashing test context, or something else?

adam-urbanczyk, 2024-05-17:
So the idea was that you debug (in pdb) the first segfaulting test and share the problematic shape as a `brep` file. BTW: how is OCCT compiled? Can you share the flags passed to cmake?

l29ah, 2024-05-17:
It's compiled by a Gentoo ebuild. `cmake -C /var/tmp/portage/sci-libs/opencascade-7.7.2-r1/work/OCCT-7_7_2_build/gentoo_common_config.cmake -G Ninja -DCMAKE_INSTALL_PREFIX=/usr -D3RDPARTY_DIR=/usr -DBUILD_CPP_STANDARD=C++17 -DBUILD_SOVERSION_NUMBERS=2 -DBUILD_DOC_Overview=no -DBUILD_Inspector=no -DBUILD_ENABLE_FPE_SIGNAL_HANDLER=no -DBUILD_USE_PCH=no -DBUILD_RELEASE_DISABLE_EXCEPTIONS=no -DINSTALL_DIR=/usr -DINSTALL_DIR_BIN=lib64/opencascade/bin -DINSTALL_DIR_CMAKE=lib64/cmake/opencascade -DINSTALL_DIR_DATA=share/opencascade/data -DINSTALL_DIR_DOC=share/doc/opencascade-7.7.2-r1 -DINSTALL_DIR_INCLUDE=include/opencascade -DINSTALL_DIR_LIB=lib64/opencascade -DINSTALL_DIR_RESOURCE=share/opencascade/resources -DINSTALL_DIR_SAMPLES=share/opencascade/samples -DINSTALL_DIR_SCRIPT=lib64/opencascade/bin -DINSTALL_DIR_TESTS=share/opencascade/tests -DINSTALL_DIR_WITH_VERSION=no -DINSTALL_SAMPLES=no -DINSTALL_TEST_CASES=no -DUSE_DRACO=no -DUSE_FFMPEG=no -DUSE_FREEIMAGE=no -DUSE_FREETYPE=no -DUSE_GLES2=no -DUSE_OPENGL=yes -DUSE_OPENVR=no -DUSE_RAPIDJSON=yes -DUSE_TBB=no -DUSE_TK=no -DUSE_VTK=yes -DUSE_XLIB=yes -D3RDPARTY_VTK_INCLUDE_DIR=/usr/include/vtk-9.3 -D3RDPARTY_VTK_LIBRARY_DIR=/usr/lib64 -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/sci-libs/opencascade-7.7.2-r1/work/OCCT-7_7_2_build/gentoo_toolchain.cmake /var/tmp/portage/sci-libs/opencascade-7.7.2-r1/work/OCCT-7_7_2`

Will try with pdb in a while.

l29ah, 2024-05-17:
I failed to export brep, but produced a minimal segfault reproducer:
```
#!/usr/bin/python3
import cadquery as cq

r_wheel = 25
w_wheel = 10
wheel = cq.Workplane("YZ").circle(r_wheel).extrude(w_wheel, both=True)
wheel.val().exportBrep("wheel.brep")
```

l29ah, 2024-05-17:
The `circle` brep: https://tinystash.undef.im/il/5MpwSPcG7mTHLufdLBqbVD9s94u5WQAxigxoYm6Hf3e3S3R1dZsFhSLKD8D9yzSv8EgfeRN5jzK1rDtNCeK2ozwm.brep

adam-urbanczyk, 2024-05-17:
That does not look too weird. I assume that `cq.Workplane("YZ").circle(r_wheel).extrude(w_wheel, both=True, clean=False)` does not segfault, right?

Looking at your cmake flags, `-DBUILD_ENABLE_FPE_SIGNAL_HANDLER=no`  might be suspicious. I never unset it, could you check what happens without it?

l29ah, 2024-05-17:
On Fri, May 17, 2024 at 10:38:20AM GMT, AU wrote:
> That does not look too weird. I assume that `cq.Workplane("YZ").circle(r_wheel).extrude(w_wheel, both=True, clean=False)` does not segfault, right?

Yes, this doesn't segfault.

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

l29ah, 2024-05-17:
> Looking at your cmake flags, -DBUILD_ENABLE_FPE_SIGNAL_HANDLER=no might be suspicious. I never unset it, could you check what happens without it?

No effect.

adam-urbanczyk, 2024-05-17:
OK, could you then check with a debug build of OCCT and share the coredump?

l29ah, 2024-05-17:
On Fri, May 17, 2024 at 11:26:42AM GMT, AU wrote:
> OK, could you then check with a debug build of OCCT and share the coredump?

I have the coredump. I doubt it would be of any use without the accompanying symbols tho, and i'm not sure how to provide them nicely.
The backtrace and a little more info when compiled with -Og -ggdb3:
```
(gdb) bt
#0  TopoDS_TShape::setFlag (theIsOn=false, theFlag=TopoDS_TShape::TopoDS_TShape_Flags_Free, this=0x0) at /var/tmp/portage/sci-libs/opencascade-7.7.2-r1/work/OCCT-7_7_2/src/TopoDS/TopoDS_TShape.hxx:170
#1  TopoDS_TShape::Free (theIsFree=false, this=0x0) at /var/tmp/portage/sci-libs/opencascade-7.7.2-r1/work/OCCT-7_7_2/src/TopoDS/TopoDS_TShape.hxx:64
#2  TopoDS_Builder::Add ***@***.***=0x7ffd23863063, aShape=..., aComponent=...) at /var/tmp/portage/sci-libs/opencascade-7.7.2-r1/work/OCCT-7_7_2/src/TopoDS/TopoDS_Builder.cxx:49
#3  0x00007f6f811b9c34 in ShapeUpgrade_UnifySameDomain::IntUnifyFaces ***@***.***=0x564f88339690, theInpShape=..., theGMapEdgeFaces=..., theGMapFaceShells=..., theFreeBoundMap=...)
    at /var/tmp/portage/sci-libs/opencascade-7.7.2-r1/work/OCCT-7_7_2/src/ShapeUpgrade/ShapeUpgrade_UnifySameDomain.cxx:3613
#4  0x00007f6f811bb4b0 in ShapeUpgrade_UnifySameDomain::UnifyFaces ***@***.***=0x564f88339690) at /var/tmp/portage/sci-libs/opencascade-7.7.2-r1/work/OCCT-7_7_2/src/ShapeUpgrade/ShapeUpgrade_UnifySameDomain.cxx:2807
#5  0x00007f6f811bb7da in ShapeUpgrade_UnifySameDomain::Build (this=0x564f88339690) at /var/tmp/portage/sci-libs/opencascade-7.7.2-r1/work/OCCT-7_7_2/src/ShapeUpgrade/ShapeUpgrade_UnifySameDomain.cxx:3900
#6  0x00007f6f897d09b7 in pybind11::cpp_function::initialize<pybind11::cpp_function::initialize<void, ShapeUpgrade_UnifySameDomain, , pybind11::name, pybind11::is_method, pybind11::sibling, char [53]>(void (ShapeUpgrade_UnifySameDomain::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [53])::{lambda(ShapeUpgrade_UnifySameDomain*)#1}, void, ShapeUpgrade_UnifySameDomain*, pybind11::name, pybind11::is_method, pybind11::sibling, char [53]>(pybind11::cpp_function::initialize<void, ShapeUpgrade_UnifySameDomain, , pybind11::name, pybind11::is_method, pybind11::sibling, char [53]>(void (ShapeUpgrade_UnifySameDomain::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [53])::{lambda(ShapeUpgrade_UnifySameDomain*)#1}&&, void (*)(ShapeUpgrade_UnifySameDomain*), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [53])::{lambda(pybind11::detail::function_call&)#1}::_FUN(pybind11::detail::function_call&) () from /usr/lib/python3.11/site-packages/OCP.cpython-311-x86_64-linux-gnu.so
#7  0x00007f6f84a27da3 in pybind11::cpp_function::dispatcher(_object*, _object*, _object*) () from /usr/lib/python3.11/site-packages/OCP.cpython-311-x86_64-linux-gnu.so
#8  0x00007f6f8f9175d3 in cfunction_call () from /usr/lib64/libpython3.11.so.1.0
#9  0x00007f6f8f8d34ff in _PyObject_MakeTpCall () from /usr/lib64/libpython3.11.so.1.0
#10 0x00007f6f8f8864e3 in _PyEval_EvalFrameDefault () from /usr/lib64/libpython3.11.so.1.0
#11 0x00007f6f8f9afbf0 in _PyEval_Vector () from /usr/lib64/libpython3.11.so.1.0
#12 0x00007f6f8f9afcb1 in PyEval_EvalCode () from /usr/lib64/libpython3.11.so.1.0
#13 0x00007f6f8f9f1653 in run_eval_code_obj () from /usr/lib64/libpython3.11.so.1.0
#14 0x00007f6f8f9f1876 in run_mod () from /usr/lib64/libpython3.11.so.1.0
#15 0x00007f6f8f9f1950 in pyrun_file () from /usr/lib64/libpython3.11.so.1.0
#16 0x00007f6f8f9f438d in _PyRun_SimpleFileObject () from /usr/lib64/libpython3.11.so.1.0
#17 0x00007f6f8f9f48f4 in _PyRun_AnyFileObject () from /usr/lib64/libpython3.11.so.1.0
#18 0x00007f6f8fa121f0 in Py_RunMain () from /usr/lib64/libpython3.11.so.1.0
#19 0x00007f6f8fa1272b in Py_BytesMain () from /usr/lib64/libpython3.11.so.1.0
#20 0x00007f6f8f65f2e0 in __libc_start_call_main () from /lib64/libc.so.6
#21 0x00007f6f8f65f399 in __libc_start_main () from /lib64/libc.so.6
#22 0x0000564f81293095 in _start ()
(gdb) down
Bottom (innermost) frame selected; you cannot go down.
(gdb) up
#1  TopoDS_TShape::Free (theIsFree=false, this=0x0) at /var/tmp/portage/sci-libs/opencascade-7.7.2-r1/work/OCCT-7_7_2/src/TopoDS/TopoDS_TShape.hxx:64
64	  void Free (Standard_Boolean theIsFree) { setFlag (TopoDS_TShape_Flags_Free, theIsFree); }
(gdb) down
#0  TopoDS_TShape::setFlag (theIsOn=false, theFlag=TopoDS_TShape::TopoDS_TShape_Flags_Free, this=0x0) at /var/tmp/portage/sci-libs/opencascade-7.7.2-r1/work/OCCT-7_7_2/src/TopoDS/TopoDS_TShape.hxx:170
170	    else         myFlags &= ~(Standard_Integer )theFlag;
(gdb) up
#1  TopoDS_TShape::Free (theIsFree=false, this=0x0) at /var/tmp/portage/sci-libs/opencascade-7.7.2-r1/work/OCCT-7_7_2/src/TopoDS/TopoDS_TShape.hxx:64
64	  void Free (Standard_Boolean theIsFree) { setFlag (TopoDS_TShape_Flags_Free, theIsFree); }
(gdb) up
#2  TopoDS_Builder::Add ***@***.***=0x7ffd23863063, aShape=..., aComponent=...) at /var/tmp/portage/sci-libs/opencascade-7.7.2-r1/work/OCCT-7_7_2/src/TopoDS/TopoDS_Builder.cxx:49
49	  aComponent.TShape()->Free(Standard_False);
(gdb) p aComponent
$1 = (const TopoDS_Shape &) @0x7ffd238631f0: {myTShape = {entity = 0x0}, myLocation = {myItems = {myNode = {entity = 0x0}}}, myOrient = TopAbs_EXTERNAL}
(gdb) up
[... cut ...]
$3 = {<TopoDS_Shape> = {myTShape = {entity = 0x0}, myLocation = {myItems = {myNode = {entity = 0x0}}}, myOrient = TopAbs_EXTERNAL}, <No data fields>}
(gdb) up
#4  0x00007f6f811bb4b0 in ShapeUpgrade_UnifySameDomain::UnifyFaces ***@***.***=0x564f88339690) at /var/tmp/portage/sci-libs/opencascade-7.7.2-r1/work/OCCT-7_7_2/src/ShapeUpgrade/ShapeUpgrade_UnifySameDomain.cxx:2807
2807	    IntUnifyFaces(exps.Current(), aGMapEdgeFaces, aGMapFaceShells, aFreeBoundMap);
```

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

adam-urbanczyk, 2024-05-21:
AFAICT `this` is null and that is the direct cause of the behavior. I'm not sure what to make out of this. Two random ideas:
(1) does changing of the toolchain help (i.e. gcc instead of clang)?
(2) what is actually added in `#3` `ShapeUpgrade_UnifySameDomain.cxx:3613`? Is `CurEdge` null (`CurEdge.IsNull()`)?

erooke, 2024-10-11:
For any future travelers: this was fixed for me by applying the [blobfish patch](https://github.com/conda-forge/occt-feedstock/blob/f95e88194e0a98ea1e92ff35cb4911af642e19aa/recipe/patches/blobfish.patch). For some reason the new set of [patches](https://github.com/conda-forge/occt-feedstock/tree/main/recipe/patches) does not seem to fix it.

l29ah, 2024-10-13:
> For any future travelers: this was fixed for me by applying the [blobfish [...]

Indeed, this patch fixes the segfaults for me, thank you for your nicely-documented investigation (https://github.com/erooke/ocp.nix/issues/2)!
I hope CadQuery will stop depending on patched opencascade in the future, as now it makes it really awkward to package it anywhere but in NixOS-like distros perhaps, since other software (like freecad, kicad, prusaslicer) would likely prefer/require vanilla opencascade.

adam-urbanczyk, 2026-09-05:
OCCT does not use the patche anymore and CQ tests do not segfault AFAICT.

closed 2026-09-05 by adam-urbanczyk