CadQuery/OCP #150: OCCT 7.8.0+ no longer uses HashCode instead uses std::hash

jdegenstein, 2024-06-06:
Not sure if this is the right place to file this issue. I am wondering if the new `std::hash` methods in OCCT 7.8.0+ are properly bound with pybind11's `py::hash` so that they are accessible from OCP?

This comment from one of the OCCT devs seems to indicate that an alternative approach would be to use `TopTools_ShapeMapHasher{}(aShape);` but I have not been able to get this to work in OCP yet.

https://dev.opencascade.org/node/84745#comment-25254

Link to relevant C++ code for TopTools_ShapeMapHasher in the OCP repo:
https://github.com/CadQuery/OCP/blob/636da4ca7c6d70d924dbeeeb52950f0a907fe99c/opencascade/TopTools_ShapeMapHasher.hxx#L25

adam-urbanczyk, 2024-06-06:
Probably not, note that OCP 7.8 is not yet released.

jdegenstein, 2024-06-06:
OK thank you. For context, I created cadquery-ocp==7.8.1 wheels from the cadquery/label/dev conda-forge package in my fork here https://github.com/jdegenstein/ocp-build-system/tree/resync 

py311 wheels that I made are here (not including MacOS arm64):
https://github.com/jdegenstein/ocp-build-system/releases/tag/7.8.1

I have just been testing 7.8.1 for breaking changes, and this one with hashing is potentially significant.

jdegenstein, 2024-06-06:
Sure, marked it as pre-release.

jdegenstein, 2024-06-17:
I tested the most recent cadquery/label/dev conda package and the hash functionality appears to be working correctly. Thank you! 

I have noticed another few issues, would you like me to document them or is it not necessary yet?

adam-urbanczyk, 2024-06-17:
For now not needed, thanks

closed 2024-06-17 by jdegenstein