CadQuery/OCP #153: PMI annotations implementation (TDF_Label.FindAttribute from OCP not working properly)

GabrielJMS, 2024-08-14:
Hello,

I'm trying to implement the reading of PMI annotations (Geometrical Dimensions and tolerances) from a step file using OCP.

I have a C++ code that is working fine and that I have already tested:
[image]

In my C++ code, I have something like this:

```cpp

m_gdtTool = XCAFDoc_DocumentTool::DimTolTool(doc->Main());

TDF_LabelSequence labels_datums, label_geoms, label_dims, label_dimTols;
m_gdtTool->GetDatumLabels(labels_datums);
m_gdtTool->GetDimensionLabels(label_dims);
m_gdtTool->GetGeomToleranceLabels(label_geoms);

int datumSize = labels_datums.Length();
int dimSize = label_dims.Length();
int geomSize = label_geoms.Length();
// Dimensional Tolerances
for (int i = 1; i <= dimSize; ++i)
{
    const TDF_Label& label_dim = label_dims.Value(i);
    Handle(XCAFDoc_Dimension) aDimAttr;
    label_dim.FindAttribute(XCAFDoc_Dimension::GetID(), aDimAttr);

    if (!aDimAttr.IsNull())
    {
        Handle(XCAFDimTolObjects_DimensionObject) aDimObject = aDimAttr->GetObject();
    }
}

```

So, I have translated into python:

```python
        labels_datums = TDF_LabelSequence()
        labels_dims = TDF_LabelSequence()
        labels_geoms = TDF_LabelSequence()

        self.gdt_tool.GetDatumLabels(labels_datums)
        self.gdt_tool.GetDimensionLabels(labels_dims)
        self.gdt_tool.GetGeomToleranceLabels(labels_geoms)

        datum_size = labels_datums.Length()
        dim_size = labels_dims.Length()
        geom_size = labels_geoms.Length()
         for i in range(1, dim_size + 1):
            label_dim = labels_dims.Value(i)
            aDimAttr = XCAFDoc_Dimension()
            found = label_dim.FindAttribute(XCAFDoc_Dimension.GetID_s(), aDimAttr)

            if found:
                aDimObject = aDimAttr.GetObject()
```
When the execution gets in the line `aDimObject = aDimAttr.GetObject()` I get this error:

[image]

I have verified and compared the label_dim variable in both python and c++ :
### python
[image]
### c++
[image]

Do you have any ideas why I'm getting this error?

I suspect that the problem is coming from TDF_Label.FindAttribute method:

I think that for my case it's falling in the third case, and maybe the pybind11 wrapping of this method is not working properly because the generic type?
[image]

I'm attaching the c++ project and the step file used to test the reading. The main file is called TestOCCT.cpp

Thank you very much for the attention. Any insight on how to solve this issue or workaround will be highly appreciated!
[TestOCCPMI.zip](https://github.com/user-attachments/files/16618172/TestOCCPMI.zip)

GabrielJMS, 2024-08-14:
Maybe the problem is in the line 59:
[image]

GabrielJMS, 2024-08-14:
The workaround was to reimplement the FindAttribute method in python using OCP
```python
    def find_attribute(label, guid):
        """
        Reimplementation of the TDF_Label::FindAttribute method in Python.

        :param label: The label to search in.
        :type label: TDF_Label
        :param guid: The GUID of the attribute to find.
        :type guid: Standard_GUID

        :return: A tuple where the first element is a boolean
            indicating if the attribute was found, and the second element is the found attribute handle.
        :rtype: (bool, TDF_Attribute)
        """
        if label.IsNull():
            raise Standard_NullObject("A null Label has no attribute.")

        itr = TDF_AttributeIterator(label)

        while itr.More():
            attr = itr.Value()
            if attr.ID().IsSame(guid):
                return True, attr
            itr.Next()

        return False, None
     ```
     
Does anyone have a suggestion for refactoring the C++ FindAttribute method to improve its compatibility with pybind11? Please let me know if you do, so I can propose a change to the OpenCASCADE project.

hyj1787848067, 2025-02-17:
18/5000
What software did you use to generate this STEP