CadQuery/OCP #167: Solution to functions that tries modifying immutable arguments passed by reference

GabrielJMS, 2025-04-08:
Hello,

Have you found any solution for handling functions that return values via output parameters passed by reference, especially when those parameters are of immutable types?

This becomes a problem with important functions like [BRep_Tool::CurveOnSurface](https://dev.opencascade.org/doc/occt-7.2.0/refman/html/class_b_rep___tool.html#aeaa70c29f0a30d827fd8a36116d74b91). In such cases, the function modifies the arguments passed by reference (e.g., to return the first and last parameters of the Geom2d_Curve), but when these are immutable types in the binding language, we can’t retrieve the modified values.

This effectively blocks access to important return data.

Some relevant references:

[pybind11 FAQ – Limitations involving reference arguments](https://pybind11.readthedocs.io/en/stable/faq.html#limitations-involving-reference-arguments)

[pyOCCT developer notes – Reference arguments](https://pyocct.readthedocs.io/en/latest/dev.html#reference-arguments)

Would appreciate any guidance or workarounds!

Thanks in advance.

adam-urbanczyk, 2025-04-11:
The solution is to add workaround method in the binding manually or via rules. What is your use case (not CQ I presume) case and why is this function so important?

GabrielJMS, 2025-04-17:
Hello @adam-urbanczyk , thank you for your quick response.

In some data science workflows involving BRep geometry, we need to work directly in the parametric (UV) domain of surfaces to perform calculations on faces. For instance, we often need to construct 2D edges and wires that represent the boundaries of a face in this parametric space.

To do this, it's essential to retrieve the associated pcurve and its parameter range. This also applies to vertices—sometimes we need to extract their (u, v) coordinates on a given surface to carry out specific operations.

Since this information is already available within the BRep data structures, it would be ideal to access it directly using functions like BRep_Tool::CurveOnSurface, rather than having to recompute or approximate it manually.

Thanks again for your support!

GabrielJMS, 2025-08-06:
Hello @adam-urbanczyk Do you think would be possible to expose the `Standard_Real` object, so we could use something like this to solve this issue:

```
from OCP.Standard import Standard_Real
from OCP import BRep_Tool
from OCP.Geom2dAPI import Geom2dAPI_ExtremaCurveCurve

first = Standard_Real()
last = Standard_Real()
curve = BRep_Tool.CurveOnSurface(edge, face, first, last)

extrema_curve = Geom2dAPI_ExtremaCurveCurve(curve1, curve2)
u1 = Standard_Real()
u2 = Standard_Real()
extrema_curve.LowerDistanceParameters(u1, u2)
```

Thank you for your attention