CadQuery/OCP #169: symbols_mangled_linux.dat missing when trying to build linux wheels

olly-writes-code, 2025-04-25:
Hi! 👋 

I'm trying to build linux aarch64 wheels from the current master branch.

I've followed the flow with pywrap but I'm hitting an issue.

When I run
```
pywrap -l $LIBCLANG_PATH/libclang-14.so.1 all ocp.toml 
```
from the root of the project I get 
```
FileNotFoundError: [Errno 2] No such file or directory: 'symbols_mangled_linux.dat'
```

It looks like this is because the ocp.toml points to the [file](https://github.com/CadQuery/OCP/blob/master/ocp.toml#L373) but since version [7.7.10](https://github.com/CadQuery/OCP/blob/7.7.1.0/symbols_mangled_linux.dat) symbols_mangled_linux.dat doesn't exist. See it's gone in [7.7.2.0](https://github.com/CadQuery/OCP/blob/7.7.2.0/symbols_mangled_linux.dat). 

I do notice the file in [pywrap though](https://github.com/CadQuery/pywrap/blob/master/symbols_mangled_linux.dat).

olly-writes-code, 2025-04-25:
It seems like this can be fixed by changing the ocp.toml to
```
[Linux]
    modules = ["Xw"]
    symbols = "pywrap/symbols_mangled_linux.dat"
```

adam-urbanczyk, 2025-04-26:
https://github.com/CadQuery/OCP/blob/01055e099a3b1a34da38bb1f5457f06a2d4c7a18/CMakeLists.txt#L33

You are supposed to use the released sources or run via cmake

adam-urbanczyk, 2025-04-26:
README.md needs to be updated regarding this step, but it does state that generating is not advised.

olly-writes-code, 2025-04-26:
Thanks @adam-urbanczyk ! What do you think the best process is for generating aarch64 wheels? 

From the README.md I was planning on doing something like
```
pywrap all ocp.toml
cmake -S OCP -B build
cmake --build build
```

adam-urbanczyk, 2025-04-27:
I don't now what is the best process for generating wheels for any platform, this repo dos not provide any. My point is: you should not try to generate the sources, just use the released ones to build with your favorite package manager.

Cf.
https://github.com/conda-forge/ocp-feedstock/blob/main/recipe/bld.bat
https://github.com/conda-forge/ocp-feedstock/blob/fdb460bb4480706ee8b72435e3b914473d09c14f/recipe/meta.yaml#L20

You probably should open this issue here:
https://github.com/CadQuery/ocp-build-system

olly-writes-code, 2025-04-28:
got it! I hadn't realized the build system had it's own repo. I'll head over there. Thanks again Adam!

closed 2025-04-28 by olly-writes-code