CadQuery/OCP #172: Build error for Debian packaging (help needed)

rolandmas, 2025-06-08:
Hi, your friendly Debian maintainer here :-)
I've been tasked with packaging CadQuery for Debian; it requires OCP, which in turn requires pywrap.
I managed to get pywrap to build (using a current Git snapshot rather than the occt7.4 tag), but now I'm stuck when trying to build OCP.
It's quite possible that I do things wrong, but the errors I get seem to go beyond what I did as a packager: lots and lots of syntax errors during compilation. I tried various compilers (gcc-11, gcc-12, gcc-13, gcc-14 and clang-19) but it doesn't seem to make a difference apart from the amount of errors they tolerate before giving up.

A few samples of the error messages I see (with clang):
```
cd /build/python-ocp-7.8.1.2/obj-x86_64-linux-gnu/OCP && /usr/bin/clang++ -DOCP_EXPORTS -I/build/python-ocp-7.8.1.2/OCP -I/build/python-ocp-7.8.1.2/OCP/OPENCASCADE_INCLUDE_DIR-NOTFOUND -isystem /usr/include/pyth
on3.13 -I/usr/include/opencascade  -Wdate-time -D_FORTIFY_SOURCE=2 -std=gnu++17 -fPIC -fpermissive -fvisibility=hidden -fvisibility-inlines-hidden -MD -MT OCP/CMakeFiles/OCP.dir/AppDef_pre.cpp.o -MF CMakeFiles/O
CP.dir/AppDef_pre.cpp.o.d -o CMakeFiles/OCP.dir/AppDef_pre.cpp.o -c /build/python-ocp-7.8.1.2/OCP/AppDef_pre.cpp
In file included from /build/python-ocp-7.8.1.2/OCP/AppDef_pre.cpp:159:
/build/python-ocp-7.8.1.2/OCP/NCollection_tmpl.hxx:96:77: error: expected unqualified-id
   96 |              (iterator (NCollection_DataMap<TheKeyType,TheItemType,Hasher>::*)() const) &NCollection_DataMap<TheKeyType,TheItemType,Hasher>::begin,
      |                                                                             ^
/build/python-ocp-7.8.1.2/OCP/NCollection_tmpl.hxx:96:78: error: expected expression
   96 |              (iterator (NCollection_DataMap<TheKeyType,TheItemType,Hasher>::*)() const) &NCollection_DataMap<TheKeyType,TheItemType,Hasher>::begin,
      |                                                                              ^
/build/python-ocp-7.8.1.2/OCP/NCollection_tmpl.hxx:99:77: error: expected unqualified-id
   99 |              (iterator (NCollection_DataMap<TheKeyType,TheItemType,Hasher>::*)() const) &NCollection_DataMap<TheKeyType,TheItemType,Hasher>::end,
      |                                                                             ^
/build/python-ocp-7.8.1.2/OCP/NCollection_tmpl.hxx:99:78: error: expected expression
   99 |              (iterator (NCollection_DataMap<TheKeyType,TheItemType,Hasher>::*)() const) &NCollection_DataMap<TheKeyType,TheItemType,Hasher>::end,
      |                                                                              ^
/build/python-ocp-7.8.1.2/OCP/NCollection_tmpl.hxx:102:83: error: expected unqualified-id
  102 |              (const_iterator (NCollection_DataMap<TheKeyType,TheItemType,Hasher>::*)() const) &NCollection_DataMap<TheKeyType,TheItemType,Hasher>::cbegin,
      |                                                                                   ^
/build/python-ocp-7.8.1.2/OCP/NCollection_tmpl.hxx:102:84: error: expected expression
  102 |              (const_iterator (NCollection_DataMap<TheKeyType,TheItemType,Hasher>::*)() const) &NCollection_DataMap<TheKeyType,TheItemType,Hasher>::cbegin,
      |                                                                                    ^
/build/python-ocp-7.8.1.2/OCP/NCollection_tmpl.hxx:105:83: error: expected unqualified-id
  105 |              (const_iterator (NCollection_DataMap<TheKeyType,TheItemType,Hasher>::*)() const) &NCollection_DataMap<TheKeyType,TheItemType,Hasher>::cend,
      |                                                                                   ^
/build/python-ocp-7.8.1.2/OCP/NCollection_tmpl.hxx:105:84: error: expected expression
  105 |              (const_iterator (NCollection_DataMap<TheKeyType,TheItemType,Hasher>::*)() const) &NCollection_DataMap<TheKeyType,TheItemType,Hasher>::cend,
      |                                                                                    ^
```

Another bunch (with gcc-14):
```
2573 |         .def("Update3dMouse",
      |         ~~~~^~~~~~~~~~~~~~~~~
 2574 |              (bool (AIS_ViewController::*)( const WNT_HIDSpaceMouse &  ) ) static_cast<bool (AIS_ViewController::*)( const WNT_HIDSpaceMouse &  ) >(&AIS_ViewController::Update3dMouse),
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 2575 |              R"#(Process 3d mouse input event (redirects to translation, rotation and keys).)#"  , py::arg("theEvent")
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 2576 |           )
      |           ~  
/usr/include/pybind11/pybind11.h:324:73:   in 'constexpr' expansion of 'pybind11::detail::descr<18, AIS_ViewController, WNT_HIDSpaceMouse>::types()'
/usr/include/pybind11/pybind11.h:324:73: error: 'constexpr' call flows off the end of the function
  324 |         PYBIND11_DESCR_CONSTEXPR auto types = decltype(signature)::types();
      |                                               ~~~~~~~~~~~~~~~~~~~~~~~~~~^~
make[3]: *** [OCP/CMakeFiles/OCP.dir/build.make:82: OCP/CMakeFiles/OCP.dir/AIS.cpp.o] Error 1
```

I'll attach the full build logs to this ticket because I'm out of ideas, could anyone provide a hint?

[log-clang19.log](https://github.com/user-attachments/files/20646502/log-clang19.log)
[log-gcc11.log](https://github.com/user-attachments/files/20646505/log-gcc11.log)
[log-gcc12.log](https://github.com/user-attachments/files/20646503/log-gcc12.log)
[log-gcc13.log](https://github.com/user-attachments/files/20646504/log-gcc13.log)
[log-gcc14.log](https://github.com/user-attachments/files/20646501/log-gcc14.log)

adam-urbanczyk, 2025-06-09:
Hey, you don't need to ( and should not) generate your own OCP sources. Simply use the released sources just like the conda forge recipe.

adam-urbanczyk, 2025-06-13:
I don't think I have the bandwidth to debug this. Likely your environment generates different sources wrt what we get in CI.

rolandmas, 2025-06-13:
Fair enough. Are the build logs from CI available somewhere, so I could investigate the differences myself?

jmwright, 2025-06-13:
@rolandmas Your first resource on this will be @adam-urbanczyk , but I'll provide some information that hopefully helps.

There are some logs on Azure Pipelines for this repo. Check steps such as the Compile step [here](https://dev.azure.com/cadquery/OCP/_build/results?buildId=4314&view=logs&jobId=a1fb9864-b881-5568-0b92-71e1df4c2b07&j=a1fb9864-b881-5568-0b92-71e1df4c2b07&t=9425d66d-6cf9-5299-f718-f7373dd2de8d).

You could also have a look at the [ocp-build-system repo](https://github.com/CadQuery/ocp-build-system/blob/main/.github/workflows/build-ocp.yml), which is responsible for building our Python wheels for OCP. Unfortunately all the logs from the builds have expired.

adam-urbanczyk, 2025-06-15:
BTW my guess is that you are using different llvm/clang version than one used for generating (15). I'm trying to switch to 20 now and I'm also getting some issues.

adam-urbanczyk, 2026-01-12:
@rolandmas does the 7.9.3 work better ?