CadQuery/OCP #175: Performance issues on Macbook Pro M1.

WirajanDASILVA, 2025-06-25:
anyone experiencing performance issues when using cadquery-ocp on a macbook pro m1? operations that take 30s in a linux or windows machine take 9 mins on the macbook. Appreciate any help.

jmwright, 2025-06-25:
@WirajanDASILVA Are you able to test using conda to see if you have the same performance issue? We need to determine if it is an OCP issue or a wheel issue.

BTW, the repo that is responsible for the wheel builds is here: https://github.com/CadQuery/ocp-build-system

WirajanDASILVA, 2025-06-25:
@jmwright, thanks for the response. I've just tested using conda, and I had same performance issue..

jmwright, 2025-06-25:
@WirajanDASILVA Is there any kind of Intel-compatibility mode involved (Rosetta) even though the wheels/packages are built for ARM? I know very little about Macs.

WirajanDASILVA, 2025-06-25:
@jmwright no...
 I've verified everything was arm based, since brew to python version and packages. the same using conda.

jmwright, 2025-06-25:
@WirajanDASILVA Thanks, good to know.